raspberrypi/firmware #645: Possible OTP fail

Thepiggy, 2016-08-22:
Pi3
I was trying to enable usb boot mode for my Pi3

But I accidentally had the program_usb_boot_mode=1 in my /boot/config.txt before updating to the 'next' branch of the firmware and now that I've updated and restarted it still shows under "vcgencmd otp_dump" that value 17: is 1020000a not the desired 3020000a, I think because I had the program_usb_boot_mode line before updating to the 'next' branch I've ruined my π's ability to usb boot forever. Is this correct or is something else preventing the value of my OTP from changing.. (Ps I've also reinstalled raspbian and tried the steps again, properly this time, and it still didn't work)

If this is something that's possible (destroy the π's ability to usb boot forever) I'd consider resolving it so it's not possible for others to bug up their π like I have mine. (Maybe change the current firmware so the program_usb_boot_mode line doesn't do anything)

If this is something that's not possible, and I'm simply doing something else wrong or this is a bug, I'd be ecstatic.

pelwell, 2016-08-23:
Reading the code shows that trying to program bit 29 is only conditional on program_usb_boot_mode being set to 1 (and not having hit a fatal error earlier - if your board is booting then this isn't the case).

Are you sure you are editing the correct config.txt? NOOBS can make this confusing. Try adding something like "dtoverlay=missing", rebooting and checking the output of `sudo vcdbg log msg` for errors. You should also try `sudo vcdbg log assert` in case the OTP programming is reporting a failure.

popcornmix, 2016-08-23:
``program_usb_boot_mode=1`` is a no-op on master branch. There is no reference to that config setting in the source tree so using that setting on master branch didn't cause a problem.

Ping @ghollingworth

Thepiggy, 2016-08-23:
@pelwell, I'm not using NOOBS, I'm using normal raspbian (not lite or NOOBS) would you like me to add "dtoverlay=missing" with or without "program_usb_boot_mode=1" also in the /boot/config.txt file?

I'll run the commands and comment the output shortly.

pelwell, 2016-08-23:
Without NOOBS it is unlikely you've got the wrong file, but the the vcdbg log assert output might be interesting.

Also, can you confirm the contents of /boot/.firmware_revision?

Thepiggy, 2016-08-23:
Contents of /boot/.firmware_revision:
http://pastebin.com/kFq3JBZb

"sudo vcdbg log assert" output:
No messages available

"sudo vcdbg log msg" output:
http://pastebin.com/feC4aySm

pelwell, 2016-08-23:
That revision is from the "master" branch, not "next". Try running `sudo BRANCH=next rpi-update` and see if that makes a difference.

Thepiggy, 2016-08-23:
I've already done that, after running the command again I get this:
http://pastebin.com/YJrxe0pR

Starting to think this is a rpi-update bug..

Thepiggy, 2016-08-23:
Can I do this without rpi-update updating the firmware? Such as manually coping the files?

dlech, 2016-08-24:
You misspelled `BRANCH`

Thepiggy, 2016-08-24:
I'm relived to say all this trouble was indeed caused by me. The π has the correct value of 3020000a now that I've spelt the command correctly. (could've sworn I used copy and paste for that command from someplace originally)

All is right with the world again, I'll have my army of network booted πs shortly after I get this one working!

Thank you everyone for your help, and dealing with my stupidity. 

I think its time to close this issue (and probably change the title to something more appropriate).

pelwell, 2016-08-24:
Phew/oops.

closed 2016-08-24 by pelwell