raspberrypi/firmware #647: Failed booting RPi 3 from USB stick

incahellas, 2016-08-25:
I did everything from https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md and my RPi 3 didn't boot neither without SD card nor with the special SD card containing only bootcode.bin. The latter showed the very initial colorful screen, and kept periodically flashing the LED.
lsusb reports my 16GB USB stick as: 'Bus 001 Device 005: ID 0951:1624 Kingston Technology DataTraveler G2'
With the normal Raspbian SD card, I can use it successfully as root filesystem.

popcornmix, 2016-08-25:
Ping @ghollingworth - is this the best place for bug reports and/or reporting devices that work/don't work?

incahellas, 2016-08-25:
From https://www.raspberrypi.org/blog/pi-3-booting-part-i-usb-mass-storage-boot/ :
"If this still doesn’t work, please open an issue in the firmware repository"

ghollingworth, 2016-08-25:
> mass-storage-boot/ :

Yeah,

For specific devices at the moment at least I'd like to be able to discuss
them in an issues list...

Have you tried adding the timeout file to the SD card?  Does this help with
booting?

One other thing to try is to access it from a standard Windows PC and check
that it appears as a mass storage device immediately (i.e. with no delay at
all)...  I had a problem where it took ages (literally 10 seconds) to
enumerate my DataTraveller on a Raspberry Pi just running normal Raspbian
from an SD card but after plugging into a Windows PC and ejecting it
properly it suddenly started responding immediately!  It would be
interesting to see if that's what's happening to yours...

Gordon

incahellas, 2016-08-25:
What do you mean by 'timeout file'? I haven't seen anything about this in the instructions.

Anyway, when I use it as root filesystem with a normal Raspian SD card, I don't notice any significant delay, Tried with rootdelay=3 and kept booting normally.

ghollingworth, 2016-08-26:
Sorry I realised I hadn't documented that in the standard docs...  Please see:

https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bootmodes/README.md

Just add a file 'timeout' to the SD card alongside the bootcode.bin to see if this helps your booting issues.

Please boot a Raspberry Pi from a standard SD card and on the command line do:

tail -f /var/log/messages

Then plug in the mass storage, what is the time between plugging in the MSD and the first message appearing on the screen (approx) is it < 2 seconds?

ghollingworth, 2016-08-26:
Can you please add a picture of the device so that I can see if I can get the same one?  Currently I've got a G3 device which works fine, but no G2 device

Gordon

incahellas, 2016-08-26:
The 'timeout' file didn't change anything, and that's probably because this is not the problem here. The test with tail -f /var/log/messages showed that RPi recognizes the USB stick almost instantaneously. I would say, definitely less than 1 second

[image]

ghollingworth, 2016-08-30:
Got one on order, will see what happens...

ghollingworth, 2016-08-31:
Just tried three different looking DataTraveller G2 devices and they all work correctly with the standard boot code...

If you still can't get it working, would you be OK to send one to me here at Raspberry Pi Towers?  Email me at gordon@raspberrypi.org and we can arrange swapping something else with you

Thanks

Gordon

incahellas, 2016-09-01:
As I emailed you, I did everything from scratch and booted OK.

I did, however, two more tests. I converted the partition table first to GPT with gdisk, in order to use the 'root=PARTUUID=...' syntax in /boot/cmdline.txt and boot from the correct stick/disk, in case there are more than one.  But it stopped booting.
Then I tried a hybrid MBR partition table. This didn't boot either.

ghollingworth, 2016-09-02:
GPT should work, although I've only tested it when formatted from scratch
from Linux, Mac and Windows machines using the standard formatting tools, I
don't know if gdisk does something different...

Can you send me an image of the disk which just includes the whole of the
first FAT partition?

Normally to do this I'd use parted on another Pi or a linux machine...

parted /dev/sda (replace with whichever the flash drive is)
units s
p

This should print out the table with start and end sectors, record
partition 1's end sector number

dd if=/dev/sda of=file.img bs=512 count=eeee

Replacing eeee with the end sector number + 1

The zip it and stick it on dropbox or something similar...

Thanks

Gordon

incahellas, 2016-09-02:
The image you asked for:
[https://www.dropbox.com/s/pvlyhwhdryz13mu/file.zip?dl=0]

ghollingworth, 2016-09-05:
Ah OK,

I've had a look and realised my mistake, although the bootrom is happy and starting up the second stage the support for GUID partitions isn't in the start.elf so it is failing to find a kernel...

This needs to be added as a separate issue though

#654

closed 2016-09-05 by ghollingworth