raspberrypi/firmware #649: Available memory restricted to 120204 kB on RPi 3 (1GB)

brice-fr, 2016-08-27:
After updating to the current next BRANCH my amount of available memory suddenly decreased with any possibility to play on gpu_mem or startx parameters to get it up again. 

sorry if this is a documented feature I overlooked.

sudo BRANCH=next rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
~...
 *** Backing up modules 4.4.11-v7+
 *** Downloading specific firmware revision (this will take a few minutes)
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 4.4.17-v7+
 *** depmod 4.4.17+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to 6272ee1c3fd9f0a87015adb36830a18f3e3b7245
 *** A reboot is needed to activate the new firmware
pi@raspberrypi:~ $ cat /proc/meminfo
MemTotal:         **882788 kB**
MemFree:          521932 kB
~...
pi@raspberrypi:~ $ sudo reboot
~...

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sat Aug 27 16:54:28 2016 from fe80::8638:35ff:fe63:3b18%eth0
pi@raspberrypi:~ $ cat /proc/meminfo
MemTotal:         **120204 kB**
MemFree:           44712 kB
~...

popcornmix, 2016-08-27:
Known problem. See: https://github.com/Hexxeh/rpi-update/issues/216

DougieLawson, 2016-08-29:
I'm getting this when I boot from a USB MSD

```
root@challenger:~# vcgencmd get_mem arm
arm=128M
root@challenger:~# vcgencmd get_mem gpu
gpu=128M
root@challenger:~#
```

I've tried tweaking gpu_mem=xxx in config.txt but that gets completely ignored. If I boot from an SDCard using the master branch (4.4.19-v7+ # 906) everything is normal

BTW @popcornmix you've created a circular reference, Hexxeh/rpi-update#216 points to here.

vpetersson, 2016-08-31:
👍 

Really hoping this one will be sorted out soon as it is blocking more or less USB boot (which I'm super excited about :)

andrewlavens, 2016-08-31:
I was able to get back to an earlier build by connecting over SSH and running 
`sudo BRANCH=next rpi-update 193f081a0ed502b62483aa707a79cf245db9b028`
which got the memory back along with USB boot. I'm new to this though so YMMV.

popcornmix, 2016-08-31:
Actually a BRANCH=next firmware was pushed yesterday that included a possible fix - does this work any better?

andrewlavens, 2016-08-31:
Ooo - I'll have a try when I'm back home tonight.

ghollingworth, 2016-08-31:
I've tested this and happy that it works correctly...  Please close when you're happy

vpetersson, 2016-08-31:
I'm in the process of testing this as we speak ;)

vpetersson, 2016-08-31:
👍 👍 👍 
Works! Thanks guys!

vpetersson, 2016-08-31:
If anyone is looking to streamline the process of migrating from SD to HDD, i put together a short script for it [here](https://github.com/vpetersson/raspberrypi-sd2hdd).

andrewlavens, 2016-08-31:
Working here. Thanks very much.

DougieLawson, 2016-08-31:
Thank you, it's back to 
```
root@challenger:~# vcgencmd get_mem gpu
gpu=64M
root@challenger:~# vcgencmd get_mem arm
arm=944M
root@challenger:~#
```
Normality is restored.

popcornmix, 2016-09-01:
@brice-fr okay to close?

brice-fr, 2016-09-02:
My memory is back

_pi@raspberrypi:~ $ cat /proc/meminfo
MemTotal:         947744 kB_
~

But for the moment I am having a very strange displayed split

_pi@raspberrypi:~ $ vcgencmd get_mem gpu
gpu=64M
pi@raspberrypi:~ $ vcgencmd get_mem cpu
cpu=0M_

Will check that and document tonight from a fresh install

pelwell, 2016-09-02:
Try `*vcgencmd get_mem *arm`.

brice-fr, 2016-09-02:
oops sorry that was that

_vcgencmd get_mem arm
arm=944M_

Ok to close

closed 2016-09-02 by pelwell