raspberrypi/firmware #650: Occasional possibly hangs in GPU firmware on reboot

Electron752, 2016-08-28:
I mostly want to know if others have reported this and if it's a possible known issue before I continue to bang my head against the wall.

As you might know, I have some modifications to the linux kernel to support 64bit on the RPI 3.  Occasionally, say 1 out of 10 reboots the kernel hangs on an mbox call to the GPU/firmware for something simple like turning the USB power on.   The thread is hung waiting on the completion event of the mbox call.   

It's possible something is wrong on my end, but I'm starting to suspect that it's the GPU/firmware side of things that is hanging.   The only way to recover is to unplug the RPI for awhile.  Sometimes, but not always, I see some video malfunctions around the time this is happening such as the monitor will switch rapidly between on mode and power saving mode(while any mbox calls are blocked of course).

Here is the kernel debug printout of the firmware version:
[    0.121402] raspberrypi-firmware soc:firmware: Attached to firmware from 2016-08-23 15:55

popcornmix, 2016-08-30:
No thoughts currently.
Is the failure rate the same when you remove power or if you ``sudo reboot`` ?

Any peripherals attached? Does the problem occur with everything disconnected?
Any overclock or other non-standard config.txt settings?
Do you ever get under-voltage reported? (check with ``vcgencmd get_throttled``)
Can you test with a different power supply?

Electron752, 2016-08-31:
1. "Is the failure rate the same when you remove power or if you sudo reboot ?"

The failure rate is much higher with sudo reboot then with the power pulled.

"Any peripherals attached? Does the problem occur with everything disconnected?
Any overclock or other non-standard config.txt settings?"

Nothing is attached except for HDMI and the ethernet cable.  I remotely control the RPI with ssh.  Nothing in config.txt at all.

"Do you ever get under-voltage reported? (check with vcgencmd get_throttled)
Can you test with a different power supply?"

I'll do more testing on this.  I have a benchtop "Lab Grade" linear power supply that can do 6A.   Since I can dial up the voltage, I'll see if this is voltage dependent. Rarely does the RPI 3 require more then .5A when all the USB ports are unconnected(except for some very short transients) so I think 6A should be more then adequate.

Again, have you had any reports or this or seen this in your testing?  I highly doubt this is 64bit vs 32bit dependent.

popcornmix, 2016-09-01:
No this is the first report like this I've seen.

I guess a useful test point would be for you to run a standard clean raspbian sdcard and check if you get the occurrence of failed reboots.

Electron752, 2016-09-03:
OK, I'm able to trace the firmware hangs to the GPU call to change the ARM clock frequency.  Since I'm using the "on-demand" governor this happens a lot during boot.   If the frequency is changed too frequently during boot, the GPU/firmware hangs.

Also, I'm seeing that the firmware calls to change the ARM frequency are very badly broken.  **It appears that a call to change the frequency to 1.2GHz works but if the frequency is ever changed back to 600MHz by the governor, then frequency stays stuck at 600MHz.   Repeated calls to change back to 1.2GHz hangs the firmware.**

I added some extra logging to the cpufreq code to query the arm frequency from the firmware as soon as the firmware call to change the frequency returns.

Here are some interesting pieces from dmesg:
[    0.351579] raspberrypi-firmware soc:firmware: Attached to firmware from 2016-09-02 11:49
...
[    2.326671] bcm2835-cpufreq: min=600000 max=1200000
[    2.333000] sdhci: Secure Digital Host Controller Interface driver
**[    2.333782] cpufreq: set clock 1200000000
[    2.336564] cpufreq: finished setting clock to 1200000000**
...
[   12.308602] cpufreq: set clock 600000000
[   12.314915] cpufreq: finished setting clock to 600000000
[   12.665573] cpufreq: set clock 1200000000
[   12.667018] cpufreq: finished setting clock to 600000000
[   13.053590] cpufreq: set clock 600000000
[   13.054981] cpufreq: finished setting clock to 600000000
**[   20.721669] cpufreq: set clock 1200000000
[   20.723119] cpufreq: finished setting clock to 600000000**
[   21.081646] cpufreq: set clock 600000000
[   21.083028] cpufreq: finished setting clock to 600000000
...

Here is my  full config.txt:
disable_overscan=1

Section of the cpufreq code that I changed to add extra logging:

static uint32_t bcm2835_cpufreq_set_clock(int cur_rate, int arm_rate)
{
        u32 rate = arm_rate * 1000;
        int ret;

        WARN_ON(rate<6000000);
        BUG_ON(rate<6000000);

        pr_err("cpufreq: set clock %u\n", rate);

        ret = bcm2835_cpufreq_clock_property(RPI_FIRMWARE_SET_CLOCK_RATE, VCMSG_ID_ARM_CLOCK, &rate);
        if (ret) {
                print_err("Failed to set clock: %d (%d)\n", arm_rate, ret);
                return 0;
        }

        rate = bcm2835_cpufreq_get_clock(RPI_FIRMWARE_GET_CLOCK_RATE);

        pr_err("cpufreq: finished setting clock to %u\n", rate*1000);

        print_debug("Setting new frequency = %d -> %d (actual %d)\n", cur_rate, arm_rate, rate);

        return rate;
}

pelwell, 2016-09-03:
The firmware won't allow the clock to increase if the temperature is too high or the voltage is too low. What does `vcgencmd get_throttled` return?

Electron752, 2016-09-03:
I don't have that command available in my environment.  I'll see if I can get it running...

Anyway, I'm now 95% sure at this point that you're correct and this is a voltage/electrical issue.   I increased the voltage on the benchtop power supply by 0.25V( to 5.25V) and the issues seem to completely disappear.  Remove the 0.2V increase and the issues suddenly reappear.

I'm wondering if 64bit is somehow more taxing on the system and it makes it more sensitive to voltage issues? I also thought the 5.25V was only required if peripherals where connected and the power supply wasn't that good.

I'll leave the issue open for a bit longer, but if I don't find anything else that suggests at this point that it isn't an electrical issue I'll close it.

closed 2016-09-04 by Electron752

popcornmix, 2016-09-04:
The power consumed by a processor will scale with number of bits that flip per second internally.
So, 64-bit may use more power than 32-bit. In reality few instructions will toggle 64-bits so I'd expect the increase to be modest (e.g. 5-10%), except for deliberate "power" stress tests.

Nothing on the Pi wants more than 5V. You do probably have some voltage drop in your cable (which increases with load) so I'm sure the voltage on the Pi is lower than 5V if you are getting crashes. You can measure this on the GPIO header if you want to be sure.

But setting the power supply to 5.25V should be fine as there will be some loss and it is still within spec.