raspberrypi/firmware #651: CONFIG_SECCOMP but no CONFIG_SECCOMP_FILTER produces systemd issues

pbatard, 2016-08-30:
A [few](https://github.com/Hexxeh/rpi-update/issues/219) [distros](bugs.debian.org/cgi-bin/bugreport.cgi?bug=832713) are currently experiencing issues with systemd due to `SECCOMP` not being fully configured with the rpi kernel options.

It appears that `CONFIG_SECCOMP` is enabled in the kernel config, but the lack of filter support (`CONFIG_SECCOMP_FILTER` not being set) means systemd cannot properly detect and set `SECCOMP`, which results in critical systems such as `journald`, `network` or `logind` failing to start.

Could we have either `CONFIG_SECCOMP_FILTER` set in future versions of the kernel?

Please see [Debian bug 832713](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=832713#314) for more.

clivem, 2016-08-30:
ISTR, the SECCOMP_FILTER is not available while they want to keep CONFIG_OABI_COMPAT. I think this came up the other day in a similar report.

pbatard, 2016-08-30:
Okay. Maybe `SECCOMP` should be disabled altogether then?

Of course, in parallel to this request, there's also work being carried out to improve `SECCOMP` and filter support detection in systemd, to try to address the current issues.

clivem, 2016-08-30:
That's up to the RPi guys, in their builds. ;) I'm pretty sure that they want to have OABI_COMPAT enabled (for a reason I have forgotton) so perhaps may be best to disable SECCOMP all together if that will make systemd happy. I SECCOMP_FILTER have in my Fedora builds, but I don't need OABI_COMPAT.

clivem, 2016-08-30:
@pbatard See https://github.com/raspberrypi/linux/issues/1605

Ferroin, 2016-08-31:
FWIW, OABI_COMPAT is pretty much useless to most people using a Pi.  For an OABI application that isn't statically linked to work, you have to have either a self-contained chroot built using OABI, or a whole system using it, and neither is something supported by any major distro I know of (and this isn't something that can be done with multilib without some serious work), and any remaining statically linked OABI applications are third party code for other systems.

JamesH65, 2017-12-21:
This is a linux kernel issue, and I suspect not particularly relevant, but if people want it looked at please open an issue on the Linux repo. Closing here.

closed 2017-12-21 by JamesH65