raspberrypi/firmware #652: Disable CONFIG_OABI_COMPAT.

Ferroin, 2016-08-31:
OABI_COMPAT is a backwards compatibility tool intended to support the old Linux ARM ABI.  This support is a very niche use case because:
1. The Raspberry Pi has never had an official distribution built in OABI mode.
2. Dynamically linked OABI applications require either a full system running in OABI mode, or a dedicated self-contained chroot containing all the libraries they need built in OABI mode, and doing this requires building the whole environment from scratch (including the toolchain).
3. Statically linked OABI applications are all legacy third party code for other systems.
4. All modern distributions are built in EABI mode.

In contrast, OABI_COMPAT exposes a very large attack surface, contributes to a significant amount of space taken up by the kernel (all syscalls it supports must have two entry points, one for OABI, one for EABI), and prevents selection of other config options (such as SECCOMP_FILTER, which is required by systemd when SECCOMP is enabled).

Based on this, I propose that OABI_COMPAT be disabled in the official kernel images and the rpi_defconfigs.

popcornmix, 2016-08-31:
See https://github.com/raspberrypi/linux/issues/766 for the reason we have OABI_COMPAT enabled.
If there is evidence that gdb no longer requires this, the we may be able to disable this, but we need evidence of that first or there will be unhappy users.

Ferroin, 2016-08-31:
The patch mentioned in that thread appears to be in the GDB VCS, I'm not sure if it's in a released version yet though, and I don't exactly have the ability to check immediately (all the Pi's I have access to right now are production systems which I can't afford to be swapping kernels on just to check something like this).  

Under the assumption that it should be an issue on any ARMv7 system, I'll try to throw something together with Buildroot to test in QEMU.

popcornmix, 2016-08-31:
Looks like gdb patch is not in debian jessie yet:
https://github.com/raspberrypi/linux/issues/766#issuecomment-243799421

so removing OABI_COMPAT will break things.

Ferroin, 2016-08-31:
I'll look at opening a bug in the Debian bug tracker then to see about getting that particular patch included, as this probably affects other ARM Debian systems too.

Until the patch gets included or a sufficiently recent version of GDB  gets included in Debian, I'd suggest this remains open as a reminder to remove this config option once it is no longer needed.

JamesH65, 2017-12-21:
@Ferroin Did this ever get sorted out? Note, this is the wrong repo for this question anyway. Should be in the Linux repo.