raspberrypi/firmware #653: Raspbian 4.4.19-v7+ Error on RPi 3: brcmf_c_preinit_dcmds

ciconia1, 2016-09-01:
New installation with rpi-update done. dmesg shows the following errors:

[    5.878431] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: May 27 2016 00:13:38 version 7.45.41.26 (r640327) FWID 01-df77e4a7
[    5.909533] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[    6.379695] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[    7.146946] brcmfmac: brcmf_add_if: ERROR: netdev:wlan0 already exists
[    7.146967] brcmfmac: brcmf_add_if: ignore IF event

If I blacklist brcmfmac and brcmutil and reboot, the error disappears, but so does wifi, as expected.

With the blacklist removed, I updated the bcm43438 firmware using these files, but the problem still persists:
https://github.com/RPi-Distro/firmware-nonfree/master/brcm80211/brcm/brcmfmac43430-sdio.bin
https://github.com/RPi-Distro/firmware-nonfree/master/brcm80211/brcm/brcmfmac43430-sdio.txt

Even with the error messages, it seems like wifi works ok. So, perhaps they are benign. I just wish I didn't have to see them because it makes me think there is something that is wrong in my setup. Is anyone else experiencing this same issue?

pelwell, 2016-09-02:
Those are expected errors. I agree it's not pretty, but that's what the Broadcom driver does.

Ruffio, 2016-09-02:
Yes, they are expected, look here: https://github.com/raspberrypi/firmware/issues/630
But I agree with you @ciconia1, I would also like them not to be there. Maybe if your write a mail to the dev (see in the above issue) he will suppress that 'error', I have already mentioned it to him...

ciconia1, 2016-09-02:
Thank you for taking the time to respond to my concern. I hadn't found thread #630, so I apologize for bringing it back up.

From thread #630- "We could add an exception for "00" in the driver, ie. ignore it without logging an error message."

Well, then, do it! haha!!

closed 2016-09-02 by ciconia1