raspberrypi/firmware #655: `dtoverlay=lirc-rpi` does not load LIRC device

flochtililoch, 2016-09-05:
However, restoring [lirc-rpi-overlay.dtb](https://github.com/raspberrypi/firmware/blob/0a96f63b75d23567320e74767d3ce11002397af7/boot/overlays/lirc-rpi-overlay.dtb) seems to fix the issue.

popcornmix, 2016-09-05:
Can you show your ``config.txt``?

You should be using ``lirc-rpi.dtbo`` not ``lirc-rpi-overlay.dtb``

Make sure you are using ``dtoverlay=lirc-rpi`` and not ``dtoverlay=lirc-rpi-overlay``
https://www.raspberrypi.org/documentation/configuration/device-tree.md

flochtililoch, 2016-09-05:
my `/boot/config.txt` file looks like this:

```
dtoverlay=lirc-rpi,gpio_in_pin=23,gpio_out_pin=22,debug=on
```

pelwell, 2016-09-05:
What does `cat /boot/.firmware_revision` show?

And `sudo vcdbg log msg |& grep -v -E "(HDMI|gpioman|clock)"`?

flochtililoch, 2016-09-05:
I do not have `/boot/.firmware_revision` file on my system (using HypriotOS 1.0)

```bash
$ sudo vcdbg log msg |& grep -v -E "(HDMI|gpioman|clock)"
000971.852: *** Restart logging
000974.035: Read command line from file 'cmdline.txt'
+dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 cgroup_enable=memory swapaccount=1 elevator=deadline fsck.repair=yes rootwait console=ttyAMA0,115200 kgdboc=ttyAMA0,115200
001535.813: Loading 'kernel7.img' to 0x8000 size 0x8e411c
001539.461: Kernel trailer DTOK property says yes
001539.475: Kernel trailer DDTK property says no
001543.551: Loading 'bcm2709-rpi-2-b.dtb' to 0x8ec11c size 0x38e7
001619.020: dtparam: pwr_led_gpio=35
001627.826: dtparam: uart0_clkrate=3000000
001661.228: Failed to load overlay 'lirc-rpi'
001716.890: dtparam: arm_freq=900000000
001757.001: dtparam: core_freq=250000000
001767.512: dtparam: cache_line_size=64
001786.329: Device tree loaded to 0x2fffc100 (size 0x3e09)
004358.543: vchiq_core: vchiq_init_state: slot_zero = 0xfac80000, is_master = 1
```

pelwell, 2016-09-05:
Which kernel version are you running (`uname -a`)?

flochtililoch, 2016-09-05:
```
$ uname -a
Linux butterfly 4.4.15-hypriotos-v7+ #1 SMP PREEMPT Mon Jul 25 08:46:52 UTC 2016 armv7l GNU/Linux
```

pelwell, 2016-09-05:
OK - that explains it. The problem is in the trailer on the kernel, which says it doesn't support Dynamic Device Tree (the DDTK flag), but this flag is also used to decide which type of overlay to load (.dtbo or -overlay.dtb), which is governed by the kernel version -> 4.1 and earlier generates -overlay.dtb, while 4.4 and later uses .dtbo.

Using current firmware, there is actually no need to use the trailer - the default is to expect that Device Tree is enabled, and it will look for .dtbo overlays before falling back to using -overlay.dtb. You can strip the trailer off the kernel like this:

1) Use scripts/knlinfo from the kernel tree (https://raw.githubusercontent.com/raspberrypi/linux/rpi-4.4.y/scripts/knlinfo) to display the trailer:
```
$ sudo cp /boot/kernel7.img{,.bak}  # Make a backup
$ ./knlinfo /boot/kernel7.img
Kernel trailer found at 4226592/0x407e20:
  KVer: "Linux version 4.4.19-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #906 SMP Tue Aug 23 15:53:06 BST 2016"
  DTOK: true
  DDTK: 01000000
  270X: 01000000
  283X: 00000000
  283x: 00000000
```
2) Truncate the kernel to the point where the trailer starts:
```
$ sudo truncate -s 4226592 /boot/kernel7.img  # Use the trailer position from knlinfo
$ ./knlinfo /boot/kernel7.img
* no trailer
```

pelwell, 2016-09-05:
I think there's an argument for changing the mkknlimg script in the 4.4 and later trees to always set the DDTK flag.

flochtililoch, 2016-09-06:
Great, thanks for the details @pelwell - will close that one for now and follow up with the Hypriot folks to resolve this.

closed 2016-09-06 by flochtililoch