raspberrypi/firmware #656: vcgencmd measure_clock sometimes prints a halfway value

Terminus-IMRC, 2016-09-06:
I wish here is the right place to post this.

For, at least, `isp` and `v3d`, `vcgencmd measure_clock` command sometimes prints a halfway value. For example on RPi2B:

```
$ vcgencmd measure_clock v3d
frequency(43)=250000000
$ vcgencmd measure_clock v3d
frequency(43)=249999000
$ vcgencmd measure_clock v3d
frequency(43)=250000000
```

Does the clock really change between `250MHz` and `249.999MHz`? Is this an expected behavior?

lowphas, 2016-09-07:
I have seen something very similar with Intel cpus. Some kind of rounding errors maybe?

pelwell, 2016-09-07:
This is expected behaviour. The "measure_clock" command uses the clock manager hardware to measure the frequency of the chosen clock by counting pulses in 1ms. Multiplying by 1000 gets a value in Hz. Depending on the clock phases when the test is started you may miss a cycle, hence the variation.

Terminus-IMRC, 2016-09-07:
Thanks! I see.

Isn't there a way to read frequencies directly and correctly?

popcornmix, 2016-09-07:
There is no ways that is more direct and correct this this.
We measure the frequency directly.

Just read the number and round it to the nearest MHz. That will always be consistent.

closed 2016-09-07 by Terminus-IMRC