raspberrypi/firmware #657: No IPV4 address allocated since 4.4.20 update

trulio, 2016-09-07:
my raspberry does not get an IPV4 address but an IPV6 since 4.4.20 update

trulio, 2016-09-08:
To be more precise, no IPV4 address obtained via DHCP on any interface (Eth0, wlan0, wlan1, ...)
I had to fix manually a static IP to get a connexion and roll back to previous firmware.

An IPV6 address is listed typing ifconfif, but i don't know if it is provided by dhcp or a default address.

Anyway, all IPV4 equipments seems to work fine on my network, even after reboot. 
For me the issue is related to the lastest kernel upgrade.

Regards,
Trulio

Ruffio, 2016-09-08:
Hmmm, strange. I have just updated my RPi3 to 4.4.20 and still get the same IP4 address as before update.

trulio, 2016-09-08:
i'm using a RPI2 with raspbian-ua-netinst minimal install

Ruffio, 2016-09-08:
I use 'standard Raspbian Lite':

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.4.20-v7+ #908 SMP Wed Sep 7 14:44:27 BST 2016 armv7l GNU/Linux

If you have made any network related configuration before you upgraded, then you should go thru these.
Currently you are the only one that have reported this issue and I think that if it was a general issue, then we would have heard from a lot of people by now, so to me it looks like a local issue. That is just what I'm thinking.

trulio, 2016-09-08:
could you share me your /etc/network/interfaces file content to see if i have some differences?

popcornmix, 2016-09-08:
Can you confirm that:
``sudo rpi-update 67a8ae6c5d622484111f9be5d8aaca624fa69c5e && sudo reboot``
(reverts to last 4.19 kernel) has working IPv4 without the static IP address?

trulio, 2016-09-15:
i will test tonight. Wait one or two days before closing the issue.

popcornmix, 2016-09-15:
Also, there is a kernel bump to 4.4.21 available with rpi-update.
While I have no evidence that fixes the problem, it is worth trying as a first step.

trulio, 2016-09-15:
Here is the results of my investingations : the problem is reproduced on 4.4.19 kernel on raspbian-ua-netinst.
now I imagine my broadband access provider updated my router firmware recently without noticing it and broke something in DHCP management.

what remain strange is that i can only reproduce the issue on raspbian-ua-netinst. The problem is not present on raspian lite.

i join dmesg extracts if someone want to investigate but you can close the issue (i switched on, raspian lite for my mopidy setup)

[dmesg_4.4.21_raspian_ua-netinst_ipV4 KO_with_DHCP_activated.txt](https://github.com/raspberrypi/firmware/files/475512/dmesg_4.4.21_raspian_ua-netinst_ipV4.KO_with_DHCP_activated.txt)
[dmesg_4.4.21_raspian_lite_DHCP_OK.txt](https://github.com/raspberrypi/firmware/files/475513/dmesg_4.4.21_raspian_lite_DHCP_OK.txt)
[dmesg_4.4.21_raspian_ua-netinst_static_ipV4OK_with_DHCP_deactivated.txt](https://github.com/raspberrypi/firmware/files/475514/dmesg_4.4.21_raspian_ua-netinst_static_ipV4OK_with_DHCP_deactivated.txt)

closed 2016-09-16 by popcornmix