raspberrypi/firmware #661: Can't boot a B+ with a USB HDD

DougieLawson, 2016-09-11:
I've got a B+ that has the rootfs on a USB HDD. That needs ```max_usb_current=1``` in /boot/config.txt and it needs ```rootdelay=5``` in /boot/cmdline.txt to allow the USB drive to spin up before mounting the rootfs. 

If I try to boot that using a 2GB SDCard with just ```bootcode.bin``` on it then nothing happens, the system doesn't get far enough to display the rainbow screen, the HDD just whirs and never spins up

What I need is ```bootcode.bin``` when it detects that there's no other firmware / start.elf / cmdline.txt / config.txt available on the SDCard to default to ```max_usb_current=1``` (and, possibly, to append ```rootdelay=5``` to the cmdline passed to the Linux kernel).

At the moment my B+ is back to running with a regular SDCard with only root=UUID=xxx-xxx-xxx in cmdline.txt and max_usb_current=1 in config.txt.

popcornmix, 2016-09-11:
Not really understanding this issue. Are you talking about the BRANCH=next and the new boot modes?
On the default branch you won't be able to even detect a USB drive from bootcode.bin - start.elf and kernel.img are required.
The rainbow screen comes from start.elf - bootcode.bin doesn't even know about displays.

DougieLawson, 2016-09-11:
It is the new boot modes from the BRANCH=next stuff. There's a posting on the forum saying that booting the non-RPi3Bs with just bootcode.bin on the SDCard is now possible. What I'm reporting is that doesn't work when you have a USB HDD that needs those couple of funky parms in config.txt & cmdline.txt.  

I've got an RPi3B booting from a USB MSD and with PXE (all good). The second experiment was to try and boot a RPiB+ with just a minimal SDCard (that failed).  

I'm keen to test anything that's possible.

popcornmix, 2016-09-11:
Ping @ghollingworth

ghollingworth, 2016-09-12:
So the only thing you need here is the max_usb_current for Pi B+ and Pi 2, assuming it correctly boots bootcode.bin then it should get everything else from the HDD anyway.

Will add this to the list for the bootcode / bootrom

DougieLawson, 2016-09-12:
Thanks Gordon. I'm willing to alpha test or beta test any code change you make.

petergloor, 2016-09-13:
@DougieLawson does it work now? On my Pi 2 I also added an empty file named timeout to the µSD card.
However, I experienced another issue. I installed everything according to the instructions and finally, after my first boots into /dev/sda2, I issued a` sudo apt-get update && sudo apt-get upgrade`. Unfortunately this downgraded the firmware from 4.4.17 to 4.4.13 (the current downloads are still on 4.4.11).

DougieLawson, 2016-09-13:
@petergloor I've got USB MSD and PXE running on my 3B. I'm currently doing this stuff on a B+ (because it's had a HDD for about two years now). I've not got anywhere near trying USB MSD or PXE on my 2B yet (my 2B is currently the TFTP/NFS server for booting my 3B with PXE).

The BRANCH=master kernel is a lot of commits ahead of BRANCH=next so I think @ghollingworth & @popcornmix possibly need to merge those commits before we go anywhere with a alpha/beta version for my B+ problem.

Ruffio, 2016-12-30:
@ghollingworth what is the status on this one?

Ruffio, 2017-01-22:
@DougieLawson has this been resolved?

DougieLawson, 2017-01-22:
Not as far as I know. But I've not yet tested it with 4.9.4 due to https://github.com/raspberrypi/firmware/issues/725 and I've not had a chance to test it with 4.4.44 as that just caused a boot failure on my B+ (probably due to failing to build the initramfs) [it's too late today to look deeper at that].

DougieLawson, 2017-04-21:
This one is still not resolved. Can we get bootcode.bin to always set 1.2A on the USB interface as my HDD won't spin up without that.

popcornmix, 2017-04-21:
Resolving this is a little awkward.
The pin used for ``max_usb_current`` can be overridden by a ``dt-blob.dtb`` file,
but parsing of device tree files is too complex for bootcode.bin.

In reality very few users will have done this but it means there is a use case we won't handle.

I have produced a test bootcode.bin which may enable the max current gpio:
https://drive.google.com/uc?id=0B-6zmEDJwxZER2ZJbTdyZ0FEdWc&export=download

At the moment it is disabled by default, but adding ``max_usb_current=1`` to config.txt (on the partition of sdcard that bootcode.bin lives on) may enable it. We may want to consider whether this should be enabled by default, or be more configurable (i.e. specify the gpio pin to assert).

I'm afraid I don't have a setup to easily test this, so apologies if it doesn't work.

DougieLawson, 2017-04-21:
Hi @popcornmix 
I tried your downloadable bootcode.bin with and without a config.txt and couldn't get that one to boot.

popcornmix, 2017-04-21:
Okay, I'll need to set up a USB boot Pi and get back to you.

DougieLawson, 2017-04-21:
The cheapo/crappo USB2.0 HDD carrier I've got on my B+ has now been discontinued by eBuyer. So if you need any more tests done, just shout.

I've got three 3Bs that are all booting from USB devices (two with HDDs, one with a USB stick). I've also got a zero with a WD HDD (and a super power supply) that I can experiment with.

ghollingworth, 2017-04-22:
@popcornmix I think you'll have to change the code to set the 1.2A usb current by default otherwise by adding the config.txt the bootcode.bin will assume that there is a full build available on the SD card and therefore won't try booting through USB.

Gordon

popcornmix, 2017-04-22:
Okay, this one sets the high current unconditially:
https://drive.google.com/uc?id=0B-6zmEDJwxZEaEh4UklqSHE3aW8&export=download

So remove config.txt from sdcard and try this.

DougieLawson, 2017-04-22:
That one sits with the green LED lit for a while, then it starts doing four flashes.

My normal config.txt and cmdline are:
```
#
# Unified config.txt @ 14/04/2017 14:50
#

[0xe685c936]
# B+ ** Apollo     : e685c936 **
dtoverlay=i2c-rtc,ds1307
gpu_mem=16
initramfs initrd.sda followkernel

# ... lots of lines for other Raspberries removed for clarity ...

[all]
dtparam=audio=on
dtparam=spi=on
dtparam=i2c_arm=on
``` 
```
dwc_otg.lpm_enable=0 console=tty1 root=UUID=2ebfb7f2-0c0b-4de2-964c-8fe9b20aacee rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait rootdelay=10 quiet splash plymouth.ignore-serial-consoles
```

DougieLawson, 2017-04-22:
PS. I tried https://drive.google.com/uc?id=0B-6zmEDJwxZEaEh4UklqSHE3aW8&export=download on my WD Pi Zero with 314GB HDD and that booted enough to get to an emergency console (because the cmdline.txt was set-up for root on a HDD). 

So the not booting problem only exists with my B+ and HDD carrier.

ghollingworth, 2017-04-22:
Does it spin up the hard drive then?

The config.txt and cmdline.txt on the HDD will have no effect on this...  the important thing is that you have the bootcode.bin that popcornmix provided on the SD card on it's own.  You can also add a file TIMEOUT (can be empty, as long as it exists on the SD card) to increased the delay to wait for the hard disk to enumerate.

Gordon

DougieLawson, 2017-04-22:
As far as I could tell the drive is spinning up. It normally runs with rootdelay=10 so I'll try a blank TIMEOUT file.

DougieLawson, 2017-04-22:
The drive doesn't seem to spin up until the green LED starts winking (four winks). That's with a null TIMEOUT file in the root of the SDCard.

While I had your special bootcode.bin I've tested with a bootable SanDisk USB stick and that B+ came up OK with that.

DougieLawson, 2017-04-22:
Somewhat interestingly after I rebooted normally I got this email from S.M.A.R.T monitoring
```
Date: Sat, 22 Apr 2017 18:23:12 +0000
From: root <root@localhost>
To: root@localhost
Subject: SMART error (ErrorCount) detected on host: apollo

This message was generated by the smartd daemon running on:

   host name:  apollo
   DNS domain: [Empty]

The following warning/error was logged by the smartd daemon:

Device: /dev/sda [USB JMicron], ATA error count increased from 0 to 17

Device info:
Hitachi HCC543216A7A380, S/N:E200424039M7UJ, WWN:5-000cca-691eea61f,
FW:ES1OA60W, 160 GB

For details see host's SYSLOG.

You can also use the smartctl utility for further investigation.
Another message will be sent in 24 hours if the problem persists.
```

popcornmix, 2017-04-24:
I wonder if enabling the max usb current earlier will help?
https://drive.google.com/uc?id=0B-6zmEDJwxZEWnkzLVBjN3BLNjA&export=download

DougieLawson, 2017-04-24:
Hi @popcornmix 
Today's bootcode.bin with a timeout file and we're up and running.

Thank you.

popcornmix, 2017-04-26:
I've tweaked the bootcode.bin to only enable the max usb current pin when sdcard booting fails and it switches to USB (so should be a little safer for users who don't use usb boot).
That is now on rpi-update - can you confirm that still works okay for you and close if issue is resolved.

DougieLawson, 2017-04-27:
The version you've just pushed with rpi-update doesn't work.

DougieLawson, 2017-04-29:
I've tried this again with https://github.com/Hexxeh/rpi-firmware/commit/a86bfee5b47a74c13056997f1e4d8b9d8090b398 and it's still a NO GO.

JamesH65, 2017-12-21:
@DougieLawson Still not working?

DougieLawson, 2017-12-22:
@JamesH65 I've changed the HDD carrier on my B+ now. I've not tried a "just the bootcode.bin" boot on that machine recently.

I'll see what I can do when I'm bored over Xmas.

JamesH65, 2018-06-29:
@DougieLawson Did you get bored over Xmas?

DougieLawson, 2018-06-30:
I've got a spare B+ sitting on a bench right now. 

I'll see if I can find the USB carrier and give that one a go with a hard disk.