raspberrypi/firmware #664: Request: Way to fix the gpu clocks

Terminus-IMRC, 2016-09-20:
I'm using RPi3B and `gpu_freq=400` in config.txt.

However, when I'm not using GPU, the clocks of v3d and isp are almost always 250MHz.
When I runs GPU programs such as omxplayer and QPU programs, the clocks of v3d and isp are almost always 400MHz.

I'm finding the way to fix the clocks to 400MHz because in low clocks QPU performs lower VPM DMA throughput. My experiment showed that when QPU is waiting for DMA load/store (reading `vpm_ld_wait` or `vpm_st_wait`) for several clocks the GPU clock would get lower though it's in DMA operation. QPU won't keep 400MHz even in DMA operation!

Is it possible to fix the GPU clocks or could you implement it?

popcornmix, 2016-09-20:
``force_turbo=1`` in config.txt will always run the arm and gpu clocks at the higher rate.
Be aware that using this in conjunction with ``over_voltage`` may set the warranty bit.

Terminus-IMRC, 2016-09-21:
I should have read the document more carefully...

closed 2016-09-21 by Terminus-IMRC