raspberrypi/firmware #665: hdmi_force_hotplug=1 is not functional

HuseyinBashan, 2016-09-21:
I  tried to use hdmi_force_hotplug=1 in config.txt in order to boot Rpi3 on Android without.
the flag is not functional. I also used hdmi safe, hdmi_edid_file=1 with dumped edid.dat l put BOOT  partition.
- kernel 4.4
- latest firmware

popcornmix, 2016-09-21:
By Android I assume you are using the experimental graphics driver? (``dtoverlay=vc4-kms-v3d``)

HuseyinBashan, 2016-09-22:
yes

popcornmix, 2016-09-22:
That is not expected to work.
config.txt display settings will only apply when using the (default) firmware display driver.
``dtoverlay=vc4-kms-v3d`` disables the firmware display driver and the arm controls the display (with an arm side "kms" driver).
There may be ways of configuring a similar behaviour, but not through the firmware.
You may have more luck asking here: https://github.com/anholt/linux/issues/

closed 2016-09-22 by popcornmix