raspberrypi/firmware #667: force_turbo sets incorrect vcore

Fourdee, 2016-10-01:
As per: https://www.raspberrypi.org/forums/viewtopic.php?p=1045575#p1045575

Setting ```force_turbo=1``` results in a lower/incorrect vcore, when measured with ```vcgencmd measure_volts```

Confirmed with RPi Zero (possibly related to all RPi's: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=160939&hilit=vcgencmd)

Test with RPi Zero:
- Zero 1.3
- 4.4.21+
- Performance CPU gov.
- Get core voltage with: vcgencmd measure_volts
- Default core voltage =1.3500V
```
force_turbo=1
volt=1.2000V
```

popcornmix, 2016-10-23:
I set ``force_turbo=1`` in config.txt and I see:
```
pi@raspberrypi:~ $ vcgencmd measure_volts
volt=1.3500V
```
Report output of
```
vcgencmd version
vcgencmd get_config int
vcgencmd measure_clock arm
vcgencmd measure_clock core
vcgencmd measure_volts
```

Fourdee, 2016-10-23:
@popcornmix 

RPi Zero.

```config.txt```, enabled entries only:
```
disable_overscan=1
framebuffer_width=1280
framebuffer_height=720
start_x=0
disable_camera_led=0
gpu_mem_256=16
gpu_mem_512=16
gpu_mem_1024=16
max_usb_current=1
disable_pvt=1
disable_splash=1
avoid_pwm_pll=1
dtparam=audio=off
dtparam=i2c_arm=off
dtparam=i2c1=off
i2c_arm_baudrate=100000
dtparam=spi=off
temp_limit=65
initial_turbo=20
force_turbo=1
```

```
root@DietPi:~# vcgencmd version
Sep 21 2016 13:14:13
Copyright (c) 2012 Broadcom
version 2eaf52cc53435b5ce67253af1487f9a4f9f96e2d (clean) (release)
root@DietPi:~# vcgencmd get_config int
arm_freq=1000
audio_pwm_mode=1
config_hdmi_boost=5
core_freq=400
disable_auto_turbo=1
disable_commandline_tags=2
disable_splash=1
enable_uart=1
force_eeprom_read=1
force_pwm_open=1
force_turbo=1
framebuffer_depth=16
framebuffer_height=720
framebuffer_ignore_alpha=1
framebuffer_swap=1
framebuffer_width=1280
gpu_freq=300
hdmi_force_cec_address=65535
ignore_lcd=1
init_uart_clock=0x2dc6c00
initial_turbo=20
mask_gpu_interrupt1=0x10000
max_usb_current=1
over_voltage_avs=0x249f0
pause_burst_frames=1
program_serial_random=1
sdram_freq=450
temp_limit=65
root@DietPi:~# vcgencmd measure_clock arm
frequency(45)=700000000
root@DietPi:~# vcgencmd measure_clock core
frequency(1)=250000000
root@DietPi:~# vcgencmd measure_volts
volt=1.2000V
```

CPUfreq reports 1ghz:
```
root@DietPi:~# cat /sys/devices/system/cpu/cpufreq/policy0/scaling_cur_freq
1000000
root@DietPi:~# cat /sys/devices/system/cpu/cpufreq/policy0/scaling_governor
performance
```

popcornmix, 2016-10-23:
What about:
```
vcgencmd measure_temp
vcgencmd get_throttled
```

Fourdee, 2016-10-23:
@popcornmix 

```
root@DietPi:~# vcgencmd measure_temp
temp=34.7'C
root@DietPi:~# vcgencmd get_throttled
throttled=0x0
```

popcornmix, 2016-10-23:
Try removing ``initial_turbo=20``

Fourdee, 2016-10-23:
@popcornmix 

> Try removing initial_turbo=20

Yep, that did it 👍 
```
root@DietPi:~# vcgencmd measure_clock arm
frequency(45)=1000104000
```

popcornmix, 2016-10-23:
Okay, that is a bug.
force_turbo and initial_turbo don't make sense together, and the end of the initial_turbo time is disabling turbo mode. Should be easy to fix.

popcornmix, 2016-10-23:
Test firmware you can try: https://www.dropbox.com/s/oo5f4ztfozgeg06/firmware_ip.zip?dl=0

Ruffio, 2016-12-30:
@Fourdee has this issue been resolved with the above firmware?

Fourdee, 2016-12-30:
@Ruffio 
Unsure, will test and confirm results tomorrow.

Fourdee, 2016-12-31:
@Ruffio 
Yep, resolved with latest firmware/kernel ```4.4.38+``` from Raspbian repo 👍 :

```
root@DietPi:~# vcgencmd measure_volts
volt=1.3500V
root@DietPi:~# cat /DietPi/config.txt | grep force_turbo
force_turbo=1
root@DietPi:~# uname -a
Linux DietPi 4.4.38+ #938 Thu Dec 15 15:17:54 GMT 2016 armv6l GNU/Linux
```

Thanks @popcornmix. @Ruffio this is now resolved and can be closed.

closed 2016-12-31 by popcornmix