raspberrypi/firmware #668: over_voltage has no effect on RPi Zero

Fourdee, 2016-10-01:
As per: https://www.raspberrypi.org/forums/viewtopic.php?p=1045575#p1045575

```over_voltage``` value has no effect on vcore when measured with ```vcgencmd measure_volts```

Notes:
- Zero 1.3
- 4.4.21+
- Performance CPU gov.
- Get core voltage with: ```vcgencmd measure_volts```
- Default core voltage =1.3500V

over_voltage=6
volt=1.3500V

current_limit_override=1
over_voltage=8 | warranty bit now set
volt=1.3500V

popcornmix, 2016-10-23:
``current_limit_override`` has been a default setting for more than 3 years and the config.txt option no longer exists.
Try setting ``force_turbo=1`` or ``temp_limit=86`` to force warranty bit and allow an ``over_voltage > 6``

Fourdee, 2016-10-23:
@popcornmix 

Caused by same issue as: https://github.com/raspberrypi/firmware/issues/667#issuecomment-255597934

Setting ```initial_turbo=x``` prevents ```over_voltage=x``` being applied.

Disabling ```initial_turbo=x``` resolves issue:
```
root@DietPi:~# vcgencmd measure_volts
volt=1.4000V
root@DietPi:~# cat /DietPi/config.txt | grep over_volt
over_voltage=2
root@DietPi:~# cat /DietPi/config.txt | grep initial_tur
#initial_turbo=20
```

JamesH65, 2018-06-29:
Closing this issue as questions answered/issue resolved.

closed 2018-06-29 by JamesH65