raspberrypi/firmware #669: Disable QPU mailbox commands when vc4 is loaded

anholt, 2016-10-04:
I'm starting work on making it so that you can run `hello_fft` under vc4.  My current thought is to intercept the mailbox commands in the firmware driver and hand them off to vc4 instead.

However, for safety, similarly to the disabling of tvservice's commands, it would be nice if the firmware could disable qpu_enable and qpu_execute mailbox commands when we've found the v3d (without `status="disabled"`) in the devicetree.  Right now if you run `hello_fft`, it locks up the GPU such a way that it never recovers.

popcornmix, 2016-10-04:
Yes, that should be possible.

popcornmix, 2016-10-11:
This is in latest rpi-update firmware.

Ruffio, 2016-12-30:
@anholt have you tested with the 'lastest' firmware? Can this issue be closed?

Ruffio, 2017-01-14:
@anholt any update on this? Have you tested with a firmware newer than 18 Pct 2016 where it was fixed?

popcornmix, 2017-01-15:
I'll close as request has been implemented.
Can always be re-opened if there are still issues.

closed 2017-01-15 by popcornmix