raspberrypi/firmware #670: network boot only works if TFTP server is in the same subnet

alexheinz, 2016-10-06:
It looks to me like network boot is partially broken in the current implementation of bootcode.bin and there is no known way to make this work with a TFTP server in a different subnet.

The DHCP discover packet sent by the Pi includes the parameter request list (option 55). This list does not contain the paremeter request list item 3 (router). That is why the information about the default gateway is not included in the DHCP offer packet sent by the DHCP server by default. It was not requested, so it is not provided.

Without the information about the default gateway, it is not possible to contact a TFTP server in a different subnet (a.k.a the Internet).

Even if I force the DHCP server to provide information about the default gateway, the Pi does not try to go through the gateway. It assumes that the TFTP server can be reached without going through the gateway although it is not in the same subnet.

Please see the attachment for evidence (tcpdump of the packets). The Pi sends an ARP request for an IP address that is not in the same subnet.

[pi3-netboot.txt](https://github.com/raspberrypi/firmware/files/514210/pi3-netboot.txt)

grigorig, 2016-10-11:
It looks like the limited TCP/IP implementation of the 1st stage bootloader simply does not support this. As that bootloader is in masked ROM, this cannot be changed.

cleverca22, 2016-12-02:
the tftp stuff in the rom also ignores the next-server field in dhcp, it only ever tries to tftp the dhcp server, even though i have a seperate tftp box

ghollingworth, 2016-12-03:
You should use option 66: server-name

Just set this to the IP address of the TFTP server (there are many different options you could give and I am unable to test them all and adding each one adds bytes to the bootrom!)

It will use the last of either option type (so if you supply both 43 and 66 it'll use which ever came last in the DHCP reply)

Gordon

rebell3x6, 2017-05-23:
The options do not work for me.

I have set the option tftp-server-name in ISC DHCP SERVER, unfortunately without success.

In Whireshark I see that the correct information are transmitted to the PI with the DHCP OFFER. (e.g. option 43 and 66)

Was anyone able to boot the pi via pxe server from different subnet?

If not, is this a bug and will it be fixed?

Thank you and have a nice day

ghollingworth, 2017-05-24:
The bootrom doesn't support TFTP servers on a separate subnet.  This has been fixed in the boocode.bin implementation (just put bootcode.bin on its own on an SD card)

closed 2017-05-24 by ghollingworth