raspberrypi/firmware #672: DPMS power saving with VGA666

grigorig, 2016-10-17:
With the HDMI output, it's possible to enable display power saving signalling with the ```hdmi_blanking``` config.txt option. There is no counterpart for DPI displays or the VGA adapter. According to the DPMS standard, it should be enough to disable VSYNC and HSYNC signals to enable the display to go to a sleep mode. Is it possible to implemented this for DPI/VGA?

popcornmix, 2016-10-17:
Can you confirm if disabling those signals works?
You should be able to do this by changing the ALT mode of the GPIO from DPI to ouput, e.g. using ``raspi-gpio``.

grigorig, 2016-10-18:
I can confirm that this works. When I set both VSYNC and HSYNC pins to a regular output and pull them to GND, the monitor gets into standby after a few seconds. If I switch back to alt function 2 ([as documented](https://www.raspberrypi.org/documentation/hardware/raspberrypi/dpi/README.md)), the display wakes up again. So I have a small C program now which does the job, which is neat.

Is it possible to get this integrated into the console and X blanking so that it Just Works like with HDMI? It would be quite nice if the firmware could handle that. Maybe it can also be handled by the Linux kernel instead.

popcornmix, 2016-10-18:
Can you try this:
https://www.dropbox.com/s/s4rcrap077o2gzp/firmware_blank.zip?dl=0

``vcgencmd display_power`` should affect the default display (hdmi or dpi).
If ``hdmi_blanking=1`` is set in config.txt then the DPMS display blanking should also affect the default display.
(Note: I haven't got a DPI display set up here, so I've only tested by stepping over the code).

grigorig, 2016-10-18:
That was quick. And it works perfectly fine! Thank you.

grigorig, 2016-10-18:
I just noticed: aren't VGA sync signals active low? So while this may work for my monitor, maybe it is not correct. Let me recheck and test that tomorrow. I'm not even sure what polarity the sync pulses of DPI are, but I can also check that tomorrow.

popcornmix, 2016-10-19:
Okay, I'm forcing the lines low. Let me know if you thing forcing high would be better.
(I'd have thought low signals to disable monitor would make more sense as it would happen when source is powered off or disconnected. It may just be "absence of clock", so either would work).

grigorig, 2016-10-19:
Forcing to high definitely makes more sense. And while I can't find any definite source that says it explicitly, various source seems to imply that forcing the signals to high is correct too, e.g. http://lateblt.tripod.com/bit74.txt

> On monitors which automatically go into power-saving mode when they receive no signal, connecting pins 13, 14, and 10 should be all that's needed to bring the monitor out of power-saving mode.

Doing this will force the pins to low, as pin 10 is the sync ground. On CRTs that should probably keep the display from going to sleep. On the other hand it means forcing the pins to high should invoke power saving.

TL;DR please change this to force pins to high, this should be  the correct way to do it.

grigorig, 2016-10-20:
The updated firmware forces the signals high as expect and works fine. Thanks again!

closed 2016-10-20 by grigorig