raspberrypi/firmware #674: Unable too boot from USB flash drive

maielsboy, 2016-10-20:
HI. Because high SDHC corruption, I was trying to boot from USB (using PI 3 model B and latest Raspbian Lite). Already modified OTP, so `vcgencmd otp_dump | grep 17:` gives me `17:3020000a`. Official guide says, to boot from USB I need to change root and boot path, copy bootcode.bin and start.elf from next firmware branch. So, the way I am doing it: using SDFormatter, I format flash drive as fat32, using win32diskimager I flash Raspbian Lite ot it. Then I copy-paste bootcode.bin and start.elf, modify cmdline.txt and fstab (modifying mmcblk0 path to /dev/sda1 and /dev/sda2 respectively). That's it. I can't boot this way. I even copied bootcode.bin to an empty SD card, but I am always stuck on splash screen (with USB plugged in). I am unable to do a copy of partition with rsync or dd because of my SD card corruption. Any idea what am I doing wrong? 
P.S. my PSU seems ok, I never saw a voltage shortage icon

popcornmix, 2016-10-20:
ping @ghollingworth

ghollingworth, 2016-10-20:
If you've got to the splash screen then start.elf has been loaded correctly by bootcode.bin

Therefore the USB MSD is being read correctly (so unlikely to be a formatting problem or something).  If start.elf was not the right version then this is what you'd see...

Can you copy the whole of the FAT32 partition to the SD card and then see if you can boot to linux?

Even without the USB drive plugged in then you should get as far as a linux kernel bug (because it won't be able to find the root filesystem).  This will at least tell you that you've got a valid booting start.elf, bootcode.bin and kernel7.img

maielsboy, 2016-10-20:
Today I was able to fix my corruption issue by simply changing the sd card. I had a 32gb class 10, unbranded sd card [(link amazon)](https://www.amazon.com/gp/product/B01CM6TSE8). It was troublesome from beginning. Today I just swapped it with a class 4 4gb SanDisk sdhc, that seems to work great. For now I will stick with that. Probably in future I will use that USB to extend my filesystem. Thanks

closed 2016-10-20 by maielsboy