raspberrypi/firmware #675: RPi3 Boot from USB nogo

johnd2117, 2016-10-25:
I have been trying to setup a USB hard drive to boot my RPi3 to no avail. I'm using the instruction "How to boot from a USB mass storage device on a Raspberry Pi3". I have tried on a WD 500Gb hard rive, a Sandisk 32Gb flash drive and now I'm trying a Seagate 500Gb hard drive. In each case, when I remove the SD card and restart, the hard drive light starts to flash and I get the colour splash followed by four raspberries and the following two lines:
[      4.260940] sd 0:0:0:0: [sda] No caching mode page found.
[      4.261119] sd 0:0:0:0: [sda] Assuming drive cache: write through.
After that the drive stops flashing and the RPi3 green led flashes every second. But nothing else. ???
I assume this must be a problem with either the instructions or my RPi3 as I can't believe that three drives can be faulty.
Is there anybody, with more wisdom than mine. who can help with this problem.

Kind regards John.

ghollingworth, 2016-10-25:
Looks like everything is working correctly from the booting standpoint since you are starting Linux.  But if that is the last thing on the screen then the problem is mostly likely due to Linux not being able to find the root filesystem.  On a dmesg log I get here after the line "Assuming drive cache: write through", I get "sda: sda1 sda2" which is what happens when Linux understands the partition layout.

Can you plug the mass storage device into the SD card booted Raspberry Pi and do:

parted /dev/sda print

You should have something like:

    Number  Start   End     Size    Type     File system  Flags
    1       1049kB  99.6MB  98.6MB  primary  fat32        lba
    2       99.6MB  30.8GB  30.7GB  primary  ext4         lba

If so, do:
 
    mkdir /tmp/fs_tmp
    sudo mount /dev/sda2 /tmp/fs_tmp
    ls -l /tmp/fs_tmp

This should look like a standard root file system containing things like etc, dev, proc, tmp, var, home

johnd2117, 2016-10-25:
Thanks for your reply Gordon.

I followed your instructions and got the feedback that you expected, except that ext4 was not lba. When I initially created this partition, and 'printed' the results ext4 was lba (??)

 It still doesn't boot though:) I am new to Linux and I'm afraid that it is all very vague to me at the moment.

Kind regards John

robtech58, 2016-10-26:
I had a similar problem after following the instructions in Magpi issue 50. Boot would start but the root filesystem was not found because for some reason (no doubt past errors on my part) the SD card that I was copying from had a different set of partitions to those assumed in the article (I had /dev/mmcblk0p6 and /dev/mmcblk0p7 for the boot and root filesystems), and so the command line edits didn't work correctly.  To fix the problem I had to:

1. Boot the Pi from the SD Card
2. Mount the USB disk
```
sudo mkdir /mnt/target
sudo mount /dev/sda2 /mnt/target/
sudo mount /dev/sda1 /mnt/target/boot/
```
3. Edit the /boot/cmdline.txt on the USB disk to point root at /dev/sda2
```
sudo nano /mnt/target/boot/cmdline.txt
```
Make sure it has `root=/dev/sda2`
4. Edit /etc/fstab to point it at the both the boot partition and the root filesystem
```
sudo nano /mnt/target/etc/fstab
```
Make sure it contains 2 lines that look something like:
```
/dev/sda1       /boot      vfat          defaults                     0       2
/dev/sda2      /              ext4         defaults,noatime       0       2
```
I don't know how significant spacing is in that file so I just edited what was there.

Then you can power down, remove the SD card and see if it works. It did for me. But like you I am very new to Linux so apologies if it doesn't help but I hope it does. (I just upgraded everything to try Pixel and broke the USB boot in the process - so plenty more to learn!)

ghollingworth, 2016-10-26:
Ah that would be because you used a NOOBS SD card rather than a Raspbian image.  The docs should be a bit more explicit about that!

ghollingworth, 2016-10-26:
OK, can you do:

    mkdir /tmp/fs_tmp
    sudo mount /dev/sda1 /tmp/fs_tmp
    cat /tmp/fs_tmp/cmdline.txt

johnd2117, 2016-10-27:
I would like to thank both of you gentlemen for your help in this matter. I Followed what robtech58 suggested and it all works properly now. Now I can get on with learning about Linux. I used to dabble in MSDOS years ago and I'm sure Linux can't be any harder, can it?

Once again, thank you very much. Kind regards John

closed 2016-10-28 by johnd2117

krokofant, 2017-09-08:
@robtech58 Many thanks! Couldn't get passed this while trying to install RetroPie on an USB.

I'm suprised this issue isn't more wide spread.

robtech58, 2017-09-08:
Thanks - glad it helped. I'm guessing not many people doing USB boots start with a NOOBS installation.

airatru, 2018-04-05:
Thanks to robtech58, I got it working too =)

davidkusch, 2018-07-05:
Thank you **robtech58**, worked for me as well! ;)

crus1313, 2018-11-18:
Hi

I followed the guid for 4 times, on an external ssd and an usb stick... cannot boot on them. Any clues?
I folloed  @robtech58 guide, tripple checked my files are the same..nothing, my Pi won't boot from external devices.
Got no errors during setup, don't know where to look for now.

crus1313, 2018-11-18:
Well, I powered my Pi second time and it booted of usb stick. Odd, because I did nothing... just unplug and replug the power. 
Now I have to duplicate the result on an usb ssd drive.
Thank you.

goldindeejay, 2019-04-22:
If it still doesn‘t boot up after robtechs instruction try this:
1. Change the boot instructions in cmdline.txt on your Micro-SD-card 

sudo nano /boot/cmdline.txt

2. Change root=.........blahbla
to root=/dev/sda2

3. Reboot and see the good results ;)
P.S.:After Rebooting my mouse and keyboard stopped working for no reason.
 -_-Working on a solution to change this mistake.

goldindeejay, 2019-04-22:
Update: it still needs the sd-card.