raspberrypi/firmware #676: usb boot - tvheadend continuity errors

fishxz, 2016-10-28:
hey,
popcornmix told me to create an issue here, so i will do :)

since i switched over to usb boot on libreelec, my tvheadend produce alot of continuity errors. it feels like my usb tv box gets not enough power for run flawless.

```
LibreELEC-wz:~ # vcgencmd get_throttled
throttled=0x0
```

power supply: original power supply for the rpi3 5V/2,5A

popcornmix, 2016-10-28:
Ping @ghollingworth

ghollingworth, 2016-11-04:
Is this just the fact that the USB bandwidth is now being shared between the mass storage and the video data stream?

Otherwise I'll wait for another reproduction of the issue...

Gordon

popcornmix, 2016-11-04:
@fishxz Can you confirm if booting from the sdcard, but using USB as the storage partition has continuity errors?
That will confirm if it is a USB contention issue, or something specific to the USB (without sdcard) boot.

fishxz, 2016-11-04:
i can confirm this. i even compared same version of libreelec on usb and sd, and i only had this problems on usb.

popcornmix, 2016-11-04:
So you are booting from sdcard, but using USB for storage?
You may be better off not using the USB for storage. In general a good sdcard will be faster and won't have contention with USB TV dongle and ethernet.

fishxz, 2016-11-04:
i dont. im talking about a full usb (with your builds and enabled branch=next firmware). thats where this problem occurs.

with a normal sdcard boot (storage on sdcard aswell) everything works fine.

popcornmix, 2016-11-04:
There are 3 scenarios that are of interest. Knowing which have problems and which don't will identify if the problem is just USB contention (which we can't fix) or USB setup from bootcode.bin (which we can fix).
1. boot and storage on USB. No sdcard connected. OTP bit set and using BRANCH=next firmware
2. boot and storage on sdcard. Default LibreELEC setup.
3. boot from sdcard, storage on USB. sdcard is plugged in with valid boot files. cmdline.txt includes ``disk=<USB device>``

Can you confirm the results for these 3 cases - I believe 1 has continuity errors, and 2 doesn't, but we are interested in 3.

fishxz, 2016-11-04:
1. continuity errors
2. works fine
3. works fine aswell

ghollingworth, 2016-11-04:
OK, one thing that would be interesting:

I'm trying to understand if the problem is due to the BRANCH=next firmware or whether it is because of the USB booting.  So I'd like you to try, taking the 3. case and copying the firmware across from USB onto the SD sdcard...

So in this case you are using the BRANCH=next firmware to boot from sdcard but then use the USB as the storage....

Gordon

fishxz, 2016-11-04:
@ghollingworth works.

Ruffio, 2016-12-30:
@fishxz if this issue has been resolved, then please close this issue...

Ruffio, 2017-01-09:
@fishxz has this issue been fixed?

fishxz, 2017-01-12:
It seems to be fixed. I'm currently back to m my SD card, but last time I test, the issue was fixed, so I gonna close this

closed 2017-01-12 by fishxz