raspberrypi/firmware #677: Add parameter to config.txt to request ARM core to be started at EL3

gonzoua, 2016-10-30:
Currently VideoCore starts kernel file at EL2 unless kernel_old=1 specified in config.txt. But kernel_old also changes logic for kernel and DTB load procedure. What would be nice is to have special knob, probably a bit in arm_control or dedicated parameter to just change default ELx level but keep all the normal boot logic intact. 

This comes useful when first ARM binary loaded is or emulates Secure Monitor to provide PSCI API to the next boot level.

pelwell, 2016-10-31:
You already have the facility to do this by modifying the [ARMv8 stub file](https://github.com/raspberrypi/tools/blob/master/armstubs/armstub8.S) and saving the assembled output in the boot partition as `arm8stub.bin`. The [Makefile](https://github.com/raspberrypi/tools/blob/master/armstubs/Makefile) in the same directory takes care of the assembler options and the conversion to a `.bin`.

See [this thread](https://github.com/raspberrypi/firmware/issues/579) for the discussion that led to the stubs and how they work. @swarren, who started the discussion and provided much of the stub code, had U-Boot in mind during the development, and even mentioned [using a custom stub to remain in EL3](https://github.com/raspberrypi/firmware/issues/579#issuecomment-206659570).

Note that you could, if you want, make U-Boot the "stub" rather than the "kernel" by  embedding the magic fields at 0xf0-0xfc within the U-Boot image; setting the kernel address field (0xfc) to 0xffffffff will prevent the firmware from trying to load the kernel image.

swarren, 2016-10-31:
U-Boot should not be a PSCI provider on any AArch64 system (this is actually true for ARMv7 too but for some bizarre reason people dumped PSCI into U-Boot on some platforms rather than using a separate secure monitor). ARM Trusted Firmware is the standard EL3 firmware.

gonzoua, 2016-10-31:
@pelwell Thanks. It does solve my problem. Thanks for pointing me in the right direction. 

@swarren I thought that it was going to be easier to use U-Boot for this but probably create ATF port is not that much of a work. I'll give it a try. Thanks

closed 2016-10-31 by gonzoua