raspberrypi/firmware #680: mmal error when executing python through bash shell in C#

crener, 2016-11-06:
When starting picamera (the python library) in a python script through a bash shell which is being executed by mono C# these errors appear.

```
mmal: mmal_vc_port_enable: failed to enable port vc.null_sink:in:0(OPQV): ENOSPC
mmal: mmal_port_enable: failed to enable connected port (vc.null_sink:in:0(OPQV))0x2170ee0 (ENOSPC)
mmal: mmal_connection_enable: output port couldn't be enabled
Traceback (most recent call last):
  File "camera.py", line 3, in <module>
    camera = picamera.PiCamera()
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 417, in __init__
    self._init_preview()
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 496, in _init_preview
    self, self._camera.outputs[self.CAMERA_PREVIEW_PORT])
  File "/usr/lib/python2.7/dist-packages/picamera/renderers.py", line 512, in __init__
    self.renderer.connect(source)
  File "/usr/lib/python2.7/dist-packages/picamera/mmalobj.py", line 1467, in connect
    self._connection = MMALConnection(source, self.inputs[0])
  File "/usr/lib/python2.7/dist-packages/picamera/mmalobj.py", line 1280, in __init__
    prefix="Failed to enable connection")
  File "/usr/lib/python2.7/dist-packages/picamera/exc.py", line 157, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to enable connection: Out of resources (other than memory)
```

Not really sure why this happens....

The process has the input redirected which seems to cause the issue, If you change `RedirectStandardInput` to false it doesn't throw the error. Why would that break mmal?

Python code
```
import picamera
camera = picamera.PiCamera()
camera.capture('image.jpg')
```

C# code:

```
        static void Main(string[] args)
        {
            ProcessStartInfo start = new ProcessStartInfo()
            {
                FileName = "/bin/bash",
                Arguments = "-c 'python camera.py'",
                UseShellExecute = false,
                RedirectStandardInput = true
            };
            Process pyThread = new Process { StartInfo = start };
            pyThread.Start();
            StreamWriter input = pyThread.StandardInput;
            
            input.WriteLine(Console.ReadLine());
        }
```
The C# is compiled with VS but executed with mono if that makes any difference (though i don't see why it would).

Ruffio, 2016-12-30:
@crener what is the status on this issue? Can you make a simple testcase without Mono ?

crener, 2017-01-09:
As far as i know there is no other way to run C# code on linux without mono...

Ruffio, 2017-01-09:
@crener I don't have PiCamera myself, but I do use C# alot to start programs and scripts, but I don't use StandardInput... I think your issue is in the last couple of lines. Try without 'Line', fx. input.Write or Console.Read

The first line of the error message looks to me like a null pointer exception (mmal: mmal_vc_port_enable: failed to enable port vc.null_sink:in:0(OPQV): ENOSPC), but I know nothing about C++.

Ruffio, 2017-01-22:
@crener have you figured it out?

crener, 2017-01-23:
I haven't managed to try it the project that that is involved in was a little broken so I had to spend the weekend fixing it so that I could get to that point in the code. I don't know if that will make a difference, but i will definitely try.

I'm going to try and rule out mono by using MS .NET core if it works with the code base.

I'll get back to you with results once I try it out. 
Sorry its taking so long :(

crener, 2017-01-29:
So I tried it and it did work by replacing the `WriteLine `with `Write`. So maybe that was the problem? I tried it with `WriteLine` just be be sure that i was still seeing the error and nothing happened...

I did update the Pi a few hours before i tried, maybe that has something to do with it. When i did try it failed but i got no exception and no image which is a bit strange. I wonder why that originally happened?

JamesH65, 2017-12-21:
@crener OK to close?

crener, 2017-12-23:
yeah, sure :)

closed 2017-12-23 by crener