raspberrypi/firmware #681: RPI3 freezes

lugaru90, 2016-11-07:
Hi,

we've got a digital signage system, using midori and omxplayer. on random intervals our system freezes. It seems, that the IO hangs. many kworkers are in D state, with WCHAN state "rpi_firmware_transaction" and "db_timer".
Freezing means: No new SSH connection possible, old ones work, but also freezing at different commands. For example: reading the GPU temperature -> freeze

dmesg output:
`
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.26-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #915 SMP Thu Oct 20 17:08:44 BST 2016
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 8 MiB at 0x36800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 225280
[    0.000000] free_area_init_node: node 0, pgdat 808c2f80, node_mem_map b6036000
[    0.000000]   Normal zone: 1980 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 225280 pages, LIFO batch:31
[    0.000000] [bcm2709_smp_init_cpus] enter (9520->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 13 pages/cpu @b5ff2000 s22592 r8192 d22464 u53248
[    0.000000] pcpu-alloc: s22592 r8192 d22464 u53248 alloc=13*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 223300
[    0.000000] Kernel command line: 8250.nr_uarts=0 dma.dmachans=0x7f35 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2709.boardrev=0xa02082 bcm2709.serial=0xff54d37f smsc95xx.macaddr=B8:27:EB:54:D3:7F bcm2708_fb.fbswap=1 bcm2709.uart_clock=48000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 874112K/901120K available (6348K kernel code, 432K rwdata, 1716K rodata, 476K init, 764K bss, 18816K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xb7800000 - 0xff800000   (1152 MB)
    lowmem  : 0x80000000 - 0xb7000000   ( 880 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x807e8584   (8066 kB)
      .init : 0x807e9000 - 0x80860000   ( 476 kB)
      .data : 0x80860000 - 0x808cc290   ( 433 kB)
       .bss : 0x808cf000 - 0x8098e1ec   ( 765 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000008] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000025] Switching to timer-based delay loop, resolution 52ns
[    0.000290] Console: colour dummy device 80x30
[    0.000489] console [tty1] enabled
[    0.000512] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000532] pid_max: default: 32768 minimum: 301
[    0.000851] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000866] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001795] Disabling cpuset control group subsystem
[    0.001828] Initializing cgroup subsys io
[    0.001859] Initializing cgroup subsys memory
[    0.001905] Initializing cgroup subsys devices
[    0.001927] Initializing cgroup subsys freezer
[    0.001947] Initializing cgroup subsys net_cls
[    0.001996] CPU: Testing write buffer coherency: ok
[    0.002061] ftrace: allocating 21225 entries in 63 pages
[    0.050761] CPU0: update cpu_capacity 1024
[    0.050794] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.050807] [bcm2709_smp_prepare_cpus] enter
[    0.050947] Setting up static identity map for 0x8240 - 0x8274
[    0.052588] [bcm2709_boot_secondary] cpu:1 started (0) 18
[... 649 lines cut ...]
4 S root     18030 18028  0  80   0 -   852 poll_s 21:35 ?        00:00:00 xset dpms 0 0 0
1 S root     18338     2  0  80   0 -     0 worker 21:37 ?        00:00:00 [kworker/1:3]
0 S root     19132     1  0  80   0 -  1153 wait   21:40 ?        00:00:00 /bin/bash /home/pi/Ipse/ipse_tv_scripts/start_browser.sh
4 S root     19134 19132  0  80   0 -   852 poll_s 21:40 ?        00:00:00 xset dpms 0 0 0
1 S zabbix   19331     1  0  80   0 -  3243 wait   21:41 ?        00:00:00 /usr/sbin/zabbix_agentd
1 S zabbix   19332 19331  0  80   0 -  3243 hrtime 21:41 ?        00:00:00 /usr/sbin/zabbix_agentd: collector [idle 1 sec]
1 S zabbix   19333 19331  1  80   0 -  3275 hrtime 21:41 ?        00:00:14 /usr/sbin/zabbix_agentd: active checks #1 [idle 1 sec]
1 S root     19349     2  0  80   0 -     0 worker 21:41 ?        00:00:00 [kworker/3:3]
1 S root     19366     2  0  80   0 -     0 worker 21:41 ?        00:00:00 [kworker/1:4]
0 S root     20196     1  0  80   0 -  1153 wait   21:45 ?        00:00:00 /bin/bash /home/pi/Ipse/ipse_tv_scripts/start_browser.sh
4 S root     20198 20196  0  80   0 -   852 poll_s 21:45 ?        00:00:00 xset dpms 0 0 0
1 S root     20411     2  0  80   0 -     0 worker 19:40 ?        00:00:01 [kworker/u8:2]
0 S root     21237     1  0  80   0 -  1153 wait   21:50 ?        00:00:00 /bin/bash /home/pi/Ipse/ipse_tv_scripts/start_browser.sh
4 S root     21239 21237  0  80   0 -   852 poll_s 21:50 ?        00:00:00 xset dpms 0 0 0
0 S root     22322     1  0  80   0 -  1153 wait   21:55 ?        00:00:00 /bin/bash /home/pi/Ipse/ipse_tv_scripts/start_browser.sh
4 S root     22324 22322  0  80   0 -   852 poll_s 21:55 ?        00:00:00 xset dpms 0 0 0
0 S root     23373   450  0  80   0 -   850 hrtime 22:00 ?        00:00:00 sleep 60
0 S root     23374     1  0  80   0 -  1153 wait   22:00 ?        00:00:00 /bin/bash /home/pi/Ipse/ipse_tv_scripts/start_browser.sh
4 S root     23376 23374  0  80   0 -   852 poll_s 22:00 ?        00:00:00 xset dpms 0 0 0
0 S root     23442  2168  0  80   0 -   850 hrtime 22:00 ?        00:00:00 sleep 30
1 S root     27680     2  0  80   0 -     0 worker 20:22 ?        00:00:00 [kworker/2:2]
4 T root     14902 12873  0  80   0 -  1695 signal 21:21 pts/0    00:00:00 sudo strace -cp 979
4 T root     14906 14902  0  80   0 -   531 signal 21:21 pts/0    00:00:00 strace -cp 979
0 Z pi        1411  1316  0  80   0 -     0 exit   18:44 ?        00:00:00 [unclutter] <defunct>
`

popcornmix, 2016-11-08:
Possibly the firmware is crashing.  Can you report output of:
```
vcgencmd version
vcgencmd get_config int
vcgencmd get_config str
```
(from before a crash).
Also we can get assert logging from the firmware if you add to config.txt ``start_debug=1`` (and reboot).
Then after you get get a crash run:
```
sudo vcdbg log msg
sudo vcdbg log assert
sudo vcdbg log reloc
sudo vcdbg log malloc
```
Also can you describe any hardware (e.g. devices connected by USB or GPIO connector) attached?
After you have been running for some time (but before the crash) report output of:
```vcgencmd get_throttled```

lugaru90, 2016-11-08:
So for now the first Information (will update later)
```
root@Ipse:~# vcgencmd version
Oct 25 2016 16:02:48
Copyright (c) 2012 Broadcom
version f74adfcae00b721627eab44590728c13860bcbc2 (clean) (release)
```
```
root@Ipse:~# vcgencmd get_config int
arm_freq=1000
audio_pwm_mode=1
avoid_warnings=1
config_hdmi_boost=5
core_freq=500
desired_osc_freq=0x36ee80
disable_commandline_tags=2
disable_l2cache=1
force_eeprom_read=1
force_pwm_open=1
framebuffer_ignore_alpha=1
framebuffer_swap=1
gpu_freq=300
hdmi_force_cec_address=65535
hdmi_force_hotplug=1
hdmi_group=1
hdmi_mode=16
init_uart_clock=0x2dc6c00
lcd_framerate=60
over_voltage=2
over_voltage_avs=0x200b2
over_voltage_sdram_p=2
pause_burst_frames=1
program_serial_random=1
sdram_freq=500
temp_limit=85
```
**Also tried non overclocked:**
```
pi@Ipse:~ $ vcgencmd get_config int
arm_freq=1200
audio_pwm_mode=1
avoid_warnings=1
config_hdmi_boost=5
core_freq=400
desired_osc_freq=0x36ee80
disable_commandline_tags=2
disable_l2cache=1
disable_splash=1
force_eeprom_read=1
force_pwm_open=1
framebuffer_ignore_alpha=1
framebuffer_swap=1
gpu_freq=300
hdmi_force_cec_address=65535
hdmi_group=1
hdmi_mode=16
init_uart_clock=0x2dc6c00
lcd_framerate=60
over_voltage_avs=0x2932e
pause_burst_frames=1
program_serial_random=1
sdram_freq=450
temp_limit=85
```
```
root@Ipse:~# vcgencmd get_config str
device_tree=-
```

Attached Hardware is normally only a Logilink WL0084B but we've also disconnected it for testing.

Device is rebooting with Debug. Will report back later.

popcornmix, 2016-11-08:
Always disable overclock if you are having unreliability issues.
``avoid_warnings=1`` is suspicious - it suggests you are hiding under-voltage or over-temperature warnings. under-voltage would be an obvious source of crashes.
Output of ``vcgencmd get_throttled`` will identify which warnings have been seen (run this after Pi has been active for a while).

lugaru90, 2016-11-08:
Turned avoid_warnings=1 off, we still have some old system, with small power adapters but our actual devices and testing devices all have a stable Power Supply with 3A.

Already getting some information over vcdbg log msg:
```
056935.632: *** No KHAN handle found for pid 1851
077869.097: *** No KHAN handle found for pid 2131
098885.869: *** No KHAN handle found for pid 2487
119869.799: *** No KHAN handle found for pid 2907
140785.938: *** No KHAN handle found for pid 3213
161819.278: *** No KHAN handle found for pid 3795
```
PID 3795 from ps aux:
```
root      3795  7.8  3.2 144784 30800 ?        Sl   15:40   0:00 /usr/bin/omxplayer.bin --
no-keys --win 716.953125 247.71875 1752 822.71875 /home/pi/Ipse/www/ipse_tv/clips/581b0cdca7bba_581b0cdca7bfb.mp4 &
```

popcornmix, 2016-11-08:
What firmware services are you using?
Obviously openmax for video (from omxplayer).
Are you using OpenGLES or OpenVG yourself?
By default openmax uses OpenVG for overlays/subs - does launching with ``--no-osd`` (which disabled this) make a difference?
What do you have ``gpu_mem`` set to? Does ``gpu_mem=256`` make a difference?

pelwell, 2016-11-08:
A few more questions:
Can you say a bit about how your application uses omxplayer?
Is it allowed to run to completion, or do you kill it mid-stream?

lugaru90, 2016-11-08:
Just set GPU mem to 256 on a few devices, can tell you in 1-3 hours...
We just use omxplayer for playing videos.

Our system works like a webserver with lighttpd and midori (also tested epiphany-browser). We have different content: Images, Videos, and HTML. On some HTML pages we have embedded Videos.
The Videos are shown in Fullscreen by default, if it is embedded in HTML we find out the position over JS and run omxplayer over an ajax request to a php script. On some Videos we have to kill the omxplayer process this happens over killall or pkill. This can be set by the user. We also tested to disable this and let the video run out, but even then the crash happens.

Also Midori has a strange error some times (not only on videos, also on normal html):
```
Tue Nov  8 16:08:29 2016
GLib-CRITICAL **: Source ID 169604 was not found when attempting to remove it
```

popcornmix, 2016-11-08:
```throttled=0x50005```
See https://www.raspberrypi.org/forums/viewtopic.php?t=147781&p=972579#p972790

Your power supply is inadequate (dropping below 4.63V) and we are displaying a under-voltage warning. That is very likely the cause of your crashes.

Use a decent power supply. Ensure get_throttled returns 0x0. Report back if you still have a problem.
https://thepihut.com/products/official-raspberry-pi-universal-power-supply would be a suitable supply.

lugaru90, 2016-11-08:
thx for your support and info. We will try it out :)

lugaru90, 2016-11-11:
So new Logs on crash:
```
pi@Ipse:~ $ sudo vcdbg log msg
001033.346: HDMI:EDID error reading EDID block 0 attempt 0
001034.609: HDMI:EDID error reading EDID block 0 attempt 1
001035.868: HDMI:EDID error reading EDID block 0 attempt 2
001037.126: HDMI:EDID error reading EDID block 0 attempt 3
001038.385: HDMI:EDID error reading EDID block 0 attempt 4
001039.643: HDMI:EDID error reading EDID block 0 attempt 5
001040.902: HDMI:EDID error reading EDID block 0 attempt 6
001042.160: HDMI:EDID error reading EDID block 0 attempt 7
001043.419: HDMI:EDID error reading EDID block 0 attempt 8
001044.677: HDMI:EDID error reading EDID block 0 attempt 9
001045.699: HDMI:EDID giving up on reading EDID block 0
001062.398: HDMI:Setting property pixel encoding to Default
001062.421: HDMI:Setting property pixel clock type to PAL
001062.439: HDMI:Setting property content type flag to No data
001062.458: HDMI:Setting property fuzzy format match to enabled
001251.174: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
001251.200: hdmi: HDMI:>>>>>>>>>>>>>Rx sensed, reading EDID<<<<<<<<<<<<<
001251.599: hdmi: HDMI:EDID error reading EDID block 0 attempt 0
001252.868: hdmi: HDMI:EDID error reading EDID block 0 attempt 1
001254.131: hdmi: HDMI:EDID error reading EDID block 0 attempt 2
001255.395: hdmi: HDMI:EDID error reading EDID block 0 attempt 3
001256.659: hdmi: HDMI:EDID error reading EDID block 0 attempt 4
001257.923: hdmi: HDMI:EDID error reading EDID block 0 attempt 5
001259.188: hdmi: HDMI:EDID error reading EDID block 0 attempt 6
001260.452: hdmi: HDMI:EDID error reading EDID block 0 attempt 7
001261.715: hdmi: HDMI:EDID error reading EDID block 0 attempt 8
001262.979: hdmi: HDMI:EDID error reading EDID block 0 attempt 9
001264.007: hdmi: HDMI:EDID giving up on reading EDID block 0
001264.038: hdmi: HDMI: No lookup table for resolution group 0
001264.055: hdmi: HDMI: hotplug attached with DVI support
001264.116: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
001264.491: hdmi: HDMI:EDID error reading EDID block 0 attempt 0
001265.758: hdmi: HDMI:EDID error reading EDID block 0 attempt 1
001267.021: hdmi: HDMI:EDID error reading EDID block 0 attempt 2
001268.284: hdmi: HDMI:EDID error reading EDID block 0 attempt 3
001269.546: hdmi: HDMI:EDID error reading EDID block 0 attempt 4
001270.809: hdmi: HDMI:EDID error reading EDID block 0 attempt 5
001272.071: hdmi: HDMI:EDID error reading EDID block 0 attempt 6
001273.335: hdmi: HDMI:EDID error reading EDID block 0 attempt 7
001274.598: hdmi: HDMI:EDID error reading EDID block 0 attempt 8
001275.861: hdmi: HDMI:EDID error reading EDID block 0 attempt 9
001276.887: hdmi: HDMI:EDID giving up on reading EDID block 0
001277.375: hdmi: HDMI: hotplug deassert
001277.390: hdmi: HDMI: HDMI is currently off
001277.405: hdmi: HDMI: changing mode to unplugged
001277.427: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
001278.472: *** Restart logging
001279.914: Read command line from file 'cmdline.txt'
dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
001531.068: Loading 'kernel7.img' to 0x8000 size 0x408b88
001534.722: Kernel trailer DTOK property says yes
001534.738: Kernel trailer DDTK property says yes
001538.036: Loading 'bcm2710-rpi-3-b.dtb' to 0x410b88 size 0x3e78
001634.230: dtparam: audio=on
001796.257: Device tree loaded to 0x2efebc00 (size 0x4384)
001798.527: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
003618.751: vchiq_core: vchiq_init_state: slot_zero = 0xee880000, is_master = 1
003622.957: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
052847.806: *** No KHAN handle found for pid 1840
073584.363: *** No KHAN handle found for pid 2112
094273.592: *** No KHAN handle found for pid 2408
115009.584: *** No KHAN handle found for pid 2834
135696.945: *** No KHAN handle found for pid 3101
156401.398: *** No KHAN handle found for pid 3397
177088.321: *** No KHAN handle found for pid 3723
197842.446: *** No KHAN handle found for pid 3990
218579.801: *** No KHAN handle found for pid 4291
[... 581 lines cut ...]
[ 1439.492768] [<8046bf3c>] (bcm2835_cpufreq_clock_property.constprop.1) from [<8046bf9c>] (bcm2835_cpufreq_driver_target_index+0x4c/0xc0)
[ 1439.492780] [<8046bf9c>] (bcm2835_cpufreq_driver_target_index) from [<80466568>] (__cpufreq_driver_target+0x1a4/0x2e8)
[ 1439.492792] [<80466568>] (__cpufreq_driver_target) from [<80469768>] (od_check_cpu+0xcc/0xd0)
[ 1439.492802] [<80469768>] (od_check_cpu) from [<8046b450>] (dbs_check_cpu+0x1b4/0x1f8)
[ 1439.492812] [<8046b450>] (dbs_check_cpu) from [<804699a8>] (od_dbs_timer+0x68/0xf8)
[ 1439.492823] [<804699a8>] (od_dbs_timer) from [<8046b80c>] (dbs_timer+0x1ac/0x1d0)
[ 1439.492843] [<8046b80c>] (dbs_timer) from [<8003c82c>] (process_one_work+0x154/0x458)
[ 1439.492855] [<8003c82c>] (process_one_work) from [<8003cb84>] (worker_thread+0x54/0x500)
[ 1439.492865] [<8003cb84>] (worker_thread) from [<80042860>] (kthread+0xec/0x104)
[ 1439.492876] [<80042860>] (kthread) from [<8000fbc8>] (ret_from_fork+0x14/0x2c)
[ 1439.492908] INFO: task kworker/3:2:167 blocked for more than 120 seconds.
[ 1439.492913]       Not tainted 4.4.30-v7+ #919
[ 1439.492917] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[... cut ...]
[ 1439.492977] [<805b8860>] (__mutex_lock_slowpath) from [<805b8978>] (mutex_lock+0x60/0x64)
[ 1439.492987] [<805b8978>] (mutex_lock) from [<8046b690>] (dbs_timer+0x30/0x1d0)
[ 1439.492997] [<8046b690>] (dbs_timer) from [<8003c82c>] (process_one_work+0x154/0x458)
[ 1439.493007] [<8003c82c>] (process_one_work) from [<8003cb84>] (worker_thread+0x54/0x500)
[ 1439.493017] [<8003cb84>] (worker_thread) from [<80042860>] (kthread+0xec/0x104)
[ 1439.493026] [<80042860>] (kthread) from [<8000fbc8>] (ret_from_fork+0x14/0x2c)
```

popcornmix, 2016-11-11:
Sorry, ``sudo vcdbg log reloc`` and ``sudo vcdbg malloc`` should have been
```
sudo vcdbg reloc
sudo vcdbg malloc
```

lugaru90, 2016-11-14:
Reloc log:
```
vcdbg reloc
 
Relocatable heap version 4 found at 0x2f000000
total space allocated is 236M, with 236M relocatable, 0 legacy and 0 offline
0 legacy blocks of size 2359296
 
free list at 0x3d727ae0
227M free memory in 3 free block(s)
largest free block is 225M bytes
 
0x2f000000: free 225M
[  29] 0x3d0c4fc0: used 168K (refcount 1 lock count 0, size   172032, align   32, data 0x3d0c4fe0, d0rual) 'resize tmp'
[  42] 0x3d0ef000: used 1.1M (refcount 1 lock count 0, size  1152000, align   32, data 0x3d0ef020, d1rual) 'ilin mapping buffer'
[   9] 0x3d208440: used 1.1M (refcount 1 lock count 0, size  1152000, align   32, data 0x3d208460, d1rual) 'ilin mapping buffer'
0x3d321880: free 1.3M
[   8] 0x3d462220: used 1.1M (refcount 1 lock count 0, size  1152000, align   32, data 0x3d462240, d1rual) 'ilin mapping buffer'
[  10] 0x3d57b660: used 1.1M (refcount 1 lock count 0, size  1152000, align   32, data 0x3d57b680, d1rual) 'ilin mapping buffer'
0x3d694aa0: free 588K
[  39] 0x3d727b00: used 496K (refcount 1 lock count 0, size   503968, align 4096, data 0x3d728000, d1ruAL) 'video_decodeRIL:image pool'
[  31] 0x3d7a3bc0: used 496K (refcount 1 lock count 0, size   503968, align 4096, data 0x3d7a4000, d1ruaL) 'video_decodeRIL:image pool'
[  32] 0x3d81fc80: used 496K (refcount 1 lock count 0, size   503968, align 4096, data 0x3d820000, d1ruaL) 'video_decodeRIL:image pool'
[  33] 0x3d89bd40: used 496K (refcount 1 lock count 0, size   503968, align 4096, data 0x3d89c000, d1ruAL) 'video_decodeRIL:image pool'
[  11] 0x3d917e00: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3d917e20, d1rual) 'RIL buffer'
[  12] 0x3d92be40: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3d92be60, d1rual) 'RIL buffer'
[  13] 0x3d93fe80: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3d93fea0, d1rual) 'RIL buffer'
[  14] 0x3d953ec0: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3d953ee0, d1rual) 'RIL buffer'
[  15] 0x3d967f00: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3d967f20, d1rual) 'RIL buffer'
[  16] 0x3d97bf40: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3d97bf60, d1rual) 'RIL buffer'
[  17] 0x3d98ff80: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3d98ffa0, d1rual) 'RIL buffer'
[  18] 0x3d9a3fc0: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3d9a3fe0, d1rual) 'RIL buffer'
[  19] 0x3d9b8000: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3d9b8020, d1rual) 'RIL buffer'
[  20] 0x3d9cc040: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3d9cc060, d1rual) 'RIL buffer'
[  21] 0x3d9e0080: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3d9e00a0, d1rual) 'RIL buffer'
[  22] 0x3d9f40c0: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3d9f40e0, d1rual) 'RIL buffer'
[  23] 0x3da08100: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3da08120, d1rual) 'RIL buffer'
[  24] 0x3da1c140: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3da1c160, d1rual) 'RIL buffer'
[  25] 0x3da30180: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3da301a0, d1ruAl) 'RIL buffer'
[  26] 0x3da441c0: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3da441e0, d1ruAl) 'RIL buffer'
[  27] 0x3da58200: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3da58220, d1rual) 'RIL buffer'
[  28] 0x3da6c240: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3da6c260, d1rual) 'RIL buffer'
[   7] 0x3da80280: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3da802a0, d1rual) 'RIL buffer'
[   6] 0x3da942c0: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3da942e0, d1rual) 'RIL buffer'
[   5] 0x3daa8300: used  576 (refcount 1 lock count 0, size      512, align    4, data 0x3daa8320, d0rual) 'ILCS VC buffer pool'
[   4] 0x3daa8540: used  576 (refcount 1 lock count 0, size      512, align    4, data 0x3daa8560, d0rual) 'ILCS VC buffer pool'
[   3] 0x3daa8780: used 1.3M (refcount 1 lock count 8, size  1382400, align 4096, data 0x3daa9000, d1rual) 'ARM FB'
[   2] 0x3dbfafa0: used  16K (refcount 1 lock count 0, size    16384, align   32, data 0x3dbfafc0, d0ruAl) 'audioplus_tmp_buf'
[   1] 0x3dbfefe0: used 4.0K (refcount 1 lock count 0, size        0, align 4096, data 0x3dbff000, d1rual) 'camera fast alloc arena'
small allocs not requested
```

Malloc
```
Pool 0xbe5caaf8 (3dffe7e8)
Malloc pool size=9.7M (pool=0x3e5caaf8-0x3ef7ff2c)
0x3e5cab20 = malloc(128K) [call from 0x3deb03b0]
0x3e5eab40 = malloc(120)  [call from 0x3de3e0a8]
0x3e5eb000 = malloc(138K) [call from 0x3dca3b7c]
0x3e60e500 = malloc(512)  [v3d_gfxh16_thre]
0x3e60e720 = malloc(7.3K) [dma_transfer_xfers]
0x3e610440 = malloc(7.0K) [Gencmd service state]
0x3e612080 = malloc(180)  [vcos_cmd_entries]
0x3e612160 = malloc(4.0K) [reaper]
0x3e613180 = malloc(8.0K) [call from 0x3df70114]
0x3e6151a0 = malloc(100)  [call from 0x3df70130]
0x3e615240 = malloc(32K)  [call from 0x3dd344b4]
0x3e61d260 = malloc(512)  [call from 0x3dde3a4a]
0x3e61d480 = malloc(272)  [call from 0x3dd34612]
0x3e61d5c0 = malloc(272)  [call from 0x3dd34612]
[... 414 lines cut ...]
0x3e7d5100 = malloc(72)   [RIL buffer header]
0x3e7d5180 = malloc(72)   [RIL buffer header]
[... cut ...]
0x3e7d5700 = malloc(72)   [RIL buffer header]
0x3e7d5780 = malloc(72)   [RIL buffer header]
0x3e7d5800 = malloc(72)   [RIL buffer header]
0x3e7d5880 = malloc(72)   [RIL buffer header]
0x3e7d5900 = malloc(18K)  [call from 0x3df16792]
0x3e7da2b0 = hole(88K)
0x3e7f0320 = malloc(72)   [RIL buffer header]
0x3e7f03a0 = malloc(72)   [RIL buffer header]
0x3e7f0420 = malloc(72)   [RIL buffer header]
0x3e7f04a0 = malloc(72)   [RIL buffer header]
0x3e7f0510 = hole(7.6M)
Malloced:2.1M Remaining:7.6M Failed allocs:0
```

pelwell, 2016-11-14:
Are we sure that the image pitch errors that led to the earlier assertion failures (and which had no other form of error handling) can't result in memory corruption?

6by9, 2016-11-14:
Assuming the compiler hasn't done awkward optimisations, then the vc_image asserts are off an 32bpp RGB[A|X] format image. That's not going to be associated with video playback, so what RGB resource is being used?
If only being use through video_render or equivalent for display then you'll probably get away with it.

6by9, 2016-11-14:
We have a reasonable guess for those vc_image asserts assuming you're using the OpenMax IL resize component with 32bpp RGB on either input or output.
There was a bug which trips that assert on tearing down an IL tunnel. popcornmix had seen them with Kodi when it generates thumbnail JPEGs, and the fix has cleared up those.

Assuming that the assert is coming from the same issue, it's highly unlikely to be the cause of any problems as the port would almost certainly be reconfigured correctly before next use.

lugaru90, 2016-11-23:
Okay it looks like, we've figured out the problem and solved it.

We've added the video as .mp4 video on the HTML page, without autoplay and started omxplayer over it to play. It seems that this makes Problems. Now we just display a Screenshot of the Video inside the Div and then start the OmxPlayer and the Problem is gone :)

Cheers and thanks for your support!

closed 2016-11-23 by lugaru90