raspberrypi/firmware #682: Retropie portable emulation station

DaTrueFlavs, 2016-11-18:
Hi I'm sort of a newbie when it comes to raspberry pi, I got a rpi 3 model B in March 2016 and wanted to turn it into an emulation station I can take on the go. So I bought this touch screen to use with it https://www.amazon.com/Kuman-Display-480x320-Raspberry-Interface/dp/B01CNJVG8K/ref=sr_1_1?ie=UTF8&qid=1470523089&sr=8-1&keywords=raspberry+pi+3+lcd+screen but the problem is I can't find any tutorials on how to use Retropie with this screen as it requires custom images. I'm looking for some help because every tutorial I see uses a different brand than mine, I guess mine is not that common so I'm reaching out here to anybody who happens to know how to complete this project. My pi has been collecting dust as I still can't decide on another thing to do with it. So to anyone who reads this and knows how or knows somebody who can help me complete this project, please tell me asap.

6by9, 2016-11-18:
You'd be better off asking this question in the forums as it  would get more visibility and this isn't a firmware issue, but most of the detail appears to be in the Amazon questions anyway:

Q- where do i get the software? 
A- you already have the software, aside from the waveshare overlay, which you can readily get from github - this LCD is supported under the May 2016 release of Raspbian and other distros - no need for special drivers
A- This screen is basically a Waveshare Spotpear, and uses the same drivers. I found the process and scripting provided here extremely helpful for switching between HDMI and LCD. http://www.waveshare.com/wiki/3.5inch_RPi_LCD_(A)

popcornmix, 2016-11-18:
This github repo is for reporting firmware bugs.
Posting in the [forum](https://www.raspberrypi.org/forums) would be a better place for general advice.

closed 2016-11-18 by popcornmix