raspberrypi/firmware #683: Output 240p (instead of 480i) on composite RCA

starquake, 2016-11-18:
I use my Raspberry PI to emulate old computers/consoles. A common trick with these systems is the output of 240p. Examples are: ZX Spectrum, Nintendo Entertainment System, Commodore 64. It would be great if I could connect my CRT monitor to it. Right now it only outputs 480i which is interlaced and flickers a lot. 240p removes this flicker and also adds scanlines which make it look more authentic.

More info: http://scanlines.hazard-city.de/

I think you would make a lot of people even happier than they already are about classic gaming/computing on the raspberry pi!

popcornmix, 2016-11-18:
I'm afraid this has been asked before and it is not supported.
I'm not certain if the hardware supports it or not, but it's certainly never been tested.

There is, a bit (PROG_SCAN) in VEC CONFIG2 register which potentially is useful, but when I asked about it:

> To change from interlaced to progressive, you can't just change a bit in VEC. You also have to turn it off in the HVS and maybe change some parameters in PixelValve, though I'm not sure what. DispmanX probably configures all this, but I forget where.

So I'm afraid there isn't a simple fix for this. It may or may not be possible to make this work, but it's very low priority so I can't promise anything.

starquake, 2016-11-18:
Thanks a lot for the reply! Wasn't expecting a quick fix.  I'll just keep hoping someday it will be possible.

Antiriad, 2016-12-21:
I'll keep hoping too. Thanks!

starquake, 2016-12-21:
I did find a way to output 240p over RGB. More info here:
http://www.retrorgb.com/rpi240pvga.html

In short you can use a Gert VGA and connect it to a Ultimate Scart Adapter.

Antiriad, 2016-12-21:
@starquake [I'm aware of this](https://forum.recalbox.com/topic/3475/recalbox-sur-tv-crt-en-rgb), thank you. But It would be useful also to output in 240p via Composite.

MrBrax, 2017-01-02:
Been researching this for the past 3-4 months, went with composite 480i in the end, flickering garbage.

I don't feel very tempted to buy a product (UMSA) that has "Just make sure that your PC outputs VGA lower resolution with 15Khz" in the description..

So I'm really hoping this is gonna work at some point!

jimbothegigolo, 2017-01-25:
I'm very interested in this as well.
I am hesitant to buy a Raspberry Pi knowing it will only output 480i and the 240p over HDMI is very lackluster.
240p on S-Video, Component or even Composite would be great as well.

Something I've seen on the PSP Go that I found interesting was the way they used the headphone jack.

[image]

MrBrax, 2017-01-25:
I've almost achieved it, getting a cheap HDMI->VGA adapter (Powered, important!), and a VGA -> RCA converter i had for my arcade cabinet (it does not produce 240p/15khz however, but it looks a lot sharper), and setting up the resolution to be 240p via hdmi_mode.

Getting an UMSA would be the last step, but i'm not really feeling like buying a €30 gizmo as said earlier. Some countries have stores with cheaper cables but i'm not that lucky to have those.

In a few years, maybe someone will have hacked the firmware on this 👌

starquake, 2017-01-25:
@jimbothegigolo The Raspberry PI 2 and later also has a headphone jack that also can output composite when using the right cable.

[image]

Unfortunately it doesn't output 240p and that's exactly what this request is all about.

starquake, 2017-01-25:
BTW Arcadeforge is working on this:
http://arcadeforge.net/PiJamma/PI2SCART-Preorder::264.html

And there's also this:
http://www.rgb-pi.com/

MrBrax, 2017-01-25:
@starquake i've seen that, looks promising! However, that means no GPIO pins left for non-controller use push buttons sadly.

jimbothegigolo, 2017-01-25:
@starquake I am aware of that, but composite over 480i isn't exactly ideal.

Antiriad, 2017-01-25:
@MrBrax @keilmillerjr @jimbothegigolo excuse me but the original request is different.

Indeed is yet possible to easily get a 240p by HDMI > VGA > SCART or GPIO (VGA 666) > SCART.

The original request is instead for 240p on **Composite** out (not component) that should allow to get a video mode faithful to some old retro-gaming consoles like NES or Genesis (that had a composite out). A kind of _sdtv_mode=3_  option would be great.

MrBrax, 2017-01-25:
@Antiriad yes, it would be amazing to do it via composite, but due to the firmware, it doesn't really sound possible. Perhaps in a future hardware revision?

Antiriad, 2017-01-25:
@popcornmix gave a small glimmer of hope. Even a RP1 can emulate very good NES, Genesis, PC Engine etc. at 60fps in full speed at low resolution (480i) on a CRT TV.
It would be great if,  with some kind of "magic", we could set in config.txt a _sdtv_mode=3_ with, e.g., a 320x224 resolution.
Not 100% accurate but surely much better than 480i.

I want to believe! ;D

starquake, 2017-01-25:
The console also looks great on 240p. You'll only get something like 80x25 characters, maybe even less but the picture is sharp and stable.

starquake, 2017-01-25:
@jimbothegigolo 

> I am aware of that, but composite over 480i isn't exactly ideal, if not eye gouging.

I agree. That's exactly what I described when creating this issue.

> There's also the Gert VGA666 if you're willing to settle for VGA. You might be able to convert it to SCART too with a Ultimate SCART Adapter (also from Arcadeforge)

Yep that's what I said in another post and it's pretty much what I'm using.

Don't want to be annoying but you might want to read up on my other posts.

keilmillerjr, 2017-01-29:
@Antiriad I meant composite. I would have thought that had been an obvious error. I know what I meant to type, but the two words are closely spelled and are both nouns for a type of analogue video transmission.

kevinfishburne, 2017-02-04:
I've been researching this for a long time now and it blows my mind there is apparently no good solution. Here's the deal: (1) Arcade 15KHz CRTs with RGB/SCART are no longer manufactured and are rare and expensive. (2) NTSC/PAL CRT TV's are also no longer manufactured, but they are cheap and plentiful at thrift stores (Goodwill in the U.S., for example) and they typically have composite and S-Video inputs. (3) Current solutions involve daisy-chaining multiple sketchy adapters; we need something more elegant. (4) HDMI is the most common modern video output format. Conclusion: In the interest of classic gaming preservation, if you do the math, what we need for maximum usefulness to the most number of people is an adapter to take an HDMI video signal and convert it to 240p/288p signal over both composite and S-Video. This will allow anyone with a RPi/PC/laptop/etc. and $10 CRT TV to run classic games exactly as they were intended to look. I say we find a communication hub (Discord/Quip/Facebook/whatever) and start planning a Kickstarter to design and fabricate this active adapter using completely open specifications. Who's with me? I started a Discord server for the idea if anyone wants to take the discussion there: https://discord.gg/dJzGM

tekn0x, 2017-02-04:
I was able to get pretty close with an off the shelf HDMI to Component cable. This is outputting true 240p. I just have a vertical sync issue. https://www.youtube.com/watch?v=VLBHWZV1dZ4

jimbothegigolo, 2017-02-04:
@kevinfishburne Your discord invite is expired.

@starquake 
If you use a Pi2SCART you should be able to breakout that signal into S-Video (S-VHS) without much trouble at all, just like there are SCART sockets on some sets that will accept all three composite, S-Video and RGB signals.  
No need to spend hundreds on fancy adapter either.
You might want to buy one of one of those RGB Scart to S-VHS adapters, they go 'round for 2 bucks each and to get composite from it, simply rid the S-Video cable and keep the RCA cables.

kevinfishburne, 2017-02-06:
@tekn0x - I've seen that video and think it may be an illusion based on the vertical scrolling of the video output. I noticed in certain TG16 games on my RPi on a CRT TV that when the background was scrolling at a speed matching the refresh rate that it created what looked like perfect scan lines. I could be wrong, but if the output were stabilized the scanlines might go away.

@starquake - Apparently Pi2SCART is still taking pre-orders. Is it open-spec? I don't see anything about that. There are HDMI to SCART converters, but no idea if they'll keep vsync and do 240p when daisy chained to an SCART to S-Video adapter. I've been researching this for a while and everything I've found is a mess within a mess to infinity and beyond.

starquake, 2017-02-06:
@kevinfishburne 

I think Pi2SCART will make an end to at least a part of the mess. The spec of GERT VGA is open. And you there's lot of schematics to find on how to convert VGA to RGBHV. Pi2SCART is just a solution with everything rolled into one.

For now I think the GERT VGA 666 + Ultimate SCART is the best solution.

popcornmix, 2017-02-28:
I have a test firmware here:
https://drive.google.com/uc?id=0B-6zmEDJwxZEME9lSEFZRmQ5RGs&export=download
 Can you set (in config.txt):
``sdtv_progressive_scan=1``
and report back.
We have a progressive scan bit in the VEC (composite) register set.
We've tested that before and it wasn't sufficient. I've also tried disabling the interlaced signal that gets set. I'm not sure if it does the right thing (my composite TV's quality is a bit too low to tell).

MrBrax, 2017-02-28:
@popcornmix How would i "install" this?

2skoops, 2017-03-01:
popcornmix:
You're amazing!  I just bought an old SOny Trinitron CRT that I'm planning on RGB modding and putting into an arcade cabinet.  I'm still waiting on a Gert's VGA adapter though, so until then I've been playing around with composite 480i... which looks like a total flickering mess!  I happened to check the thread today and saw your new firmware, so I just briefly tested it out.

It's such a huge improvement running this monitor over 240p.  No more flickering, just beautiful stable scanlines.  Everything seemed to be solid for the few minutes I played around with it.  The only thing I noticed was that my picture was noticeably warped and slightly titled near the edges of the screen.  It could just be my TV, since it's opened up right now, but I didn't notice it when I was running 480i.  Could be an issue with the sync over composite... not sure.  I'll have to do a better comparison with 240p vs. 480i on the set tomorrow.  (There was also a high-pitched whine from the speakers, but I'm guess that's just because the TV's guts are exposed right now)

I still plan on moving to 15kHz 240p VGA -> RGB once I get the parts I need, but having a 240p composite hookup is just so convenient and will please a lot of retro gamers!  Thanks again for your hard work!

MrBrax:
Basically you need to copy all of the files into your /boot/ folder.  Probably the easiest way is to power off your Pi, take out the microSD card, and plug it into a card reader on your computer.  Then copy all of the files from the zip over into the boot folder (remember to back up the old versions first).  After that, open /boot/config.txt and add the line popcornmix mentions above:
sdtv_progressive_scan=1
Save, close, eject, put it back into your Pi, and boot it up.

Antiriad, 2017-03-01:
@popcornmix awesone!
Will it work also on a good old Raspberry Pi 1 B?

Henrito, 2017-03-01:
@popcornmix 240p composite works with my Raspberry Pi 1 B and 2.

Antiriad, 2017-03-01:
Thank you @Henrito !!! Can't wait to test it with Lakka!

starquake, 2017-03-01:
Amazing, will test as soon as possible!

danmons, 2017-03-01:
Thank you so much, this is a long wanted feature. Will attempt to test on a couple of different boards as soon as possible.

MrBrax, 2017-03-01:
@2skoops oh i thought i had to flash it. i'll test it the moment i get home from work!!

starquake, 2017-03-01:
Can anyone post a picture of what it looks like?

Henrito, 2017-03-01:
@starquake 
[image]
[image]

MrBrax, 2017-03-01:
Is that really pixel mapped 1:1? Looks sort of blurry.

danmons, 2017-03-01:
It's still CVBS, so it will still be a little blurry, even if it is progressive scan.  The progressive scan will make moving pictures much better, as you won't get the interlace flicker.

starquake, 2017-03-01:
SCART RGB will give you better quality. But this will hopefully make the composite out on the Raspberry Pi look like composite on a Famicom for example.

MrBrax, 2017-03-01:
@danmons yes, but are you sure the scaling/retroarch resolution etc is set correctly to 1:1?

starquake, 2017-03-01:
@MrBrax what do you mean with 1:1?

Composite is analog. There are no pixels. If we're talking about NTSC there is a resolution of 240 lines vertically. You can use any resolution horizontally. Though you won't be able to see a resolution like 1920. 

@danmons What resolution is output by the Pi? 720x240?

keilmillerjr, 2017-03-01:
I’m so excited to try this out! Will probably order a new pi3 and the composite breakout cable, so it might be a few weeks for me.

MrBrax, 2017-03-01:
@starquake but the emulator has options for scaling & aspect ratios, it's not analog until it comes out of the rca connector

hmusta, 2017-03-01:
@starquake I guess a better way to word the question is:

Let's say you're running at a resolution of 256x224 with a custom hdmi_timings settings. If you set the progressive scan flag for composite video, does it output 240 lines or 224 lines? IIRC, an SNES outputs 224 lines (262 if you include blanking)

MrBrax, 2017-03-01:
^ retroarch has to be configured to output at that resolution also, with a custom aspect ratio

i can't wait to get home now haha

starquake, 2017-03-01:
@hmusta @MrBrax  Ah so the question is if the lines are mapped 1:1

I read somewhere it outputs 720x480 in ntsc mode so I guess it will be 720x240 for progressive. Will have to see when I get home. Don't know if hdmi_timings has any effect at all for composite out.

Right now, I use Lakka with RGUI forced to a 4:3 aspect ratio with a 1920x240 resolution without interpolation so both 256x240 and 320x240 look great.

In Lakka I set the system config and the core config to NOT crop the borders so it outputs 240p. It works with most cores. There is a bug with the SNES core though, it doesn't respect the no cropping and scales either 224p to 240p or 239p (not sure). Will have to report a bug about it somewhere someday.

(P.S. we should move this discussion to some forum :-) )

MrBrax, 2017-03-01:
yep yep

totally depends on the emulators for sure, for nes, they're all widely different on how they crop the picture, and i think either fcemu or nestopia had the least cropping.

well, reddit has a whole bunch of subs/posts about this :)

Antiriad, 2017-03-01:
@starquake hdmi_timings does not affect composite out, only HDMI or Gert's VGA.

240p via composite will not give a "pixel perfect" video mode for some retro systems, but it's a **GREAT** "quick and easy" solution for a better emulation experience (although not perfect).

For a 1:1 video emulation experience take a look to AdvanceMAME and AdvanceMESS + Gert's VGA 666 + VGA to SCART cable on CRT or Arcade monitor. ;)

starquake, 2017-03-01:
@MrBrax Just turn off all cropping in system and in the core config. QuickNES and Nestopia will output 240p
Maybe continue discussion here? https://np.reddit.com/r/crtgaming/comments/5wuai2/raspberry_pi_now_support_240p_over_composite/

@Antiriad 
It depends on the width resolution, can't wait to get home

netux79, 2017-03-01:
What I need to get it working? I get that I need to copy the files to /boot but I'm asking about connecting the RPi to a CRT? What kind of cable I need? Do I need the headphones cable posted above or something else? Thanks

hizzlekizzle, 2017-03-01:
@netux79 If you have an older RPi with the composite-out jack, you just hook that over RCA cable to your TV. If it's one with the 3.5mm combo jack, you'll need a breakout cable that converts the 3.5mm jack to the standard red/white/yellow RCA plugs.

netux79, 2017-03-01:
@hizzlekizzle : I have a RPi3, so I think it would be the later option. Thanks hunterk!

bparker06, 2017-03-01:
>Let's say you're running at a resolution of 256x224 with a custom hdmi_timings setting

@hmusta I don't know how the HDMI hardware in an RPi works specifically, but on typical transmitter chips like Analog Devices they usually have a minimum clock frequency, and it may not like 240p directly. How TVs handle this is described in the EIA/CEA-861 standard, which is to use pixel repetition. So a 320x240 signal over HDMI is actually output over the wire as 2880x240 and then in the AVI InfoFrame packet it specifies what the repetition factor is (9 * 320 = 2880). I think the firmware in an RPi would have to explicitly support that functionality, and I'm not sure it does. Also these "game formats" were not added until revision B of that standard and are completely optional to implement.

hmusta, 2017-03-01:
@bparker06 since HDMI has a 25MHz lower bound on pixel clock, setting the resolution that way with hdmi_timings won't work if you use the HDMI port and you have to do some pixel multiplication. it does work, however, if you're outputting over VGA with VGA666

starquake, 2017-03-01:
@popcornmix Works great! Just tested NTSC progressive and I have a stable image.

Lakka thinks it is using a 720x480@60Hz screen which is great because the aspect ratio is correct this way (as opposed to 720x240) I guess every line is skipped?

Did some tests with the 240p suite but I can confirm it does 240p at 60hz. When showing horizontal stripes you see black and white lines, just as is supposed to happen:
[image]
The striped sprite also show there is no filtering and the 240p works as you would expect:
[image]

The flashing sprite also flashes which proves things are running at 60 fps.

For people using Lakka or RetroPie. When scaling the NES width (256) to 720 you can see some slight distortion because 720 is not a multiple of 256 so not every pixel is scaled evenly. You can enable integer scale which resolves this and still matches the original aspect ratio closely. Don't know how that works because 512 is not very close to 720 but the image aspect ratio looks fairly good.

It all looks very much like a Famicom AV

jfroco, 2017-03-01:
@popcornmix  Thank you! At least.. one of the most wanted features.
Just tested it using a fresh Retropie 4.1 installation and it works great.

I'm playing Castlevania/NES that was impossible to play previously because of the flickering. I will post a video tonight.

starquake, 2017-03-01:
@jfroco Maybe you already know this but you can use the 240p test suite to see if the 240p output of the emulator core matches up with the 240p output of the pi. If it doesn't you will see thicker black or white lines across the screen.

popcornmix, 2017-03-01:
I've pushed this change to rpi-update with a slight change.
``sdtv_progressive_scan`` is no more.
You should add 16 onto ``sdtv_mode`` if you want 240p output.
That fits slightly more cleanly with how the firmware handles ``sdtv_mode``.

Examples:
```
sdtv_mode=16    # progressive NTSC
sdtv_mode=0x10  # progressive NTSC (using hex notation)
sdtv_mode=18    # progressive PAL
sdtv_mode=0x12  # progressive PAL (using hex notation)
```

MrBrax, 2017-03-01:
The screen is just scrolling downwards for me, did i miss some setting? PAL over here.

popcornmix, 2017-03-01:
@MrBrax are you testing the firmware zip file or the latest rpi-update firmware?
What did you add to config.txt?

MrBrax, 2017-03-01:
I just did an update but apt-get probably doesn't have the latest one here, do i have to replace certain files from this repo?

popcornmix, 2017-03-01:
You need to run:
``sudo rpi-update``
but be aware this gets you the bleeding edge firmware.
If you are unsure about doing this then wait for the firmware to reach apt (in perhaps a few weeks).

[161 more replies]

closed 2017-04-18 by starquake