raspberrypi/firmware #684: RAM corruption using firmware

amurzeau, 2016-11-20:
Hi,

Using the latest firmware from master (commit af555f6), I get RAM corruptions:
```
memtester version 4.3.0 (32-bit)
Copyright (C) 2001-2012 Charles Cazabon.
Licensed under the GNU General Public License version 2 (only).

pagesize is 4096
pagesizemask is 0xfffff000
want 0MB (65536 bytes)
got  0MB (65536 bytes), trying mlock ...locked.
Loop 1:
  Stuck Address       : testing   9FAILURE: possible bad address line at offset 0x00003b64.
Skipping to next test...
  Random Value        : ok
  Compare XOR         : ok
  Compare SUB         : ok
  Compare MUL         : ok
  Compare DIV         : ok
  Compare OR          : ok
  Compare AND         : ok
  Sequential Increment: ok
  Solid Bits          : ok
  Block Sequential    : ok
  Checkerboard        : ok
  Bit Spread          : testing  15FAILURE: 0x00008000 != 0x00028000 at offset 0x00000720.
FAILURE: 0x00008000 != 0x00028000 at offset 0x00001ca0.
FAILURE: 0x00008000 != 0x00028000 at offset 0x000020c0.
FAILURE: 0x00008000 != 0x00028000 at offset 0x00002420.
  Bit Flip            : testing 128FAILURE: 0x00010000 != 0x00000000 at offset 0x00002ffc.
  Walking Ones        : ok
  Walking Zeroes      : ok
  8-bit Writes        : ok
  16-bit Writes       : ok
```

This cause any RAM intensive usage to fail (like .deb decompression, checksum files, kernel panic at random location, etc ...).
I don't use overclocking. The only lines in /boot/config.txt that are not commented are:
```
disable_overscan=1
gpu_mem=128
dtparam=audio
```
I use raspberry pi 1 B.

Reverting the firmware only files to 20151118 (bootcode.bin, fixup*.dat and start*.elf only) and keeping the latest kernel from raspbian (4.4.32+) makes RAM issues go away (memtester report no error and I can do RAM intensive tasks without error).

How can I investigate more ?

Maybe its related, but here are the different kernel command lines:
2015 firmware:
`Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708.boardrev=0xd bcm2708.serial=0x2479fcfd smsc95xx.macaddr=B8:27:EB:79:FC:FD bcm2708_fb.fbswap=1 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p6 rootfstype=ext4 elevator=deadline rootwait`
Lastest firmware:
`Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708.boardrev=0xd bcm2708.serial=0x2479fcfd smsc95xx.macaddr=B8:27:EB:79:FC:FD bcm2708_fb.fbswap=1 bcm2708.uart_clock=48000000 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p6 rootfstype=ext4 elevator=deadline rootwait`

The `sdhci-bcm2708.emmc_clock_freq=250000000` in the 2015 firmware were replaced by `bcm2708.uart_clock=48000000` in latest kernel.
I tried to add `sdhci-bcm2708.emmc_clock_freq=250000000` to /boot/cmdline.txt but I still have memtester failures.

popcornmix, 2016-11-22:
Can you identify the exact update which caused this. See:
https://github.com/Hexxeh/rpi-firmware/commits/master

If you click on each commit the end of the url contains a git hash. Run
```sudo rpi-update <hash>```
to revert back to that version. Report the first version with this error.

amurzeau, 2016-11-27:
The first non working commit is ac0c0bf with message:

    firmware: sdram: Remove disable_pvt option and always do the nopvt calibration
    firmware: bootcode.bin: Set PVT explicitly with config.txt option
    
    firmware: dtoverlay: Use fewer shell commands, tidy up
    
    firmware: imx219_cypto: use 0 size sub-addresses and remove I2C_FLAGS_READ_REQUIRES_STOP
    See: https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=147744

With good commit (previous one) 297432f (extra: Update for support of Pi zero v2):
```
root@raspberrypi:/boot# uname -a
Linux raspberrypi 4.4.32+ #924 Tue Nov 15 18:06:42 GMT 2016 armv6l GNU/Linux
root@raspberrypi:/boot# vcgencmd version
May 17 2016 17:29:50
Copyright (c) 2012 Broadcom
version 72491c2e16892dd7e2a4e5b253ef499e2b6a028f (clean) (release)
root@raspberrypi:/boot# md5sum *.dat *.elf *.bin
b4d07df40cd83ded57eb79d7c6a0ee6a  fixup_cd.dat
3a6d6e97d57758dba7e1f82e4dd456b6  fixup.dat
b73e3b88c60c352b35962663613682c9  fixup_db.dat
5fa56041e87f5752b9cb23c6040b009a  fixup_x.dat
84cf98bed681335c29c5ee1a4d2c8f47  start_cd.elf
d8c0743b6299194f5a4a2ea0ab32a505  start_db.elf
e0b80e7a679a0e26bda055176bd022b9  start.elf
8aee170ed7bec219ee829a8c310dc082  start_x.elf
0a844d5ceadba293d3945ef542851af9  bootcode.bin
root@raspberrypi:/boot# memtester 20M 1
memtester version 4.3.0 (32-bit)
Copyright (C) 2001-2012 Charles Cazabon.
Licensed under the GNU General Public License version 2 (only).

pagesize is 4096
pagesizemask is 0xfffff000
want 20MB (20971520 bytes)
got  20MB (20971520 bytes), trying mlock ...locked.
Loop 1/1:
  Stuck Address       : ok
  Random Value        : ok
  Compare XOR         : ok
  Compare SUB         : ok
  Compare MUL         : ok
  Compare DIV         : ok
  Compare OR          : ok
  Compare AND         : ok
  Sequential Increment: ok
  Solid Bits          : ok
  Block Sequential    : ok
  Checkerboard        : ok
  Bit Spread          : ok
  Bit Flip            : ok
  Walking Ones        : ok
  Walking Zeroes      : ok
  8-bit Writes        : ok
  16-bit Writes       : ok

Done.
```

With bad commit ac0c0bf (firmware: sdram: Remove disable_pvt option and always do the nopvt calibration):
```
root@raspberrypi:/boot# uname -a
Linux raspberrypi 4.4.32+ #924 Tue Nov 15 18:06:42 GMT 2016 armv6l GNU/Linux
root@raspberrypi:/boot# vcgencmd version
May 20 2016 19:01:31
Copyright (c) 2012 Broadcom
version faf071dd4885c5ac1a89483d35a5326e7f69495f (clean) (release)
root@raspberrypi:/boot# md5sum *.dat *.elf *.bin
00cb9db35988e4d8b626bc43d669eab8  fixup_cd.dat
9bc95d4b6d09d4cda491e6e0c887fa1f  fixup.dat
1aff1946462622b130b30823577e03ea  fixup_db.dat
509838174c667e2ee6d630031e52f04e  fixup_x.dat
a51e3fc92eef8be7af823c8d2ed4ccb7  start_cd.elf
36e9b6c91b37376cdc9413194c0d2ee2  start_db.elf
3e16fde656f979f67d943586a65e01b9  start.elf
1d0d7b89ab42b9f41bcdddaea0ec5dee  start_x.elf
1a4881102a0d8170d9f6d39ad6cd7f47  bootcode.bin
root@raspberrypi:/boot# memtester 20M 1
memtester version 4.3.0 (32-bit)
Copyright (C) 2001-2012 Charles Cazabon.
Licensed under the GNU General Public License version 2 (only).

pagesize is 4096
pagesizemask is 0xfffff000
want 20MB (20971520 bytes)
got  20MB (20971520 bytes), trying mlock ...locked.
Loop 1/1:
  Stuck Address       : testing   0FAILURE: possible bad address line at offset 0x0055d064.
Skipping to next test...
  Random Value        : ok
  Compare XOR         : ok
  Compare SUB         : ok
  Compare MUL         : ok
  Compare DIV         : ok
  Compare OR          : ok
  Compare AND         : ok
  Sequential Increment: ok
  Solid Bits          : ok
  Block Sequential    : ok
  Checkerboard        : testing  13FAILURE: 0x55555555 != 0x55455555 at offset 0x002756dc.
  Bit Spread          : testing  41FAILURE: 0x01000000 != 0x01400000 at offset 0x009fe1c0.
  Bit Flip            : testing 128FAILURE: 0x00010000 != 0x00000000 at offset 0x0001a2fc.
FAILURE: 0x00010000 != 0x00000000 at offset 0x0001a59c.
FAILURE: 0x00010000 != 0x00000000 at offset 0x0001a73c.
[... 6843 lines like above ...]
FAILURE: 0x00010000 != 0x00000000 at offset 0x009fd93c.
FAILURE: 0x00010000 != 0x00000000 at offset 0x009fd95c.
  Walking Ones        : ok
  Walking Zeroes      : ok
  8-bit Writes        : ok
  16-bit Writes       : ok

Done.
```

So it seems to be related to PVT changes.
The raspberry pi run around 47°C usually (between 45°C and 50°C) if that matter.
At the time of writting, I get this:
```
root@raspberrypi:/boot# cat /sys/class/thermal/thermal_zone0/temp
46540
```

Maybe I need to set something in config.txt with newer firmwares ?

amurzeau, 2016-11-27:
Additional information:
The working commit also works fine even with `disable_pvt=1` in config.txt.
`vcgencmd measure_clock` return almost the same values with both firmwares. The values are not fixed but slightly change (like 700000000 to 700074000 for arm freq).

```
root@raspberrypi:/boot# for src in arm core h264 isp v3d uart pwm emmc pixel vec hdmi dpi ; do \
>     echo -e "$src:\t$(vcgencmd measure_clock $src)" ; \
> done
arm:    frequency(45)=700074000
core:   frequency(1)=250000000
h264:   frequency(28)=250000000
isp:    frequency(42)=250000000
v3d:    frequency(43)=250000000
uart:   frequency(22)=3000000
pwm:    frequency(25)=0
emmc:   frequency(47)=250000000
pixel:  frequency(29)=0
vec:    frequency(10)=108000000
hdmi:   frequency(9)=0
dpi:    frequency(4)=0
root@raspberrypi:/boot# for id in core sdram_c sdram_i sdram_p ; do \
>     echo -e "$id:\t$(vcgencmd measure_volts $id)" ; \
> done
core:   volt=1.2000V
sdram_c:        volt=1.2000V
sdram_i:        volt=1.2000V
sdram_p:        volt=1.2250V
root@raspberrypi:/home/doc# for i in {0..255}; do vcgencmd measure_clock $i; done | grep -v "=0"
frequency(1)=250000000
frequency(10)=108000000
frequency(22)=3000000
frequency(26)=4800000
frequency(28)=250000000
frequency(42)=250000000
frequency(43)=250000000
frequency(45)=700000000
frequency(47)=250000000
frequency(50)=1920000
frequency(56)=4559000
```

popcornmix, 2016-11-28:
Can you report the output of
``vcgencmd schmoo pvt``
with the last good and first bad firmwares? Also include the good firmware with ``disable_pvt=1``.

amurzeau, 2016-11-28:
Last good firmware without disable_pvt:
```
root@raspberrypi:/boot# vcgencmd schmoo pvt
aphy=000c4400, dphy=000c4400
```

Last good firmware with disable_pvt=1:
```
root@raspberrypi:/boot# vcgencmd schmoo pvt
aphy=000c4400, dphy=000c4400
```

First bad firmware:
```
root@raspberrypi:/boot# vcgencmd schmoo pvt
aphy=000c2100, dphy=000c2100
```

popcornmix, 2016-11-28:
With the bad firmware can you add ``force_pvt=0`` and report output of ``vcgencmd schmoo pvt`` and memory test?

amurzeau, 2016-11-28:
With force_pvt=0 and the bad firmware, I still have memory test failures:
```
root@raspberrypi:/boot# vcgencmd version
May 20 2016 19:01:31
Copyright (c) 2012 Broadcom
version faf071dd4885c5ac1a89483d35a5326e7f69495f (clean) (release)

root@raspberrypi:/boot# vcgencmd schmoo pvt
aphy=000c2100, dphy=000c2100

root@raspberrypi:/boot# grep -v -e '^#' -e '^$' /boot/config.txt
disable_overscan=1
gpu_mem=128
dtparam=audio
force_pvt=0

root@raspberrypi:/boot# vcgencmd get_config force_pvt
force_pvt is unknown

root@raspberrypi:/boot# memtester 20M 1
memtester version 4.3.0 (32-bit)
Copyright (C) 2001-2012 Charles Cazabon.
Licensed under the GNU General Public License version 2 (only).

pagesize is 4096
pagesizemask is 0xfffff000
want 20MB (20971520 bytes)
got  20MB (20971520 bytes), trying mlock ...locked.
Loop 1/1:
  Stuck Address       : testing  13FAILURE: possible bad address line at offset 0x00519c60.
Skipping to next test...
  Random Value        : -^C
```

I also tried with force_pvt=1 but also have memory errors and same schmoo pvt output.

popcornmix, 2016-11-28:
``force_pvt`` is a bootcode.bin option, so doesn't show in ``vcgencmd get_config``.

Can you try bad firmware (``start*.elf``/``fixup*.dat``) with good ``bootcode.bin``?
``vcgencmd schmoo pvt`` and memtester output.

amurzeau, 2016-11-28:
Still failures with bad firmware and good bootcode.bin:
```
root@raspberrypi:/boot# vcgencmd version
May 20 2016 19:01:31
Copyright (c) 2012 Broadcom
version faf071dd4885c5ac1a89483d35a5326e7f69495f (clean) (release)

root@raspberrypi:/boot# grep -v -e '^#' -e '^$' /boot/config.txt
disable_overscan=1
gpu_mem=128
dtparam=audio
force_pvt=0

root@raspberrypi:/boot# md5sum *.dat *.elf *.bin
00cb9db35988e4d8b626bc43d669eab8  fixup_cd.dat
9bc95d4b6d09d4cda491e6e0c887fa1f  fixup.dat
1aff1946462622b130b30823577e03ea  fixup_db.dat
509838174c667e2ee6d630031e52f04e  fixup_x.dat
a51e3fc92eef8be7af823c8d2ed4ccb7  start_cd.elf
36e9b6c91b37376cdc9413194c0d2ee2  start_db.elf
3e16fde656f979f67d943586a65e01b9  start.elf
1d0d7b89ab42b9f41bcdddaea0ec5dee  start_x.elf
0a844d5ceadba293d3945ef542851af9  bootcode.bin

root@raspberrypi:/boot# vcgencmd schmoo pvt
aphy=000c2100, dphy=000c2100

root@raspberrypi:/boot# memtester 20M 1
memtester version 4.3.0 (32-bit)
Copyright (C) 2001-2012 Charles Cazabon.
Licensed under the GNU General Public License version 2 (only).

pagesize is 4096
pagesizemask is 0xfffff000
want 20MB (20971520 bytes)
got  20MB (20971520 bytes), trying mlock ...locked.
Loop 1/1:
  Stuck Address       : testing   1FAILURE: possible bad address line at offset 0x006d1da4.
Skipping to next test...
  Random Value        : \^C
```

popcornmix, 2016-11-28:
Can you try removing ``audio_pwm_mode=1``?

amurzeau, 2016-11-28:
You mean adding audio_pwm_mode=0 to config.txt ? I don't have "audio_pwm_mode=1" currently in config.txt

popcornmix, 2016-11-28:
Yes, you are right - I though you were using the experimental higher quality analogue audio option (``audio_pwm_mode=2``). ``audio_pwm_mode=1`` is a default so is fine.

amurzeau, 2016-11-28:
I tried several combinations with the bad firmware + good bootcode.bin with without success:
- without `dtparam=audio`
- without `dtparam=audio`, with `audio_pwm_mode=2`
- without `dtparam=audio`, with `audio_pwm_mode=0`

I get memory errors with all these three combinations.
With the last, schmoo pvt is still aphy=000c2100, dphy=000c2100.

popcornmix, 2016-11-28:
Yes, ``aphy=000c2100, dphy=000c2100`` is the cause of the corruption.
We should be explicitly setting ``aphy=000c4400, dphy=000c4400`` in bootcode.bin.

So either that setting is failing on your Pi, or it is being reset later (in start.elf).

Can you try bad firmware with this bootcode.bin
https://www.dropbox.com/s/fo7f01m5geo828n/bootcode.bin?dl=0

By default it should be the same as latest (so bad). Add to config.txt
```
pvt_delay1=1000000
pvt_delay2=1000000
pvt_delay3=1000000
```
and boot. It should take tens of seconds longer to boot (lots of delays).
Any difference to aphy/dphy?

amurzeau, 2016-11-28:
I've installed your bootcode.bin along with bad firmware.
Without the additional pvt_delay config, I indeed still have the same bad aphy/dphy. But with the delays I still have the bad aphy/dphy.

```
root@raspberrypi:/home/doc# grep -v -e '^#' -e '^$' /boot/config.txt
disable_overscan=1
gpu_mem=128
dtparam=audio
pvt_delay1=1000000
pvt_delay2=1000000
pvt_delay3=1000000
root@raspberrypi:/home/doc# vcgencmd schmoo pvt
aphy=000c2100, dphy=000c2100
```
I tried larger delays (x3 and x10) and also adding `force_pvt=0` but to no avail. Still `aphy=000c2100, dphy=000c2100`.

Additionally, I also tried the bad bootcode.bin with good fixup*.dat and start*.elf and I got `aphy=000c4400, dphy=000c4400`. So maybe it's start.elf reseting it.

popcornmix, 2016-11-28:
Just to confirm the settings were applied correctly, did booting take a long long time?
Can you report any text on the main processor IC2 (e.g. hynix/micron/samsung)
Do you have another Pi to test?

amurzeau, 2016-11-28:
Ah forgot to tell about that. No the boot process wasn't much longer (around 1min 05 to reboot, +/- 2s, whatever the delay values).

Sorry I don't have another Pi to test. The main processor (IC2) is a Micron 3DAI8 D9QHN.

popcornmix, 2016-11-28:
With new bootcode.bin and delay settings from power being applied I get a blank screen of 16 seconds before the usual rainbow splash appears (which is normally virtually instant).

Are you sure you have correct version?
```
pi@domnfs:~ $ md5sum /boot/bootcode.bin 
7e9452ba7dc86c966059c2cf29d7cb2b  /boot/bootcode.bin
```
Are you using a standard raspbian install (i.e. without NOOBS or other dual boot partitioning which may complicate things). Report output of ``mount | grep boot``

amurzeau, 2016-11-28:
With HDMI, the rainbow splash is instant.
I confirm the md5sum, I have the same for bootcode.bin in /boot.

But I use NOOBS. I guess this is why I have this ram issue. (I wasn't aware of the NOOBS stuff and "recovery" partition until now ...)
Here is the infos:
```
# mount | grep /boot
/dev/mmcblk0p5 on /boot type vfat (rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)

# lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0  7.3G  0 disk
├─mmcblk0p1 179:1    0  1.2G  0 part /mnt
├─mmcblk0p2 179:2    0    1K  0 part
├─mmcblk0p5 179:5    0   56M  0 part /boot
└─mmcblk0p6 179:6    0  6.1G  0 part /
root@raspberrypi:/mnt# ls -l
total 21984
-rwxr-xr-x 1 root root    17808 Jun 28  2013 bootcode.bin
-rwxr-xr-x 1 root root      268 Jun 28  2013 BUILD-DATA
drwxr-xr-x 2 root root    32768 Jun 28  2013 images
-rwxr-xr-x 1 root root       66 Jan  1  1980 recovery.cmdline
-rwxr-xr-x 1 root root   468536 Jun 28  2013 recovery.elf
-rwxr-xr-x 1 root root        0 Jun 28  2013 RECOVERY_FILES_DO_NOT_EDIT
-rwxr-xr-x 1 root root  2064512 Jun 28  2013 recovery.img
-rwxr-xr-x 1 root root 19800656 Jun 28  2013 recovery.rfs
-rwxr-xr-x 1 root root     9728 Jun 28  2013 riscos-boot.bin
drwxr-xr-x 9 root root    32768 Jun 28  2013 slides
root@raspberrypi:/mnt# cat BUILD-DATA
Build-date: 2013-06-26
NOOBS Version: 1.2.1
NOOBS Git HEAD @ 13821d09a000234d5f7aafe1904b592654093db8
rpi-userland Git HEAD @ 725b03007b0f53525767539bb435bdcfeed5ea66 refs/heads/master
rpi-firmware Git HEAD @ a0c98fa6bc0f93bad8cbfcf749f7be7dc2276192 refs/heads/master
```

popcornmix, 2016-11-28:
NOOBS complicates things.
mmcblk0p1 will contain the firmware for NOOBS
mmcblk0p5 will contain the firmware for raspbian

But I think bootcode.bin always comes from mmcblk0p1 (ping @ghollingworth to confirm).
Assuming that is correct then your problem may have been from a stale bootcode.bin in mmcblk0p1.

Can you try updating the bootcode.bin from mmcblk0p1 (i.e. the /mnt version)?

pelwell, 2016-11-28:
> But I think bootcode.bin always comes from mmcblk0p1 (ping @ghollingworth to confirm).

It has to - the boot ROM doesn't understand the NOOBS partition scheme.

popcornmix, 2016-11-28:
> It has to - the boot ROM doesn't understand the NOOBS partition scheme.

Well, bootcode.bin from mmcblk0p1 could in theory load bootcode.bin from mmcblk0p5 to ensure firmware is consistent, but I'm sure that isn't how it works.

amurzeau, 2016-11-28:
When updating the NOOBS /mnt/bootcode.bin file to yours and removing pvt_delay configs, the Pi boot but when it should boot the linux kernel, it seems to hang on a black screen.
I get this scenario:
- Instant rainbow splash screen for a couple of seconds
- Noob splash screen shows for another couple of seconds (saying to press shift for recovery mode)
- The screen goes black with the HDMI display saying "no input signal" for 2 seconds, then a black screen appears (but "no input signal" disappear). I don't have anything (it should show the rasbian kernel booting).

Maybe some other files are needed so the soft reboot works ? (like fixup*.dat ?)

popcornmix, 2016-11-28:
I would have expected that to work.
Can you try also copying ``start*.elf`` and ``fixup*.elf`` to mmcblk0p1?

amurzeau, 2016-11-28:
Not working either (same black screen).
I will try to bypass completely NOOBS by copying the entire /boot from raspbian to the mmcblk0p1 partition.

popcornmix, 2016-11-28:
Your NOOBS install is pretty old (looks like 3.5 years old) which could be causing issues with updating its firmware.

If you have a spare sdcard then I'd be interested if a standard raspbian install (before and after rpi-update) works okay with memtester (I believe it will but would be good to confirm that).

amurzeau, 2016-11-28:
I put everything from /boot (bad firmware + kernel images + overlays + config/cmdline) into mmcblk0p1.
It worked fine, `vcgencmd schmoo pvt` now report `aphy=000c4400, dphy=000c4400` and I have no memtester errors.

I will try with the latest (and with updating the /boot mount to mmcblk0p1) but it seems NOOBS was the issue in this story.

Thanks very much for your help on this :)

popcornmix, 2016-11-28:
Yes, thanks for running the tests.
We need to think about ways we can avoid the issue of old NOOBS with latest firmware causing issues as others might hit this problem.

amurzeau, 2016-11-28:
Latest firmware using rpi-update works fine:
```
root@raspberrypi:/home/doc# vcgencmd version
Nov 25 2016 16:03:50
Copyright (c) 2012 Broadcom
version 48a26a2ae46c497139b3d5a9c8d15485c7b3bfbc (clean) (release)
root@raspberrypi:/home/doc# vcgencmd schmoo pvt
aphy=000c4400, dphy=000c4400
```
And memtester shows no error.

So everything is now fine with latest firmware.
I'm now closing this issue, thanks very much for this :)

closed 2016-11-28 by amurzeau