raspberrypi/firmware #686: h264: video_decode stops sending output buffers

minfrin, 2016-11-27:
I have for a long time been trying to get to the bottom of a hang while trying to transcode video from DVB-T/DVT-T2 using gstreamer's omxh264dec and omxh264enc elements.

Looking at the vcdbg log, during the video_decode stage, we see the following below. Would it be possible to confirm whether this behaviour is normal or whether there is a bug?

pi@towerofpi9:~$ sudo vcdbg log msg 2>&1 | grep decode

// we start well

222323.608: video_decode:3:RIL: setup image requirements (544x576(544x576)x7)
222327.784: video_decode:3:RIL: decoding 24140 bytes, 2361053us, flags 30, (cb:0x3e4e9780)
222327.797: video_decode:3:RIL: asking for more
222327.848: video_decode:3:RIL: decoding 8352 bytes, 2200522us, flags 10, (cb:0x3e4e979c)
222327.862: video_decode:3:RIL: asking for more
222328.261: video_decode:3:RIL: new image requirements (544x576(544x576)x7)
222331.257: video_decode:3:RIL: decoding 4153 bytes, 2120348us, flags 10, (cb:0x3e4e97b8)
222331.269: video_decode:3:RIL: asking for more
222334.031: video_decode:3:RIL: decoding 1999 bytes, 2080321us, flags 10, (cb:0x3e4e97d4)
222334.042: video_decode:3:RIL: asking for more
222380.267: video_decode:3:RIL: decoding 2174 bytes, 2160310us, flags 10, (cb:0x3e4e97f0)
222380.279: video_decode:3:RIL: asking for more
222403.209: video_decode:3:RIL: decoding 5503 bytes, 2280310us, flags 10, (cb:0x3e4e980c)
222403.220: video_decode:3:RIL: asking for more
222419.430: video_decode:3:RIL: decoding 2611 bytes, 2240304us, flags 10, (cb:0x3e4e9828)
222419.441: video_decode:3:RIL: asking for more

// we start receiving "discard" callbacks, not sure why. It could possibly be because gstreamer has asked the Pi to give it RGBA frames, which the Pi doesn't support.

222419.733: video_decode:3:RIL:cb:next_frame_time:2080321000
222419.751: video_decode:3:RIL:cb:discard (cb:0x3e4e97d4), w:1 v:1 f:0 s:0
222547.894: video_decode:3:RIL: decoding 1927 bytes, 2320304us, flags 10, (cb:0x3e4e9844)
222547.906: video_decode:3:RIL: asking for more
222548.201: video_decode:3:RIL:cb:discard (cb:0x3e4e97b8), w:1 v:1 f:0 s:0
222627.263: video_decode:3:RIL: decoding 15224 bytes, 2680303us, flags 10, (cb:0x3e4e9860)
222627.275: video_decode:3:RIL: asking for more
222627.589: video_decode:3:RIL:cb:discard (cb:0x3e4e97f0), w:1 v:1 f:0 s:0
222653.226: video_decode:3:RIL: decoding 9048 bytes, 2520303us, flags 10, (cb:0x3e4e987c)
222653.240: video_decode:3:RIL: asking for more
222653.569: video_decode:3:RIL:cb:discard (cb:0x3e4e979c), w:1 v:1 f:0 s:0
222676.996: video_decode:3:RIL: decoding 4368 bytes, 2440303us, flags 10, (cb:0x3e4e9898)
222677.009: video_decode:3:RIL: asking for more
222677.308: video_decode:3:RIL:cb:discard (cb:0x3e4e9828), w:1 v:1 f:0 s:0
222691.396: video_decode:3:RIL: decoding 1874 bytes, 2400303us, flags 10, (cb:0x3e4e98b4)
222691.410: video_decode:3:RIL: asking for more
222691.700: video_decode:3:RIL:cb:discard (cb:0x3e4e980c), w:1 v:1 f:0 s:0
222733.372: video_decode:3:RIL: decoding 2100 bytes, 2480303us, flags 10, (cb:0x3e4e98d0)
222733.384: video_decode:3:RIL: asking for more
222733.678: video_decode:3:RIL:cb:discard (cb:0x3e4e9844), w:1 v:1 f:0 s:0
222753.796: video_decode:3:RIL: decoding 5020 bytes, 2600303us, flags 10, (cb:0x3e4e98ec)
222753.810: video_decode:3:RIL: asking for more
222754.149: video_decode:3:RIL:cb:I time:2361053, cb:0x3e4e9780, ar:64/33, crop:0x0 544x576, interval:40000000, interlace:1,1,0
222754.206: video_decode:3:RIL:cb:B time:2400303, cb:0x3e4e98b4, ar:64/33, crop:0x0 544x576, interval:40000000, interlace:1,1,0

// We get "resolution changing", not sure why. Could be because gstreamer decides to change the format to I420 from RGBA.

222754.226: video_decode:3:RIL:resolution changing
223368.123: video_decode:3:RIL:output dec_frames_full:3e4ea700 out:3e512460
223368.987: video_decode:3:RIL:output done, dec_frames_full:3e4ea774 out:0
223370.926: video_decode:3:RIL:output dec_frames_full:3e4ea774 out:3e5124e0
223371.879: video_decode:3:RIL:output done, dec_frames_full:0 out:0
223376.495: video_decode:3:RIL: decoding 1963 bytes, 2560303us, flags 10, (cb:0x3e4e9908)
223376.509: video_decode:3:RIL: asking for more
223376.846: video_decode:3:RIL:cb:B time:2440303, cb:0x3e4e9898, ar:64/33, crop:0x0 544x576, interval:40000000, interlace:1,1,0
223376.870: video_decode:3:RIL:output dec_frames_full:3e4ea7e8 out:3e512560
223378.386: video_decode:3:RIL:output done, dec_frames_full:0 out:0
223389.831: video_decode:3:RIL: decoding 1911 bytes, 2640303us, flags 10, (cb:0x3e4e9924)
223389.847: video_decode:3:RIL: asking for more
223390.172: video_decode:3:RIL:cb:B time:2480303, cb:0x3e4e98d0, ar:64/33, crop:0x0 544x576, interval:40000000, interlace:1,1,0
223390.197: video_decode:3:RIL:output dec_frames_full:3e4ea85c out:3e5125e0
223391.665: video_decode:3:RIL:output done, dec_frames_full:0 out:0
223397.317: video_decode:3:RIL: decoding 14845 bytes, 3000303us, flags 10, (cb:0x3e4e9940)
[... 248 lines cut ...]
[  18] 0x3d972e40: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3d972e60, d1rual) 'RIL buffer'
[  19] 0x3d986e80: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3d986ea0, d1rual) 'RIL buffer'
[  20] 0x3d99aec0: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3d99aee0, d1rual) 'RIL buffer'
[  21] 0x3d9aef00: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3d9aef20, d1rual) 'RIL buffer'
[  22] 0x3d9c2f40: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3d9c2f60, d1rual) 'RIL buffer'
[  23] 0x3d9d6f80: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3d9d6fa0, d1ruAl) 'RIL buffer'
[  24] 0x3d9eafc0: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3d9eafe0, d1rual) 'RIL buffer'
[  25] 0x3d9ff000: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3d9ff020, d1rual) 'RIL buffer'
[  43] 0x3da13040: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3da13060, d1rual) 'RIL buffer'
[  44] 0x3da27080: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3da270a0, d1rual) 'RIL buffer'
[  45] 0x3da3b0c0: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3da3b0e0, d1rual) 'RIL buffer'
[  46] 0x3da4f100: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3da4f120, d1rual) 'RIL buffer'
[  53] 0x3da63140: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3da63160, d1rual) 'RIL buffer'
[  52] 0x3da77180: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3da771a0, d1rual) 'RIL buffer'
[  51] 0x3da8b1c0: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3da8b1e0, d1rual) 'RIL buffer'
[  50] 0x3da9f200: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3da9f220, d1rual) 'RIL buffer'
[  49] 0x3dab3240: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3dab3260, d1rual) 'RIL buffer'
[  48] 0x3dac7280: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3dac72a0, d1rual) 'RIL buffer'
[  47] 0x3dadb2c0: used  80K (refcount 1 lock count 0, size    81920, align   32, data 0x3dadb2e0, d1rual) 'RIL buffer'
[   5] 0x3daef300: used  576 (refcount 1 lock count 0, size      512, align    4, data 0x3daef320, d0rual) 'ILCS VC buffer pool'
[   4] 0x3daef540: used  576 (refcount 1 lock count 0, size      512, align    4, data 0x3daef560, d0rual) 'ILCS VC buffer pool'
[   3] 0x3daef780: used 1.0M (refcount 1 lock count 8, size  1091584, align 4096, data 0x3daf0000, d1rual) 'ARM FB'
[   2] 0x3dbfafa0: used  16K (refcount 1 lock count 0, size    16384, align   32, data 0x3dbfafc0, d0ruAl) 'audioplus_tmp_buf'
[   1] 0x3dbfefe0: used 4.0K (refcount 1 lock count 0, size        0, align 4096, data 0x3dbff000, d1rual) 'camera fast alloc arena'
small allocs not requested

minfrin, 2016-11-27:
Same log, this time with debugging enabled:

pi@towerofpi9:~$ sudo vcdbg log msg 2>&1 | grep decode
480906.837: OMX.broadcom.video_decode:init
480906.941: top:create_component video_decode
480907.946: video_decode:RIL:GetParameter(3e4e92a4)
480907.957: video_decode:RIL:GetParameter(3e4e92a4)
480907.967: video_decode:RIL:GetParameter(3e4e92a4)
480907.977: video_decode:RIL:GetParameter(3e4e92a4)
480907.997: video_decode:RIL:GetParameter(3e4e92a4)
480908.010: video_decode:RIL:GetParameter(3e4e92a4)
480908.020: video_decode:RIL:GetParameter(3e4e92a4)
480908.031: video_decode:RIL:GetParameter(3e4e92a4)
480908.040: video_decode:RIL:GetParameter(3e4e92a4)
480908.051: video_decode:RIL:GetParameter(3e4e92a4)
480908.060: video_decode:RIL:GetParameter(3e4e92a4)
480908.079: video_decode:RIL:SetParameter(7f0000a9)
480908.088: video_decode:0:SetCallbacks
480908.100: video_decode:0:SetParameter(7f000001)
480908.111: video_decode:0:GetParameter(7f000002)
480908.123: video_decode:0:SetParameter(7f000004)
480908.292: video_decode:0:SetCallbacks
480908.566: video_decode:0:GetState(1)
480909.125: video_decode:0:GetParameter(2000001)
480909.135: video_decode:0:RIL:GetParameter(3e4e92a4)
480909.613: video_decode:0:GetParameter(2000001)
480909.624: video_decode:0:RIL:GetParameter(3e4e92a4)
480915.385: video_decode:0:GetParameter(2000001)
480915.396: video_decode:0:RIL:GetParameter(3e4e92a4)
480916.066: video_decode:0:SetParameter(2000001)
480916.077: video_decode:0:RIL:SetParameter(2000001)
480916.086: video_decode:0:RIL:PortDef, port 130
480916.092: video_decode:0:RIL: set input format
480916.542: video_decode:0:GetParameter(2000001)
480916.551: video_decode:0:RIL:GetParameter(3e4e92a4)
480917.080: video_decode:0:GetParameter(2000001)
480917.091: video_decode:0:RIL:GetParameter(3e4e92a4)
480917.422: video_decode:0:SetParameter(2000001)
480917.432: video_decode:0:RIL:SetParameter(2000001)
480917.441: video_decode:0:RIL:PortDef, port 130
480917.449: video_decode:0:RIL: set input format
480917.769: video_decode:0:GetParameter(2000001)
480917.778: video_decode:0:RIL:GetParameter(3e4e92a4)
480918.261: video_decode:0:GetParameter(2000001)
480918.272: video_decode:0:RIL:GetParameter(3e4e92a4)
480919.303: video_decode:0:GetParameter(6000001)
480919.315: video_decode:0:RIL:GetParameter(3e4e92a4)
480919.636: video_decode:0:GetParameter(6000001)
480919.648: video_decode:0:RIL:GetParameter(3e4e92a4)
480920.276: video_decode:0:SetParameter(6000001)
480920.287: video_decode:0:RIL:SetParameter(6000001)
480920.802: video_decode:0:GetParameter(2000001)
480920.812: video_decode:0:RIL:GetParameter(3e4e92a4)
480921.318: video_decode:0:SendCommand(2,131,3e451b94) Disable 131
480921.331: video_decode:0:sending cmd 2 to 131
480921.350: video_decode:0:starting command 2
480921.360: video_decode:0:RIL:GetParameter(3e4e92a4)
480921.378: video_decode:0:processingCommand=0
480922.853: video_decode:0:GetParameter(2000001)
480922.863: video_decode:0:RIL:GetParameter(3e4e92a4)
480923.497: video_decode:0:GetParameter(2000001)
480923.508: video_decode:0:RIL:GetParameter(3e4e92a4)
480924.181: video_decode:0:GetParameter(2000001)
480924.191: video_decode:0:RIL:GetParameter(3e4e92a4)
480924.648: video_decode:0:GetParameter(2000001)
480924.660: video_decode:0:RIL:GetParameter(3e4e92a4)
480925.251: video_decode:0:SendCommand(0,2,3e451b94) State to idle
480925.267: video_decode:0:starting command 0
480925.279: video_decode:0:RIL:GetParameter(3e4e92a4)
480925.352: video_decode:0:RIL:GetParameter(3e4e92a4)
[... 915 lines cut ...]
485085.795: video_decode:0:qb RIL:12
485171.027: video_decode:0:EmptyThisBuffer(3e4ef140,1387,0->130)
485171.038: video_decode:0: queueBuffer (input : 1387)
485171.047: video_decode:0:qb RIL:13
485217.562: video_decode:0:EmptyThisBuffer(3e4ef1c0,11875,0->130)
485217.573: video_decode:0: queueBuffer (input : 11875)
485217.583: video_decode:0:qb RIL:14
485249.589: video_decode:0:EmptyThisBuffer(3e4ef240,4999,0->130)
485249.599: video_decode:0: queueBuffer (input : 4999)
485249.607: video_decode:0:qb RIL:15
485266.427: video_decode:0:EmptyThisBuffer(3e4ef2c0,3329,0->130)
485266.438: video_decode:0: queueBuffer (input : 3329)
485266.447: video_decode:0:qb RIL:16
485274.656: video_decode:0:EmptyThisBuffer(3e4ee940,1477,0->130)
485274.666: video_decode:0: queueBuffer (input : 1477)
485274.675: video_decode:0:qb RIL:17
485305.839: video_decode:0:EmptyThisBuffer(3e4ee9c0,1599,0->130)
485305.850: video_decode:0: queueBuffer (input : 1599)
485305.866: video_decode:0:qb RIL:18
485322.454: video_decode:0:EmptyThisBuffer(3e4eea40,2843,0->130)
485322.466: video_decode:0: queueBuffer (input : 2843)
485322.475: video_decode:0:qb RIL:19
485330.320: video_decode:0:EmptyThisBuffer(3e4eeac0,1312,0->130)
485330.331: video_decode:0: queueBuffer (input : 1312)
485330.341: video_decode:0:qb RIL:20

minfrin, 2016-11-27:
Can anyone see any reason in these logs for the GPU to stop returning callbacks from the decoder?

6by9, 2016-11-28:
```video_decode:3:RIL:resolution changing``` is the codec reports that the stream has changed format, ie there was a new set of header bytes in the stream.
The codec processing will stop until you disable the output port and reconfigure it for the new resolution/format as buffers need reallocating. 

```waiting for a cb structure``` means that you have fed in the maximum number of frames to the input FIFO and it is refusing to take any more, normally because you aren't draining the output (because you haven't dealt with the resolution change). There are 32 callback structures allocated which more than covers most normal situations.

minfrin, 2016-12-07:
> video_decode:3:RIL:resolution changing is the codec reports that the stream has changed format, ie there was a new set of header bytes in the stream.
The codec processing will stop until you disable the output port and reconfigure it for the new resolution/format as buffers need reallocating.

Thanks for your response, this is making more sense.

What event/callback/option do I look out for that will tell me the resolution has changed?

6by9, 2016-12-07:
```OMX_EventPortSettingsChanged``` on the video_decode output port.

minfrin, 2016-12-08:
Looking at the debug logging posted above, and checking the gstreamer code, we're definitely trying to handle OMX_EventPortSettingsChanged.

// First, receive the callback

480967.296: video_decode:0:RIL:resolution changing
480967.312: video_decode:0:cb:callback parameter changed 131,0
480969.791: video_decode:0:GetParameter(2000001)
480969.807: video_decode:0:RIL:GetParameter(3e4e92a4)
480971.138: video_decode:0:GetParameter(2000001)
480971.148: video_decode:0:RIL:GetParameter(3e4e92a4)
480971.841: video_decode:0:GetParameter(2000001)
480971.852: video_decode:0:RIL:GetParameter(3e4e92a4)
484628.792: video_decode:0:GetParameter(2000001)
484628.804: video_decode:0:RIL:GetParameter(3e4e92a4)
484629.332: video_decode:0:GetParameter(2000001)
484629.343: video_decode:0:RIL:GetParameter(3e4e92a4)
484630.243: video_decode:0:GetParameter(2000001)
484630.252: video_decode:0:RIL:GetParameter(3e4e92a4)
484630.840: video_decode:0:GetParameter(2000001)
484630.850: video_decode:0:RIL:GetParameter(3e4e92a4)
484631.296: video_decode:0:GetParameter(2000001)
484631.306: video_decode:0:RIL:GetParameter(3e4e92a4)
484632.037: video_decode:0:GetParameter(2000001)
484632.047: video_decode:0:RIL:GetParameter(3e4e92a4)
484741.619: video_decode:0:GetParameter(2000001)
484741.629: video_decode:0:RIL:GetParameter(3e4e92a4)

// Set the new output format

484742.182: video_decode:0:SetParameter(2000001)
484742.195: video_decode:0:RIL:SetParameter(2000001)
484742.206: video_decode:0:RIL:PortDef, port 131
484742.218: video_decode:0:RIL: set output format 20
484742.629: video_decode:0:GetParameter(2000001)
484742.643: video_decode:0:RIL:GetParameter(3e4e92a4)
484743.844: video_decode:0:GetParameter(2000001)
484743.855: video_decode:0:RIL:GetParameter(3e4e92a4)
484745.303: video_decode:0:GetParameter(2000001)
484745.314: video_decode:0:RIL:GetParameter(3e4e92a4)

// Re-enable the port

484746.220: video_decode:0:SendCommand(3,131,3e451b94) Enable 131
484746.231: video_decode:0:sending cmd 3 to 131
484746.251: video_decode:0:starting command 3

What I'm not seeing is an attempt to disable the port before trying to configure the port. Would you agree that that's what's missing above, or have I misunderstood this?

minfrin, 2016-12-08:
Looking further into the gstreamer code, it looks like there is code to disable the port after receiving OMX_EventPortSettingsChanged - but this code only runs if the port is already enabled.

Looking at the debug log above can you see any reason why the port would be in the state "disabled" directly after the OMX_EventPortSettingsChanged callback?

6by9, 2016-12-09:
The port should be enabled - the firmware won't have magically changed that.
Which version of gst-omx are you running, and is that the version you're looking at the source for?

Checking the code, clearing the flag needs either the output port to be transitioned from disabled to enabled, or a SetParameter(OMX_IndexParamPortDefinition) on the output port, which pretty much has to be done with the port disabled or component dropped back to LOADED.

minfrin, 2016-12-10:
I am currently running git master of gst-omx.

What I have discovered is that when gst_omx_video_dec_reconfigure_output_port() runs, it attempts to gst_video_decoder_set_output_state() to GST_VIDEO_FORMAT_RGBA. Gstreamer then tries to gst_video_decoder_negotiate() downstream twice, first with GST_CAPS_FEATURE_MEMORY_GL_MEMORY (which fails in gstreamer, omxh264dec supports GL_MEMORY, while omxh264enc does not), then second with just RGBA - this also fails, because omxh264enc doesn't support RGBA. Having failed to gst_video_decoder_negotiate() we plow on regardless through a no_egl path, which appears to try and unconfigure and disable the port. The rest of the code path in gst_omx_video_dec_reconfigure_output_port() leaves the port disabled and carries on regardless.

We start writing packets to the disabled port, and during this process we receive a downstream request to renegotiate to I420, which is supported. This triggers the OMX_EventPortSettingsChanged callback, which signals gst_omx_video_dec_loop() to reconfigure like this:

    GST_DEBUG_OBJECT (self, "Port settings have changed, updating caps");

    /* Reallocate all buffers */
    if (acq_return == GST_OMX_ACQUIRE_BUFFER_RECONFIGURE
        && gst_omx_port_is_enabled (port)) {
      err = gst_omx_port_set_enabled (port, FALSE);
      if (err != OMX_ErrorNone)
        goto reconfigure_error;

      err = gst_omx_port_wait_buffers_released (port, 5 * GST_SECOND);
      if (err != OMX_ErrorNone)
        goto reconfigure_error;

      err = gst_omx_video_dec_deallocate_output_buffers (self);
      if (err != OMX_ErrorNone)
        goto reconfigure_error;

      err = gst_omx_port_wait_enabled (port, 1 * GST_SECOND);
      if (err != OMX_ErrorNone)
        goto reconfigure_error;
    }

    if (acq_return == GST_OMX_ACQUIRE_BUFFER_RECONFIGURE) {
      /* We have the possibility to reconfigure everything now */
      err = gst_omx_video_dec_reconfigure_output_port (self);
      if (err != OMX_ErrorNone)
        goto reconfigure_error;
    } else {

Because we are in the disabled state, the "Reallocate all buffers" section doesn't run - we plow into gst_omx_video_dec_reconfigure_output_port() again, but this time we've hung before trying to actually reconfigure the port.

I am still trying to get my head around where in this code we're going wrong. It seems we can't call gst_omx_video_dec_reconfigure_output_port(), have it fail to negotiate, then call gst_omx_video_dec_reconfigure_output_port() again without a hang occurring.

JamesH65, 2017-12-21:
@minfrin Is there anything to add to this issue? Can it be closed?

closed 2018-06-29 by JamesH65