raspberrypi/firmware #688: Build details

N1N, 2016-11-28:
Hi,

I am using Rpi2 Board.

1) What is the build/compilation procedure for firmware-master/ source code?
2) I have UEFI Source and firmware source. 
When i run 'BuildPi2Board.bat' in UEFI, i will get a Kenrel.img file under build directory. And we have one more kernel.img in the firmware under firmware-master\boot\.

What is the difference between those 2 kernel.img files? Finally which one i need to copy to my SD card?

Regards,
NaN

pelwell, 2016-11-28:
1. The kernel build process is described here: https://www.raspberrypi.org/documentation/linux/kernel/building.md
It also compiles the Device Tree blobs and overlays.
  You can't build the remainder of the firmware (start.elf, bootcode.bin) because neither the tools nor source code are public.

2. The UEFI build instructions for the Pi2 are here: https://github.com/ms-iot/RPi-UEFI/blob/ms-iot/README.md
  The resulting kernel.img is treated by the firmware as a replacement OS, even though it is just another boot loader. UEFI has its own rules about where to put the real kernel image.

Be aware that a Pi2/Pi3 will try to load kernel7.img first...

N1N, 2016-11-28:
Hi pelwell,

Thanks for the reply! 

What does this start.elf and bootcode.bin consists of? 

What is the role of bootcode.bin and start.elf? Which one will be loaded first, bootcode.bin or kernel.img?

I just want to know,What is the boot sequence?

Regards,
NaN

pelwell, 2016-11-28:
The sequence is:

1) Boot ROM (baked into the chip - it does the minimum required to load the next image by a variety of methods into the L2 cache - SDRAM hasn't been initialised yet)
2) bootcode.bin (sets up the SDRAM, locates and loads the next loader)
3) start.elf (reads config.txt, the "kernel" and the Device Tree blob)
4) kernel{,7,8}.img (either your OS or the next loader)

And if this is a U-Boot or UEFI system then you can add:

5) OS kernel

It gets even more complicated in a NOOBS system, where steps 1-3 or -4 are run twice with a reset in-between.

N1N, 2016-11-28:
Thanks for the quick response pelwell ! It helps me alot!!

Ruffio, 2016-11-28:
@N1N please close this 'issue' if it has been resolved. Thanks.

closed 2016-11-28 by N1N