raspberrypi/firmware #692: dtoverlay loader truncates parameter string to 80 characters and drops parameters

nbonniere, 2016-12-04:
- /boot/config.txt  contains:
- dtoverlay=ads7846,cs=1,penirq=25,penirq_pull=2,speed=50000,swapxy=1,xmin=400,ymin=300,xmax=3900,ymax=3900

- sudo vcdbg log msg

001619.094: Loaded overlay 'ads7846'
001619.109: dtparam: cs=1
001620.083: dtparam: penirq=25
001622.316: dtparam: penirq_pull=2
001623.083: dtparam: speed=50000
001624.085: dtparam: swapxy=1
001625.269: dtparam: xmin=400
001626.324: dtparam: ym=true
001629.758: Unknown dtparam 'ym' - ignored

- It seems that the string containing the parameters gets truncate after 80 characters by the overlay loader.
- the parameters are needed. How do you get around this problem ?

pelwell, 2016-12-04:
Put them as dtparam directives on subsequent lines. The previous overlay stays in scope until the next one is loaded, and it will be searched for parameters before trying the base .dtb.

pelwell, 2016-12-04:
See section 3.2 of the documentation: https://www.raspberrypi.org/documentation/configuration/device-tree.md

nbonniere, 2016-12-05:
Thank you so much for taking the time to reply.

I relied on published set-up procedure for LCD touchscreens.

All the information I found for LCD touchscreen are all the same and put everything on one line. A lot of people have got his wrong obviously.

Thank you for solving my problem.

Regards,

Nick

--------------------------------------------------------------------------------
From: Phil Elwell 
Sent: Sunday, December 04, 2016 5:40 PM
To: raspberrypi/firmware 
Cc: nbonniere ; Author 
Subject: Re: [raspberrypi/firmware] dtoverlay loader truncates parameter string to 80 characters and drops parameters (#692)

See section 3.2 of the documentation: https://www.raspberrypi.org/documentation/configuration/device-tree.md

Ruffio, 2016-12-06:
@nbonniere if this issue is solved, then please close this issue. Thanks...

closed 2016-12-06 by nbonniere