raspberrypi/firmware #694: No firmware support for linux kernel address randomization

Electron752, 2016-12-06:
I noticed that some of the more recent linux kernels have support for randomizing the address of the linux kernel. Since people are sort of considering the RPI for server applications, I wonder if this would be a good feature to add to the firmware.

Here is the text from menuconfig. All it takes is passing a random values in the DT which could be as simple as combining the board serial number with a few power voltage samples.

CONFIG_RANDOMIZE_BASE:

Randomizes the virtual address at which the kernel image is
loaded, as a security feature that deters exploit attempts
relying on knowledge of the location of kernel internals.

It is the bootloader's job to provide entropy, by passing a
random u64 value in /chosen/kaslr-seed at kernel entry

popcornmix, 2016-12-06:
As far as I can tell this is only an option on arm64 from 4.6 and later, and is still not an option for arm even on 4.9 tree, so it's not something we can make use of with default kernels.

However the firmware does have access to a hardware random number generator, so it could populate ``/chosen/kaslr-seed`` if that is all you need.

pelwell, 2016-12-06:
At the cost of an increased boot time.

Electron752, 2016-12-06:
I saw an article on the www.raspberrypi.org page that described a 64 bit kernel being used in a server distro.   And I think this is by far one of the easiest ways to protect against random attacks from say the Internet.  

Yes, I think all it would need would be to populate /chosen/kaslr-seed.  And no I would not expect this to work on default kernels since they are 32 bit only.

Another option would be to insert some RPI specific ASM code into the startup route to generate the random number before the MMU is turned on.  Or possibly use something like u-boot with chained loaders.

If boot time is a concern, perhaps the firmware should only do this if it is going to load a kernel in 64 bit mode, but of course that has the disadvantage that 64 bit isn't really supported at this time.

popcornmix, 2016-12-06:
Yes, Phil is correct. It currently takes about a second between enabling and reading the first random number, so we don't want to enable the naive solution by default.
Possibly we can try to enable earlier (maybe even in bootcode.bin) to reduce this time.
The other option is a config.txt option to enable it.

Electron752, 2016-12-06:
The config.txt option seems reasonable.

Also, I suspect the firmware has access to a A/D converter to monitor the power line voltages.  One simple solution that wouldn't add much to boot time would be to hash the board serial number with a few voltage samples.   That wouldn't be a great random number, but it would be better then nothing.  Also it would prevent all RPIs from being identical since the serial number or mac address would need to be guessed by an attacker.  I know guessing the mac is trivial with IPv6, but it isn't with IPv4 and most servers are going to at least be behind some kind of firewall with either port forwarding or in the DMZ.

popcornmix, 2016-12-06:
There is no A/D converter. We get a single GPIO line from a voltage comparator with 4.63V for the under-voltage detection.

Electron752, 2016-12-06:
That's a bummer...

Perhaps we should keep this as a open issue for a bit while I ponder some other ways to make this work.  Like I said, the worst case would be to modify u-boot to generate this and used chained loaders.

I was partially against requiring u-boot from the beginning since IMHO u-boot has some issues and I wanted the option to keep things simple.

Electron752, 2016-12-06:
Looking at the kernel source code, the startup environment is setup enough before the randomization point to call C code.   I'm also seeing that the random number generator is very easy to use.

I think I'm going to make an attempt to prototype this by modifying the kernel startup code to generate the random number.

popcornmix, 2016-12-06:
> I think I'm going to make an attempt to prototype this by modifying the kernel startup code to generate the random number.

I think if we do support this downstream, then using the ``/chosen/kaslr-seed`` would be preferable (it feels like the right solution).
I have locally tested enabling the RNG hardware in bootcode.bin, before start.elf is loaded which means it can be used in start.elf without the 1 second delay.
We also may speed up kernel booting a little with https://github.com/raspberrypi/linux/commit/0d9be57ab1ee960d8a2d796014a71620da62f9c8

Electron752, 2016-12-06:
I agree that /chosen/kaslr-seed is the best option.   I'm just giving alternatives.

I've looked at the kernel source that does the magic, and I can locally test it by hard coding various values.   If/when you have a firmware that sets /chosen/kaslr-seed, I'm more then willing to test it.

Sorry for the delay in replying. I was busy with something else.

Ruffio, 2016-12-30:
@popcornmix this sounds good. Is is going to be implemented soon?

Electron752, 2017-01-13:
This has been active for awhile.

Does this have any chance of being implemented in the firmware, or should I be looking at other options?

Two other options are to either add custom code in the kernel startup, or use u-boot or similar as a second stage loader.

popcornmix, 2017-01-13:
We not unit the HW RNG earlier (bootcode.bin) so I believe we can populate this without a delay.
Let me have a look.

popcornmix, 2017-01-13:
Can you try this firmware?
https://drive.google.com/uc?id=0B-6zmEDJwxZEMkNCNlVEMUJvdDg&export=download

It should create the ``/chosen/kaslr-seed`` dt node.

Electron752, 2017-01-13:
Sure will.  Thanks for getting it so fast.

Electron752, 2017-01-13:
OK, cool.
I see kaslr-seed under /proc/device-tree/chosen and it appears to be 8 bytes of random data.

I need to rebuild my kernel with randomization enabled and do some testing.

Electron752, 2017-01-13:
Awsome! This is rather cool.   I did multiple boots and I can see that the virtual kernel memory layout is different on each boot.

I'll need to update the config files for 64 bit, but other then that I don't think it's going to require any other kernel changes.

BOOT 1:
[    0.000000] Virtual kernel memory layout:
[    0.000000]     modules : 0xffffff8000000000 - 0xffffff8008000000   (   128 MB)
[    0.000000]     vmalloc : 0xffffff8008000000 - 0xffffffbebfff0000   (   250 GB)
[    0.000000]       .text : 0xffffff856c280000 - 0xffffff856c810000   (  5696 KB)
[    0.000000]     .rodata : 0xffffff856c810000 - 0xffffff856ca20000   (  2112 KB)
[    0.000000]       .init : 0xffffff856ca20000 - 0xffffff856cb70000   (  1344 KB)
[    0.000000]       .data : 0xffffff856cb70000 - 0xffffff856cbcc808   (   371 KB)
[    0.000000]        .bss : 0xffffff856cbcc808 - 0xffffff856ccc41fc   (   991 KB)
[    0.000000]     fixed   : 0xffffffbefe7fd000 - 0xffffffbefec00000   (  4108 KB)
[    0.000000]     PCI I/O : 0xffffffbefee00000 - 0xffffffbeffe00000   (    16 MB)
[    0.000000]     vmemmap : 0xffffffbf00000000 - 0xffffffc000000000   (     4 GB maximum)
[    0.000000]               0xffffffbf86000000 - 0xffffffbf86ec0000   (    14 MB actual)
[    0.000000]     memory  : 0xffffffe180000000 - 0xffffffe1bb000000   (   944 MB)

BOOT 2:
[    0.000000] Virtual kernel memory layout:
[    0.000000]     modules : 0xffffff8000000000 - 0xffffff8008000000   (   128 MB)
[    0.000000]     vmalloc : 0xffffff8008000000 - 0xffffffbebfff0000   (   250 GB)
[    0.000000]       .text : 0xffffff93a7680000 - 0xffffff93a7c10000   (  5696 KB)
[    0.000000]     .rodata : 0xffffff93a7c10000 - 0xffffff93a7e20000   (  2112 KB)
[    0.000000]       .init : 0xffffff93a7e20000 - 0xffffff93a7f70000   (  1344 KB)
[    0.000000]       .data : 0xffffff93a7f70000 - 0xffffff93a7fcc808   (   371 KB)
[    0.000000]        .bss : 0xffffff93a7fcc808 - 0xffffff93a80c41fc   (   991 KB)
[    0.000000]     fixed   : 0xffffffbefe7fd000 - 0xffffffbefec00000   (  4108 KB)
[    0.000000]     PCI I/O : 0xffffffbefee00000 - 0xffffffbeffe00000   (    16 MB)
[    0.000000]     vmemmap : 0xffffffbf00000000 - 0xffffffc000000000   (     4 GB maximum)
[    0.000000]               0xffffffbf3c000000 - 0xffffffbf3cec0000   (    14 MB actual)
[    0.000000]     memory  : 0xffffffcf00000000 - 0xffffffcf3b000000   (   944 MB)

BOOT 3:
[    0.000000] Virtual kernel memory layout:
[    0.000000]     modules : 0xffffff8000000000 - 0xffffff8008000000   (   128 MB)
[    0.000000]     vmalloc : 0xffffff8008000000 - 0xffffffbebfff0000   (   250 GB)
[    0.000000]       .text : 0xffffff92d7880000 - 0xffffff92d7e10000   (  5696 KB)
[    0.000000]     .rodata : 0xffffff92d7e10000 - 0xffffff92d8020000   (  2112 KB)
[    0.000000]       .init : 0xffffff92d8020000 - 0xffffff92d8170000   (  1344 KB)
[    0.000000]       .data : 0xffffff92d8170000 - 0xffffff92d81cc808   (   371 KB)
[    0.000000]        .bss : 0xffffff92d81cc808 - 0xffffff92d82c41fc   (   991 KB)
[    0.000000]     fixed   : 0xffffffbefe7fd000 - 0xffffffbefec00000   (  4108 KB)
[    0.000000]     PCI I/O : 0xffffffbefee00000 - 0xffffffbeffe00000   (    16 MB)
[    0.000000]     vmemmap : 0xffffffbf00000000 - 0xffffffc000000000   (     4 GB maximum)
[    0.000000]               0xffffffbfd5000000 - 0xffffffbfd5ec0000   (    14 MB actual)
[    0.000000]     memory  : 0xfffffff540000000 - 0xfffffff57b000000   (   944 MB)

Electron752, 2017-01-13:
@popcornmix 
Any idea when this can be added to the firmware distributed on github?

popcornmix, 2017-01-13:
It will be in next update.
Should be within the week.

Electron752, 2017-01-13:
Awesome!!!   I'll keep the issue open until the next update appears on github, then I'll close the issue.

I want to do some testing with both the old and new firmware with randomization enabled in the kernel, then I submit a request for changing the 64-bit build configs.  Looking at the kernel code, enabling the feature will allow the kernel to still function even on old firmware.   I even see a command line option for disabling randomization for debugging purposes.

Thanks.

Electron752, 2017-01-15:
I confirmed that the latest github drop works, so I'm closing the issue.

closed 2017-01-15 by Electron752