raspberrypi/firmware #695: Mapping MMAL video surfaces as GL textures?

wm4, 2016-12-08:
The RPI firmware seems to be support for mapping video surfaces as textures with eglCreateImageKHR/glEGLImageTargetTexture2DOES. I'm fairly sure there used to be examples around for doing this with OMX. There is evidence that this works with MMAL too: https://github.com/raspberrypi/userland/blob/master/host_applications/android/apps/vidtex/vidtex.c 

When someone (@wang-bin) tried to reimplement this with ffmpeg's MMAL decoder, it apparently failed with OUT_OF_MEMORY. Is there any special configuration needed to make this work?

What are the performance characteristics of using this? Worse than the native MMAL video overlay and using an extra dispmanx layer to render OSD/subtitles over it?

(I think I asked this a long time ago, but maybe the situation has changed.)

popcornmix, 2016-12-08:
Yes, every video frame will need to be converted from YUV format to an RGBA format which is expensive in both GPU cycles and sdram bandwidth.
We went down this route in kodi a while back, but switched to rendering the YUV layer directly (in addition to a separate RGB GL framebuffer) as performance wasn't sufficient for 1080p (720p is okay going through GL).

6by9, 2016-12-08:
> performance wasn't sufficient for 1080p (720p is okay going through GL).

Don't tell anyone, but on some projects where we needed to video decode into GL objects, anything above 720P would get downscaled by a factor of 2 in each direction during the conversion to reduce the processing cost. No one ever noticed :-o Mainly it was used for rotation transition effects under Android, and it reverted to a YUV overlay as soon as the transition was complete.

popcornmix, 2016-12-08:
OMX example code is here: https://github.com/raspberrypi/userland/tree/master/host_applications/linux/apps/hello_pi/hello_teapot
In theory MMAL should do this, but there may be some missing plumbing if no one has tried using it before.
But the performance issue will be the same whether using MMAL or OMX.

wang-bin, 2016-12-12:
From vidtex example `MMAL_BUFFER_HEADER_T.data` looks like this: 0x77, 0x15, 0x63, 0x34..., while from ffmpeg mmal decoder it's: 0x133ae000, 0x133aea00, 0x133af500.... Why the results are different?

6by9, 2016-12-12:
Have you looked at the encoding used on the two totally different apps?

Looking at the FFMPEG output, I'd say that was RGBA, with the alpha always at 0.

vidtex sets the port encoding format to MMAL_ENCODING_OPAQUE at https://github.com/raspberrypi/userland/blob/master/host_applications/android/apps/vidtex/svp.c#L241, so unless you're looking at some other buffer, it contains an opaque reference to images generated from the decoder - you can't decode it on the ARM.

wm4, 2016-12-12:
FFmpeg also outputs MMAL_ENCODING_OPAQUE if you request the AV_PIX_FMT_MMAL pixel format: http://git.videolan.org/?p=ffmpeg.git;a=blob;f=libavcodec/mmaldec.c;h=3f92bf5ec9bef0ecfb2eba768c074d14a66157dd;hb=HEAD#l297

ni1ight, 2017-06-08:
Is this thread still alive? Did anybody manage to use this stuff properly?

JamesH65, 2017-12-21:
I suspect this issue is better dealt with on the forum, is seems dead from a lack of activity anyway. 

Closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

closed 2017-12-21 by JamesH65