raspberrypi/firmware #696: VC audio service locks up

gonzoua, 2016-12-10:
I am working on improving VCHI audio driver for FreeBSD and while debugging I came across what I believe is a firmware issue. Firmware gets into weird state: audio plays fine for couple hours then kernel receives VC_AUDIO_MSG_TYPE_COMPLETE callback with error bit set (bit 30) and some invalid data in count part. After that no audio can go through, any submit operation ends with the same callback: error bit set. 

Driver works like this:
- On initialization it connects to audio service and passes VC_AUDIO_MSG_TYPE_OPEN message
- When playback starts it passes VC_AUDIO_MSG_TYPE_START
- Samples are submitted in 4000 bytes packets, and maximum 32000 bytes are allowed to be buffered on VC side
- When playback stops VC_AUDIO_MSG_TYPE_STOP message is sent to VC

I use headphones output and my test is mpg123 playing bunch of mp3s. mpg123 does not issue stop/start between files, so essentially it's: start playback, submit audio samples for 2 hours, firmware audio service locks up

There is no vcdbg for FreeBSD so I can't check VC logs or examine its state in any other way. Any help with troubleshooting this issue would be appreciated.

gonzoua, 2016-12-12:
This seems to be a problem only with PWM audio. HDMI audio didn't stuck for almost two days. I tested it with destination set to HDMI but without actual hardware connected. I do not have audio-enabled HDMI device here. I will try to tun the test with HDMI audio device too.

gonzoua, 2016-12-15:
Thanks for explanation of bit 30. 

When preparing image I realized that I used old firmware binaries for test. After updating to the latest master I was able to play audio for more than 20 hours so far. I'll run the test over weekend to make sure that this lock-up is indeed fixed. I noticed several cases of 1-second or so pauses in playback though. They were not caused by samples starvation because VC never reported bit 30 in callback.

Although the issue doesn't seem to be in latest fw I'd still like to get vcdbg running on FreeBSD if possible, that would give better visibility of what's happening on VC side and probably make debugging easier. Is there sources for vcdbg or do you need image with build environment to prepare binaries? Most of userland VC tools should be compilable on FreeBSD as-is or with minor changes.

JamesH65, 2017-12-21:
Closing this issue as questions answered/issue resolved.

closed 2017-12-21 by JamesH65