raspberrypi/firmware #697: OMX: OMX_COLOR_FormatYUV420PackedSemiPlanar - does it map to GST_VIDEO_FORMAT_NV12?

minfrin, 2016-12-12:
I am trying to add support for OMX_COLOR_FormatYUV420PackedSemiPlanar to gstreamer's gst-omx.

Does OMX_COLOR_FormatYUV420PackedSemiPlanar map to gstreamer's GST_VIDEO_FORMAT_NV12?

If not, can you confirm which gstreamer video format this would correspond to?

https://gstreamer.freedesktop.org/data/doc/gstreamer/head/gst-plugins-base-libs/html/gst-plugins-base-libs-gstvideo.html

6by9, 2016-12-12:
Yes, OMX_COLOR_FormatYUV420PackedSemiPlanar is NV12.
It's already mapped in gst-omx - https://github.com/GStreamer/gst-omx/blob/master/omx/gstomxvideo.c#47

minfrin, 2016-12-12:
I came up with a patch for git master as follows:

https://bugzilla.gnome.org/show_bug.cgi?id=775959

Can you confirm the patch is correct? There was a question in that bug report as follows:

> This sounds to me like the Packed version is not actually planar but indeed packed. So
> not the same as NV12. Can you confirm that it is the same on the RPi? If so, we should
> probably add some configuration to gstomx.conf, some new hack flag, that maps the
> Packed*Planar things to the same as the non-Packed*Planar.

minfrin, 2016-12-12:
OMX_COLOR_FormatYUV420SemiPlanar is supported at https://github.com/GStreamer/gst-omx/blob/master/omx/gstomxvideo.c#47.

OMX_COLOR_FormatYUV420PackedSemiPlanar is not yet supported by gstreamer.

6by9, 2016-12-12:
My bad, I'd missed that it wasn't the packed flavour.

The Packed flavours only make a difference if nSliceHeight < nFrameHeight. They dictate that each slice will be complete in having all the planes in the normal order.
eg nFrameWidth=640, nFrameHeight=480, nStride=640, nSliceHeight=16, will have buffers of size 640*16*1.5 = 15360 bytes, with 640*16  samples of Y, followed by 320*8 samples of UV interleaved. You'd get 30 buffers to make up a full frame.
Each buffer can be considered to be NV12, but you can't simply concatenate them to make a complete NV12 image.

TBH I'm not 100% clear on what should be delivered under those situations with OMX_COLOR_FormatYUV420SemiPlanar (not packed). From reading the spec I'd say that that is not permitted, or at least undefined.

They've already accepted that OMX_COLOR_FormatYUV420Planar is synonymous with OMX_COLOR_FormatYUV420PackedPlanar as equivalent to I420 for their use case, so I'm not sure why the debate over SemiPlanar.

minfrin, 2016-12-13:
The gstreamer guys applied the update, thanks for this.

https://bugzilla.gnome.org/show_bug.cgi?id=775959

closed 2016-12-13 by minfrin