raspberrypi/firmware #698: OMX_BUFFERFLAG_CODECCONFIG across multiple split buffers

minfrin, 2016-12-12:
I am currently getting to the bottom of a negotiation problem with gstreamer, triggered by what seems to be the splitting of OMX_BUFFERFLAG_CODECCONFIG over multiple buffers:

0:06:56.111636218 24259 0x73fe0920 ERROR             omxh264enc gstomxh264enc.c:601:gst_omx_h264_enc_handle_output_frame:<omxh264enc-omxh264enc0> OMX_BUFFERFLAG_CODECCONFIG: length 19, bytes 00000001
0:06:56.112307095 24259 0x73fe0920 ERROR             omxh264enc gstomxh264enc.c:601:gst_omx_h264_enc_handle_output_frame:<omxh264enc-omxh264enc0> OMX_BUFFERFLAG_CODECCONFIG: length 9, bytes 00000001
0:06:56.131532481 24259 0x73fe0920 ERROR             omxh264enc gstomxh264enc.c:601:gst_omx_h264_enc_handle_output_frame:<omxh264enc-omxh264enc0> OMX_BUFFERFLAG_CODECCONFIG: length 12, bytes 00000001
0:06:56.132180754 24259 0x73fe0920 ERROR             omxh264enc gstomxh264enc.c:601:gst_omx_h264_enc_handle_output_frame:<omxh264enc-omxh264enc0> OMX_BUFFERFLAG_CODECCONFIG: length 7, bytes 09361803

We usually receive a 9 byte OMX_BUFFERFLAG_CODECCONFIG, then 19 bytes, then 9, and so on. Then suddenly we receive a 12 byte buffer followed by a 7 byte buffer instead of the 19 byte buffer.

Is it normal for the RPi to be sending OMX_BUFFERFLAG_CODECCONFIG buffer split? Do we need to implement some kind of reassembly logic, or is there a flag that keeps OMX_BUFFERFLAG_CODECCONFIG buffers together?

https://bugzilla.gnome.org/show_bug.cgi?id=775962

6by9, 2016-12-12:
Presumably you've got inline headers enabled for whatever reason, otherwise you wouldn't be getting headers in the middle of the stream.

The only situation I can think of is that internally the codec has a fixed sized buffer to write to. At the wrap point in that buffer it will spit out the frame as 2 fragments. That is permitted within the IL spec as long as the nFlags field is set appropriately (OMX_BUFFERFLAG_ENDOFFRAME set only on the second buffer). It's a shame you haven't logged nFlags as that would confirm.
To hit that wrap point reliably with 12 and 7 bytes of headers sounds pretty unlikely, but possible.

There is a parameter OMX_IndexConfigMinimiseFragmentation to reassemble those frames where possible. I hadn't enabled by default as it was a moderate risk in changing the behaviour like that, however the V4L2 camera driver has been using it like that for a good while now so it's probably worth changing the default if it makes life that much easier for GStreamer.
@popcornmix would you be happy with that change?

popcornmix, 2016-12-12:
@6by9 no objections if you think it is best.

6by9, 2016-12-12:
I'd like to get confirmation from @minfrin first that the flags are as I expect, but I'll sort a patch on gitlab.
It's changing a single default value, and it's pretty well tested now via V4L2.

Regression risk is very low as it's actually making life easier for the client.

minfrin, 2016-12-13:
Here are the flags along with the frames:

0:19:19.853439083 27785 0x73f8ef80 ERROR             omxh264enc gstomxh264enc.c:601:gst_omx_h264_enc_handle_output_frame:<omxh264enc-omxh264enc0> OMX_BUFFERFLAG_CODECCONFIG: length 19, bytes 00000001, flags 00000480
0:19:19.854463394 27785 0x73f8ef80 ERROR             omxh264enc gstomxh264enc.c:601:gst_omx_h264_enc_handle_output_frame:<omxh264enc-omxh264enc0> OMX_BUFFERFLAG_CODECCONFIG: length 9, bytes 00000001, flags 00000490
0:19:19.935048490 27785 0x73f8ef80 ERROR             omxh264enc gstomxh264enc.c:601:gst_omx_h264_enc_handle_output_frame:<omxh264enc-omxh264enc0> OMX_BUFFERFLAG_CODECCONFIG: length 19, bytes 00000001, flags 00000480
0:19:19.938798082 27785 0x73f8ef80 ERROR             omxh264enc gstomxh264enc.c:601:gst_omx_h264_enc_handle_output_frame:<omxh264enc-omxh264enc0> OMX_BUFFERFLAG_CODECCONFIG: length 9, bytes 00000001, flags 00000490
0:19:19.959799450 27785 0x73f8ef80 ERROR             omxh264enc gstomxh264enc.c:601:gst_omx_h264_enc_handle_output_frame:<omxh264enc-omxh264enc0> OMX_BUFFERFLAG_CODECCONFIG: length 19, bytes 00000001, flags 00000480
0:19:19.960788136 27785 0x73f8ef80 ERROR             omxh264enc gstomxh264enc.c:601:gst_omx_h264_enc_handle_output_frame:<omxh264enc-omxh264enc0> OMX_BUFFERFLAG_CODECCONFIG: length 9, bytes 00000001, flags 00000490
0:19:19.965270740 27785 0x73f8ef80 ERROR             omxh264enc gstomxh264enc.c:601:gst_omx_h264_enc_handle_output_frame:<omxh264enc-omxh264enc0> OMX_BUFFERFLAG_CODECCONFIG: length 19, bytes 00000001, flags 00000480
0:19:19.966214895 27785 0x73f8ef80 ERROR             omxh264enc gstomxh264enc.c:601:gst_omx_h264_enc_handle_output_frame:<omxh264enc-omxh264enc0> OMX_BUFFERFLAG_CODECCONFIG: length 9, bytes 00000001, flags 00000490
0:19:20.000193351 27785 0x73f8ef80 ERROR             omxh264enc gstomxh264enc.c:601:gst_omx_h264_enc_handle_output_frame:<omxh264enc-omxh264enc0> OMX_BUFFERFLAG_CODECCONFIG: length 19, bytes 00000001, flags 00000480
0:19:20.006958376 27785 0x73f8ef80 ERROR             omxh264enc gstomxh264enc.c:601:gst_omx_h264_enc_handle_output_frame:<omxh264enc-omxh264enc0> OMX_BUFFERFLAG_CODECCONFIG: length 9, bytes 00000001, flags 00000490
0:19:20.019422709 27785 0x73f8ef80 ERROR             omxh264enc gstomxh264enc.c:601:gst_omx_h264_enc_handle_output_frame:<omxh264enc-omxh264enc0> OMX_BUFFERFLAG_CODECCONFIG: length 7, bytes 00000001, flags 00000480
0:19:20.025249724 27785 0x73f8ef80 ERROR             omxh264enc gstomxh264enc.c:601:gst_omx_h264_enc_handle_output_frame:<omxh264enc-omxh264enc0> OMX_BUFFERFLAG_CODECCONFIG: length 12, bytes 28ac2b40, flags 00000480

6by9, 2016-12-13:
Thanks.
I'm fairly certain that you are triggering the end of buffer wrap, although I'm surprised it is so repeatable as 12 & 7 bytes. You also seem to be running with a very low intra-I-period, and inline headers, but that is valid if quirky.

I believe your logs should be interpreted as:
```
0:19:20.000193351 27785 0x73f8ef80 ERROR             omxh264enc gstomxh264enc.c:601:gst_omx_h264_enc_handle_output_frame:<omxh264enc-omxh264enc0> OMX_BUFFERFLAG_CODECCONFIG: length 19, bytes 00000001, flags 00000480
0:19:20.006958376 27785 0x73f8ef80 ERROR             omxh264enc gstomxh264enc.c:601:gst_omx_h264_enc_handle_output_frame:<omxh264enc-omxh264enc0> OMX_BUFFERFLAG_CODECCONFIG: length 9, bytes 00000001, flags 00000490
<encoded data buffer(s)>
0:19:20.019422709 27785 0x73f8ef80 ERROR             omxh264enc gstomxh264enc.c:601:gst_omx_h264_enc_handle_output_frame:<omxh264enc-omxh264enc0> OMX_BUFFERFLAG_CODECCONFIG: length 7, bytes 00000001, flags 00000480
0:19:20.025249724 27785 0x73f8ef80 ERROR             omxh264enc gstomxh264enc.c:601:gst_omx_h264_enc_handle_output_frame:<omxh264enc-omxh264enc0> OMX_BUFFERFLAG_CODECCONFIG: length 12, bytes 28ac2b40, flags 00000480
```
and I would expect another 9 byte buffer with flags 0x490 following.
19 bytes of SPS, and then 9 bytes of PPS make up the end of frame (OMX_BUFFERFLAG_ENDOFFRAME = 0x10).
Each of them have OMX_BUFFERFLAG_CODECCONFIG (0x80) set as they are header bytes. OMX_BUFFERFLAG_ENDOFNAL (0x400) is a bit of a curious one as I'd only expect it on the 12 and 9 byte buffers, but it's a proprietary extension and I'd need to check the code to see how it is being signaled.

If you want to test ahead of me sorting a patched firmware, then something along the lines of 
```
OMX_CONFIG_BOOLEANTYPE b;
GST_OMX_INIT_STRUCT (&b);
b.bEnabled = OMX_TRUE;
gst_omx_component_set_parameter (comp,
        OMX_IndexConfigMinimiseFragmentation, &b);
```
should have the same effect. It'd be interesting to know that that solves the problem.
I suspect it will result in you getting a single 28 byte header byte buffer as it accumulates data until it needs to signal OMX_BUFFERFLAG_ENDOFFRAME, or nFilledLen == nAllocLen.

6by9, 2016-12-15:
I'm curious now as to how come you get 19 and 9 bytes. The codec appears to be sending out 18 and 9 bytes.
I have confirmed that setting the MinimiseFragmentation parameter (either by default or manually) produces a single combined header bytes buffer of 27 bytes. Could you confirm if GStreamer is happy getting SPS and PPS in the one buffer please?

popcornmix, 2016-12-19:
@minfrin this test firmware:
https://www.dropbox.com/s/qc34rhnydgoexcu/firmware_ipp.zip?dl=0

defaults MinimiseFragmentation to true. Are you able to test to confirm that resolves your issue?

6by9, 2017-01-23:
@minfrin Please confirm if the change helped at all.
The change actually got merged to the mainline firmware but has been flagged up as possibly causing issues with MJPEG. In the absence of any input from you our simplest approach is to revert it. If you wish this to be progressed any further then we need some feedback.

minfrin, 2017-01-23:
The h264 encoder broke in my test environment - I suspect this change might have caused the breakage.

What this breakage did do is make an unrelated sporadic gstreamer bug happen all the time, and so I've stopped everything in an effort to find and fix that problem while the problem can be reproduced on every invocation.

Bugs bugs bugs.

JamesH65, 2017-12-21:
@minfrin Is there a status update on this? Is everything now miraculously working? Can this issue be closed?

closed 2018-06-29 by JamesH65