raspberrypi/firmware #700: MMAL advanced deinterlacing

arppl, 2016-12-13:
Hi,

with the latest firmware files on a RPI2 the MMAL advanced deinterlacing of MPEG2 streams ends up with green/red/blue flickering:

[image]

popcornmix, 2016-12-13:
Can you identify the exact update which caused this. See:
https://github.com/Hexxeh/rpi-firmware/commits/master

If you click on each commit the end of the url contains a git hash. Run
```sudo rpi-update <hash>```
to revert back to that version. Report the first version with this error.

popcornmix, 2016-12-13:
A description of how to reproduce the issue would be useful.
E.g. provide a sample file and omxplayer command to run.

arppl, 2016-12-14:
Hi popcornmix,

here is a sample file: https://www.dropbox.com/s/7ax4biwz2wkyme1/sample.ts?dl=0

I cannot reproduce the behaviour using omxplayer directly on the command line. The --advanced option works fine. How can I switch omxplayer to MMAL Bob or MMAL Advanced deinterlacing like in?

However, if I play this file in kodi directly, the MMAL Advanced deinterlacing option results in the messed up picture as in my initial post. MMAL Bob works fine.

Here is the kodi debug log if it helps: http://pastebin.com/rxZ3st4e

Thanks

popcornmix, 2016-12-14:
Okay I can reproduce.
It works okay if you have the MPEG-2 codec licence.
Something is wrong when software decoded.

arppl, 2016-12-14:
So why is kodi not using the MPG2 hardware decoding?

popcornmix, 2016-12-14:
It does work for me. Have you got MMAL enabled in video acceleration options?

arppl, 2016-12-14:
Oh... yes I do have OMXPlayer and MMAL enabled in kodi settings.
And I also have the MPG2 license installed:
`vcgencmd codec_enabled MPG2
MPG2=enabled`

popcornmix, 2016-12-14:
Any different with omxplayer disabled?
Can you post complete kodi log? You've cut out some of the useful info.

arppl, 2016-12-14:
No differences if both are enabled, OMXPlayer alone or MMAL alone.

Here is a full kodi log. I started playing the sample.ts file with advanced deinterlacing, then switched to MMAL Bob. https://ptpb.pw/hy3p

popcornmix, 2016-12-15:
I can only reproduce an effect like this with firmware from Nov 25 2016 and latest kodi using software decode of video (i.e. omxplayer and MMAL disabled).
This is due to that firmware not supporting YUV420 advanced deinterlace. Updating to firmware from Dec 3 2016 or later fixes that for me.

I suspect I'll need a Kodi 16.1 build to reproduce (I'm on Kodi 17).

arppl, 2016-12-30:
Oups, sorry...

I've updated to Kodi 17, which fixes that and seems to have better deinterlacing compared to 16. I don't know if it's a Kodi or a Firmware thing, I didn't have the time to investigate further :(

popcornmix, 2016-12-30:
Yes, Kod 17 now uses the "advanced" deinterlace for 1080p (previous was only used for SD and a simpler deinterlace was used for HD).
Okay, I'll close for now as I don't believe there is an issue with everything up to date.
Feel free to reopen if you spot a problem

closed 2016-12-30 by popcornmix

Ruffio, 2016-12-30:
@arppl IMHO this issue should be closed as fixed