raspberrypi/firmware #701: Camera annotation issue with size >60

waveform80, 2016-12-15:
There appears to be an issue with annotation text in still captures when the vertical text size exceeds 60. The following test script reproduces the issue for me under Raspbian with firmware revision 930 (latest stable from apt):

```bash
#!/bin/bash

raspistill -w 640 -h 480 -t 2000 -a "one line" -ae 59 -o one_line_59.jpg
raspistill -w 640 -h 480 -t 2000 -a "one line" -ae 60 -o one_line_60.jpg
raspistill -w 640 -h 480 -t 2000 -a "two\nlines" -ae 29 -o two_lines_29.jpg
raspistill -w 640 -h 480 -t 2000 -a "two\nlines" -ae 30 -o two_lines_30.jpg
```

The annotations appear correctly in the preview in all cases, but in the resulting JPEGs will only appear in `one_line_59.jpg` and `two_lines_29.jpg`. In the two other files, the annotation will not appear. Output resolution and aspect ratio appear to make no difference (I only used 640x480 above to keep the test files small), and the annotation appears correctly at all sizes in video recordings from raspivid so this is something specific to the still port.

At greater sizes that the truncated bottom of the annotated text starts to appear in the output; it's as though the bounding box / location for the annotation is being incorrectly calculated somewhere.

For reference the issue also affects picamera, but I'm fairly confident this is something firmware-side.

6by9, 2016-12-15:
I'm not getting anything abnormal happening using https://github.com/Hexxeh/rpi-firmware/commit/a6ddca1723f9122280ded7cbb68581fa60c684bf.
Could you confirm the output from ```vcgencmd version```? I don't use the apt based firmware revisions. 
I'm assuming this is a recent regression you're seeing, or has it happened on other versions?

waveform80, 2016-12-15:
Ah, sorry - it's:

```
 pi@raspberrypi ~ $ sudo vcgencmd version
Nov 25 2016 16:09:34 
Copyright (c) 2012 Broadcom
version 48a26a2ae46c497139b3d5a9c8d15485c7b3bfbc (clean) (release)
```

Incidentally, this issue originally came from Marc at some picademy-like event (https://twitter.com/Coding2Learn/status/809370423892787200) so it'll also be present in whatever firmware they're using on their standard image (which I'm guessing will be something based on a fairly recent Raspbian release).

6by9, 2016-12-15:
As it happens, that's exactly the same revision I was running with https://github.com/Hexxeh/rpi-firmware/commit/a6ddca1723f9122280ded7cbb68581fa60c684bf. I did have some userland changes, but those shouldn't make any difference.

From those Twitter images I have a thought, but can't reproduce the failure at present (and have other changes in progress to confuse matters). I'll have a further play when I get a chance. I suspect it is the automatic line wrap that is causing the issue, but don't know for certain.
FYI Raspistill always runs preview at 1024x768 except with the -fp option (which normally causes display underruns). I know from previous conversations that PiCamera runs preview at whatever resolution is requested for output.

waveform80, 2016-12-16:
> From those Twitter images I have a thought, but can't reproduce the failure at present (and have other changes in progress to confuse matters). I'll have a further play when I get a chance. I suspect it is the automatic line wrap that is causing the issue, but don't know for certain.

I'd be a little surprised if line-wrap was the issue given that the one-line annotation disappears at size 60 too (still appears on the preview as a single line, but there's nothing on the JPEG output), but then I've no idea what the line wrap code actually does so ... maybe?

> FYI Raspistill always runs preview at 1024x768 except with the -fp option (which normally causes display underruns). I know from previous conversations that PiCamera runs preview at whatever resolution is requested for output.

Yup, absolutely true - PiCamera's preview defaults to the resolution configured for camera output (since 1.11 or 1.12 it can be configured separately, but the default remains as the output res). Still, I don't *think* that's relevant in this case given that I see the same issue with both PiCamera (running the preview at output res) and raspistill (running the preview at 1024x768).

I threw together a quick script to run captures with annotations at a variety of sizes and stitch captures from the still and video ports together side by side. Here's a few bits of the output (still capture on the left, video capture on the right; video capture always shows the annotation correctly).

One line of text, size 59:
[image]

One line of text, size 60:
[image]

Two lines of text, size 29:
[image]

Two lines of text, size 30:
[image]

Three lines of text, size 19:
[image]

Three lines of text, size 20:
[image]

And to demonstrate that something funky is going on with the bounding box, here's three lines of text at 29:
[image]

And then at 57:
[image]

And 58:
[image]

So ... something weird at or near text with a vertical height totalling "size 60"?

6by9, 2017-02-07:
Finally getting around to looking at this.
It appears to only affect V1.3 (OV5647) sensors, not V2.1 (IMX219). That explains why I wasn't seeing an issue back in December as my Pi had a V2.1 camera. I happen to have one of each attached to my CM3 so can flick between them easily.

I need to do some more digging, but I'm suspecting that the text is being written too soon, so the ISP then overwrites it. That wouldn't explain the corrupt text posted to Twitter though. Puzzling.....

6by9, 2017-02-08:
Starting to get to the bottom of this - something is wrong in the still denoise processing stage.

For me with raspistill -a "one line" -ae 60 ..., it I've added ```mmal_port_parameter_set(camera->control, MMAL_PARAMETER_STILLS_DENOISE, MMAL_FALSE)```, then the annotation is working correctly. Set it to MMAL_TRUE and the text disappears. 

I'll do a quick test in PiCamera using ```image_denoise=false``` to see if that also sorts the other issue, and then dig into what the algorithm is doing wrong.

waveform80, 2017-02-08:
Ah, very nice! The `image_denoise` workaround certainly works it for me on a V1 module.

6by9, 2017-02-08:
Hmm, using the code from the Twitter image I still get issues even with ```image_denoise=False```.

The staggered effect is probably that the ISP is working in tiles and hasn't finished writing the lines when annotate starts adding to them. Early tiles on a line are complete, but the later ones annotate writes to before the ISP so the text gets overwritten. The pipeline passes down the number of lines supposedly complete, so I'm a little puzzled as to why it is going wrong. More analysis required.

```image_denoise``` is probably just changing the timing slightly as when it is on it requires a minimum of 16 lines of context, and processes in 64 line blocks. 16+64 = 80, which is the number of lines annotate needs to write one line of size 60 text.
I'm also getting raspistill to lose the top line of 3 when using size 100 text. Size 120 went to 4 lines and loses most of it. 
All most confusing.

waveform80, 2017-02-08:
Argh - yes, I only tested the one line example. Just dug out the test script I used above and added image_denoise to it: much the same result as you describe above (issues at size 46 for two lines, 32 for three lines, etc).

6by9, 2017-02-08:
Think I've got it, although it wasn't quite where I thought it was.

The annotation stage wasn't telling the rest of the pipe that it hadn't finished processing lines, so the JPEG encoder was being given the lines before they were annotated. The number of lines that are produced at a time by the ISP varies depending on how the image gets tiled, but typically the first stripe was around 80 lines.
The JPEG block, working a 16x16 macroblock at a time, was therefore racing the ISP and software processing stages to see which could complete their bit first. Disabling stills denoise seemed to work because it sped up the ISP & processing enough to allow annotate to have written the text before JPEG encoder could pick it up for the basic cases.

Correctly signalling that the annotate stage hasn't finished with the lines seems to fix this. I'll sort out a firmware patch.
Did I mention that the annotate stage was only ever used for debug purposes previously?! :-o

waveform80, 2017-02-08:
> Did I mention that the annotate stage was only ever used for debug purposes previously?! :-o

Yup ;)

6by9, 2017-02-08:
Fix released if you do ```sudo rpi-update```.
Don't do this on a critical machine - that is the cutting edge branch so may have issues.

6by9, 2017-10-10:
@waveform80 If you're happy would you close the issue please?

closed 2017-10-10 by waveform80