raspberrypi/firmware #703: Could bootcode decompress start.elf ?

dp111, 2016-12-21:
Reading start.elf from SDCARD can be slow. If it was compressed the decompressor should endup faster than raw reading from SDCARD

ghollingworth, 2016-12-21:
Depends upon what we can do for a decompression engine and whether it'll fit into 9k of remaining memory...

Are you sure it's the start.elf and not the kernel.img that you need accelerating?  Is this for bare metal?

Gordon

dp111, 2016-12-21:
Depends on algorithm but 2K is possible. Kernel.img would be useful too. It is bare metal , I have Kernel down to <400K currently. With some of the changes over the last 6 months to bootcode and start_cd.elf the increase in boot time means I now miss the first packet of data.

dp111, 2016-12-21:
PD SDK here http://www.7-zip.org/sdk.html 
Might be a bit slow

Ferroin, 2016-12-22:
While I don't know how small they can be made, LZ4 and Snappy are the first options that come to mind for me, they get reasonable compression ratios and are insanely fast.  For this to actually be worth it, it needs to be measurably faster than the SD card (which is not hard, but is still important).

I've actually done the math for the Linux kernel's built-in compressed image support (both for the kernel itself, and an initrd), and LZ4 and LZO improve boot times relative to the same amount of data uncompressed, GZIP breaks even, and BZIP2 and XZ actually make boot times worse (just barely, we're talking fractions of a percent with them).  An assembly optimized version of GZIP or BZIP2 might make things better there, but probably wouldn't fit the required space, and would probably not be able to beat LZ4 and LZO.

popcornmix, 2017-01-15:
> With some of the changes over the last 6 months to bootcode and start_cd.elf the increase in boot time means I now miss the first packet of data

Can you measure what the time was 6 months ago compared with now?
start_cd.elf is currently ~630K It was ~470K six months ago.
Having a look, the increase is fairly gradual, rather than being from a single update.

But ~160K extra should only be taking a fraction of a second to load from sdcard - I'm interested in what difference you are seeing.

closed 2017-05-29 by ghollingworth