raspberrypi/firmware #704: 4.4.38 fails to boot on a rpi2

jonozzz, 2016-12-22:
I just upgraded to 4.4.38 yesterday and noticed that the RPi2 fails to boot up again. The screen shows the 4 different colors, with no other activity. Green LED is off, keyboard is unusable, no error messages of any kind.

I replaced the start.elf and kernel7.img with the previous working version and it booted back up normally.

Anyone has seen this issue?

argos66, 2016-12-22:
I have exactly the same problem.
Temporary solution is to return to the precedent /boot

nmaas87, 2016-12-28:
The same thing with rpi1 - it does not boot anymore, no leds come on, except the normal power led (red)
4.4.37 works.
Seems like support for brcm2708 and brcm2709 is missing in 4.4.38.

ghollingworth, 2016-12-29:
How did you update to 4.4.38?

Did you just do sudo rpi-update

Gordon

nmaas87, 2016-12-29:
@ghollingworth Yes, that is correct, on a RPi 1 v 2.0 Wheezy Maschine

ghollingworth, 2016-12-29:
Wheezy?  So you're not using the latest Jessie builds?

Gordon

nmaas87, 2016-12-29:
I do use Jessie on some of my RPis but I also got some Appliances which still run Wheezy, but are currently on the way to being Dockerized :)

wapolinar, 2016-12-29:
I also had issues with 4.4.38 on my RPI2, but after updating fixup.dat, start.elf and bootcode.bin, it started successfully.

pelwell, 2016-12-29:
When you say you updated those firmware files, where did the replacements come from (including versions)?

wapolinar, 2016-12-29:
Sorry, I should have described it in greater detail:
a) I compiled the kernel from raspberrypi/linux (rpi-4.4.y), and replaced my /boot/kernel7.img.
After that, the RPI2 did not boot.
(I followed: https://www.raspberrypi.org/documentation/linux/kernel/building.md )
b) I copied the files fixup.dat, start.elf and bootcode.bin from raspberrypi/firmware (master, current version) and replaced my (apparently) old versions.
After that, the RPI2 did boot.
(And now I'm thinking that I maybe should create an issue against the documentation which is also on Github (did not see that - until now))

dimovnike, 2017-01-16:
I am affected by this issue too. Not only my custom compiled kernel is not working (on Pi model 2) but neither the kernel from rpi-firmware commit 2b0494cd412 (last master) is. Also i tried git bisect and the first commit that breaks it is c05b9e6 ( kernel: Bump to 4.4.6 ). What went wrong and how to fix it?

popcornmix, 2017-01-16:
Are you saying a clean raspbian install will fail to boot after
```sudo rpi-update c05b9e6```
Do you have any custom settings in config.txt / cmdline.txt?

dimovnike, 2017-01-16:
I probably have custom settings in config.txt / cmdline.txt but i'm testing now with a clean raspbian install.

dimovnike, 2017-01-16:
sudo rpi-update c05b9e6 worked fine. So it's probably related to my options in cmdline.txt/config.txt.

popcornmix, 2017-01-16:
Well if you post your options we might spot something that is problematic.

dimovnike, 2017-01-16:
my options are:
cmdline.txt:

`dwc_otg.lpm_enable=0 console=tty1 elevator=deadline`

config.txt

```
disable_overscan=1
device_tree=
hdmi_force_hotplug=1
hdmi_group=1
hdmi_mode=19
hdmi_drive=2
config_hdmi_boost=4
gpu_mem=192
initramfs initrd.cpio 0x00a00000
initrd=0xa00000,0x208580

```
It worked like this for years. I use squashfs as root, mounted from initrd. Also the kernels before c05b9e6 are throwing kernel panic (which is normal because i didn't provide initrd for testing) but with newer kernels - it is stuck in the colored boot screen. Also i tried testing without initrd options - the result was the same.

dimovnike, 2017-01-16:
Tried on a clean install and at least device_tree= is breaking it. Any idea how to disable overlays on a newer kernel?

pelwell, 2017-01-16:
> Any idea how to disable overlays on a newer kernel?

You can't - the 4.4 kernel relies on Device Tree. Why do you want to disable it?

The other reason your boot failed is the explicit load address for your initramfs - it is now too low and overlaps with the kernel; it is much safer to allow the firmware to choose a safe place to load it.

To get your Pi to boot again:

1) change "0x00a00000" to "followkernel"
2) delete the "initrd=" line - it isn't a recognised option
3) delete "device_tree="

This will leave you with:
```
disable_overscan=1
hdmi_force_hotplug=1
hdmi_group=1
hdmi_mode=19
hdmi_drive=2
config_hdmi_boost=4
gpu_mem=192
initramfs initrd.cpio followkernel
```

dimovnike, 2017-01-16:
thanks a lot! as for 

> You can't - the 4.4 kernel relies on Device Tree. Why do you want to disable it?

it was a matter of personal preference of avoiding many overlay files in /boot.

pelwell, 2017-01-16:
Overlays (files found in /boot/overlays) are only loaded if your config.txt includes a "dtoverlay=" line referring to them. Disable Device Tree (if it worked) wouldn't remove the files, but if you really want to delete the overlays directory you can. However, any update will create it again.

dimovnike, 2017-01-16:
Yes I understand that, I usually don't use rpi-update, my kernel is custom compiled. It was just simpler for me to enable all peripherals and then control what is loaded by kernel config (modprobe). So this is not possible anymore? Is there an "enable-everything" setting?

pelwell, 2017-01-16:
So you find blacklisting and modprobe-ing simpler? You can still do that, but you will need to enable the devices in DT. With standard .dtb files you have to use a dtparam= or dtoverlay= in config.txt. Alternatively you can use a custom dtb file, and there is a command that will make it easier: dtmerge
```
dtmerge /boot/bcm2709-rpi-2-b.dtb hacked.dtb - spi=on i2c_arm=on
```
This will take the original bcm2709-rpi-2-b.dtb, apply two parameters (spi=on, i2c_arm=on) and write the composite to hacked.dtb. If you want to apply an overlay, replace the '-' with the name of the overlay file.
Now you can copy hacked.dtb to /boot and add "device_tree=hacked.dtb" to config.txt. If you don't want any config.txt modifications then you can overwrite the original bcm270-rpi-2.b.dtb.

The dtdiff command compares dtb files, which can be useful to see what the various parameters and overlays do. Since running dtmerge on an original .dtb makes a few harmless changes, you may prefer to generate a new base dtb for comparison:
```
dtmerge /boot/bcm2709-rpi-2-b.dtb base.dtb -
dtdiff base.dtb hacked.dtb
```

dimovnike, 2017-01-16:
Looks like it's time to switch from blacklisting and modprobe-ing to overlay system :) Thank you for the information, I didn't know you can merge those files.

pelwell, 2017-01-16:
@jonozzz @wapolinar @basisbit @plasma0713 Are you also initramfs users? Does the above advice solve your problem?

dimovnike, 2017-01-16:
removed the unsupported settings from config.txt, now i have:
```
disable_overscan=1
hdmi_force_hotplug=1
hdmi_group=1
hdmi_mode=19
hdmi_drive=2
config_hdmi_boost=4
gpu_mem=192
```

but still no luck (this is with bootloader and kernels from 2b0494cd41). I also tried with initrd as per your @pelwell suggestion but it also didn't work.

jonozzz, 2017-01-16:
I'm not using initramfs... Do we need to merge the dtb's to get it to boot?

config.txt:
```dtparam=audio=on
start_x=1
gpu_mem=128
device_tree=
disable_splash=0
boot_delay=1
```

cmdline.txt:
`dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.mode=force fsck.repair=yes rootwait`

pelwell, 2017-01-17:
There is no need to use anything but the standard dtbs - I was just giving @dimovnike an alternative to using dtoverlay and dtparam in config.txt.

Remove the `device_tree=` line from your config.txt - the 4.4 kernel requires Device Tree in order to work. You've been getting away with it until the 4.4.38 release because kernel images included a trailer that indicated that it was DT-capable, causing the firmware to ignore the device_tree= setting. From 4.4.38 onwards we aren't adding the trailer - we can assume that all kernels are DT-aware unless config.txt says otherwise, but that means we can no longer ignore `device_tree=`.

jonozzz, 2017-01-23:
Removed the device_tree= line and upgraded to 4.4.44 and it booted back fine:

`Linux raspberrypi 4.4.44-v7+ #950 SMP Sun Jan 22 16:20:42 GMT 2017 armv7l GNU/Linux
`

Closing the issue.

closed 2017-01-23 by jonozzz