raspberrypi/firmware #706: no wan0 after upgade

tim3dman, 2017-01-01:
After updating my Raspberry Pi2 from 4.4.34 to 4.4.38 I lose wan0 and am unable to reconfigure?

pelwell, 2017-01-02:
1. Which distribution are you running?
2. How did you upgrade?
3. What is the output of `dmesg` after booting?
4. Do you mean `wan0` , or is that a typo?

tim3dman, 2017-01-03:
I'm using jessie lite, I used apt-get upgrade dist I think? not sure about dmesg as I've rolled back to a backed up 4.4.38. By wan0, when I ifconfig I only have lan and loopback but know wifi.

popcornmix, 2017-01-03:
What is the wifi device?
Output of ``lsusb`` and ``lsmod`` on working and not working systems may be useful.

Ruffio, 2017-01-09:
@tim3dman what do you day?

Ruffio, 2017-01-14:
@tim3dman what do you say to @popcornmix commands? Has this issue been resolved?

closed 2017-01-19 by tim3dman