raspberrypi/firmware #708: Please allow the ability of rotate screen while system alive

ajefr, 2017-01-02:
Hello,
 As issue #637 will it be possible to rotate the screen orientation while the system is alive ?
 A comand like :
`vcgencmd display_rotate 0`
or 
`vcgencmd display_rotate 1`
 will be perfect.
Thanks in advance !

popcornmix, 2017-01-03:
This is unlikely. There are some start of day allocations needed when running in rotated mode that may be tricky to implement dynamically.

You may have more luck with the [experimental arm side graphics driver](https://www.raspberrypi.org/blog/another-new-raspbian-release) which should support the standard xrandr mechanism of rotating the frame buffer.

ajefr, 2017-01-03:
Hi'
 Thanks for your answer.
 The aim of my request is still for emulation purpose.
 If I launch an emulator with screen rotated (rotate=1), I get some video tearing on the dpi output.
 While runing it with rotate=0 give a perfect output.
 Sadly, I have to use rotate=1 to get correct orientation for selection menu.
 So the game will be to run pi with rotate=1, when launching emulator, switch to rotate=0 and when returing back to selection menu, restore the rotate=1.
 I hope my explanation is clear enough.
 Or the another solution will be to avoid tearing on rotate=1 mode.
 FYI I'm using the very last firmware (updated yesterday)
Best regards

popcornmix, 2017-01-04:
How does emulator write to display?

If it is using dispmanx for display, you can rotate on demand.
Obviously OpenGLES would also allow cheap rotations to be performed.

They would be better/cheaper options that using display_rotate.

JamesH65, 2017-12-21:
@ajefr Can this be closed? We are unlikely to provide run time changing of orientation.

closed 2018-06-29 by JamesH65