raspberrypi/firmware #711: missed rt2x00leds module in stock

Kotove, 2017-01-04:
Hello,
can anyone from maintainers explain, why you didn't including rt2x00leds module while compiling driver module?
Source tree has this 3y old file
https://github.com/raspberrypi/linux/tree/master/drivers/net/wireless/rt2x00

but I wasn't lucky to find it under any 'firmware' repos. 
https://github.com/raspberrypi/firmware/tree/master/modules/4.4.38-v7+/kernel/drivers/net/wireless/rt2x00 for example.

```
root@raspberrypi:/# lsmod | grep rt2
rt2800usb              17554  0
rt2800lib              71894  1 rt2800usb
rt2x00usb               8943  1 rt2800usb
rt2x00lib              37529  3 rt2x00usb,rt2800lib,rt2800usb
mac80211              532162  3 rt2x00lib,rt2x00usb,rt2800lib
crc_ccitt               1278  1 rt2800lib
cfg80211              427855  3 brcmfmac,mac80211,rt2x00lib
```

```
root@raspberrypi:/# ls /lib/modules/4.4.38-v7+/kernel/drivers/net/wireless/rt2x00/
rt2500usb.ko  rt2800lib.ko  rt2800usb.ko  rt2x00lib.ko  rt2x00usb.ko  rt73usb.ko
```

I've found that accidentally when noticed that my Ralink-based wifi dongle's led does never flash.
While there was led_class subtree /sys/class/leds/rt2800usb-phy2*, with [none] in triggers, I tried to change triggers and received kernel panic/oops, not sure what exactly. But Pi was dead unfortunately.

Is there any chance to have this module be shipped pre-compiled?

popcornmix, 2017-01-04:
Can you be clean which option you want enabled?
```
pi@domnfs:~ $ zcat /proc/config.gz |grep RT2X
CONFIG_RT2X00=m
CONFIG_RT2X00_LIB_USB=m
CONFIG_RT2X00_LIB=m
CONFIG_RT2X00_LIB_FIRMWARE=y
CONFIG_RT2X00_LIB_CRYPTO=y
CONFIG_RT2X00_LIB_LEDS=y
# CONFIG_RT2X00_DEBUG is not set
```
It looks like CONFIG_RT2X00_LIB_LEDS=y is present currently.

Kotove, 2017-01-04:
ok, got it. I should have checked makefile first before posting issue. Now I see that LED support is built-in into rt2x00lib.ko. 
Fortunately, I managed to light up LED by echo 1 > brightness (facepalm).

popcornmix, 2017-01-04:
Okay to close?

Kotove, 2017-01-04:
yeap, sure. Thanks for your attention.

closed 2017-01-04 by popcornmix