raspberrypi/firmware #712: nTimestamp member in OMX_BUFFERHEADERTYPE is overwritten

carlonluca, 2017-01-06:
Hello, I'd need to pass custom data from the input buffer provided to the OpenMAX video_decode component to the output buffer structure I get in FillBufferDone callback of the tunnelled egl_render component. I understand input buffers cannot be 1-to-1 mapped to output buffers, but I'd need to somehow pass an index representing an input buffer and receive it in the output buffer. I tried to abuse the nTimestamp field of the OMX_BUFFERHEADERTYPE: I set it before EmptyThisBuffer and I read it in FillBufferDone. OpenMAX specifications indicate that "a component should propagate this field from an input buffer to its associated output buffer". What I see is that some output buffers properly contain the value I sent (so it seems ok for my case), but some others report completely different values, like OpenMAX was replacing the value intentionally. Is this correct? Of course I understand I'm misusing that field, but it seemed legal. Am I misunderstanding the OpenMAX specifications or should it be preserved?
Another element I see in the docs is the OMX_EventMark event, but I'm unsure if it is implemented for the egl_render component: can I set it in the input buffer to the video_decode component and expect the event to be raised on the egl_render component with the data I associated it to? I read in the forums this is not implemented for every component. Thanks.

6by9, 2017-01-07:
What values are you seeing in nTimestamp? You're not just seeing frame reordering due to B-frames?
I'm assuming you're not using image_fx to deinterlace content as that does mess with timestamps.

There is code in both video_decode and egl_render which looks to copy hMarkTargetComponent and pMarkData from input to output. I can't remember the detail of OMX_EventMark, but those 2 fields may be easier to abuse than nTimestamp. Gut feeling is to use odd numbers or 0 in hMarkTargetComponent just to avoid unintentional matches with an actual component handle (will always be a multiple of 4) and actually triggering a buffer mark callback.

carlonluca, 2017-01-08:
I set nTimestamp to a ordered sequence of integers in [1, n]. What I get in the output is a sequence of integers, with some "holes", which I expect, but also where some are simply replaced by high values, over 4000. So I may get something like: 6, 7, 9, 11, 4169, 15, 17, 4489 etc...

Setting the handle to 0 and passing data into the mark field seems to be working properly, thanks. Not sure if nTimestamp is supposed to behave like that... however I suppose the issue can be closed for the moment if you want to.

6by9, 2017-01-08:
That is slightly odd.
I take it you have OMX_SKIP64BIT defined in your build, otherwise OMX_BUFFERHEADERTYPE mismatches between ARM and GPU.
I can't remember where and when egl_render may drop frames which, may explain the holes. Is there a scheduler in your pipeline somewhere, or is it only paced by the speed of video_decode?
Is your input data actually framed, otherwise you're totally right that you can't rely on a 1-1 mapping, and I would expect to get frames out with nFlags including OMX_BUFFERFLAG_TIME_UNKNOWN.

Ah, just spotted a comment in the source:
```
         // keep track of the next frame time, in case the next
         // frame has a time unknown flag set.
```
So it will try to guess the correct timestamp based on either the value reported in the header bytes, or failing that using format.video.xFramerate off the input port PortDefinition. This will only apply where output frames have OMX_BUFFERFLAG_TIME_UNKNOWN set due to not being a 1-1 mapping of an input buffer.

carlonluca, 2017-01-12:
Yes, I do define OMX_SKIP64BIT. Is this wrong? And yes, my code includes the clock and the scheduler.
So you're saying that I should expect nTimestamp to be written by OpenMAX for those frames that are not 1-1 mapping of an input buffer? Am I understanding correctly? This might explain why I keep seeing "holes" even using pMarkData.
The solution you proposed seems to be working: pMarkData seems to never be "overwritten" by OpenMAX. But, assuming 2 input buffers are mapped to 1 frame, and pMarkData is different for the two input buffers, which pMarkData is returned in the output buffer?
Thank you for your help, it was very very useful.

6by9, 2017-01-13:
> Yes, I do define ```OMX_SKIP64BIT```. Is this wrong?

No, 100% right. Without it the alignment of OMX_BUFFERHEADERTYPE at and after the nTimeStamp field goes wrong (so particularly nFlags). It's a nasty in the OpenMax IL spec that hadn't been noticed during the Broadcom days, and now there is no easy way to make a backwards compatible change.

I believe if a single input buffer contains 2 complete frames, then nTimestamp of the second frame decoded will be calculated based on the framerate data either in the stream header, or xFramerate defined in the input port's PortDefinition. It is assumed that the timestamps are in microseconds, as specified in the IL spec, so you increase of ~4200 implies something is telling it the clip is ~240fps. Slightly odd.

Do you have no way of providing the stream split into frames? What is your data source? Anything where there is a chance of one IL buffer containing 2 video frames, or 1 video frame being split over 2 IL frames is going to cause some grief.

JamesH65, 2017-12-21:
@carlonluca Any status update?

carlonluca, 2018-03-03:
This question was related to implementing a accelerated decoder for Qt WebEngine (chromium) and somehow the problem remained. I had something running but I had to pause it. As I guess nothing unexpected was found in OpenMAX I'd say it makes sense to close this. Thanks for your help.

closed 2018-06-29 by JamesH65