raspberrypi/firmware #714: Layers wrapping on X-axis at 2048px

mihalop, 2017-01-10:
We have several Pi's working at 4K resolution (3840x2160 at 25Hz), as discussed on [the relevant forum thread](https://www.raspberrypi.org/forums/viewtopic.php?f=38&t=79330&start=75). They have been working great, being quite stable at 25Hz, with a few layers. The compositor seems able to cope with it.

But we have an issue with placing layers over 2048px on the X-axis. It seems that the firmware wraps values for the X-axis on 2048. So, the following:

`omxplayer --win '2047 0 3840 2160' video.mp4`

works fine, but this one:

`omxplayer --win '2048 0 3840 2160' video.mp4`

wraps the X coordinate for the layer position to 0. This does not happen with the Y-axis.

I know that 4K is not officially supported, but it any chance we can get this fixed? Thanks!

pelwell, 2017-01-10:
As far as I can tell, you have hit a hardware limitation. The specification for the HVS says that the X coordinate of the start of the image is a 12-bit signed integer (so -2048 to 2047) - the sign is to allow an image to be placed off to the left of the display. For reference, the Y coordinate is an unsigned 11-bit integer (0-2047).

popcornmix, 2017-01-10:
As Phil says, this is a hardware issue that can't be fixed.
Ideally both X and Y coordinates should have used 12-bit unsigned numbers (you can cope with negative dest coordinates in software by calculating the corresponding positive source coordinate).
But that is not how the hardware is, so placing a dispmanx layer at position >=2048 is not supported.

mihalop, 2017-01-10:
_Disclaimer: doing tests now with omxplayer, so placement might be handled differently within omxplayer._

@pelwell The Y-coordinate seems to be at least an unsigned 12-bit int, since I can correctly place a layer on (0, 2100).

@popcornmix I used extreme negatives (-12000), testing both axes, and it worked fine, stretching the output as expected. Do you handle this within omxplayer or is it processed by the firmware somehow before reaching the HVS?

In any case, thanks for the clarification.

popcornmix, 2017-01-10:
For negative dest coordinates we force them to zero and adjust the source coordinates.

mihalop, 2017-01-11:
Thanks for the clarification, just confirmed that with some test code.  :-(

closed 2017-01-11 by mihalop