raspberrypi/firmware #715: Weird hang with next branch

asavah, 2017-01-10:
Pi3
Custom OS, kernel 4.9.2 (defconfig) from rpi-4.9.y branch

I' ve decided to try the next branch of firmware and I stumbled upon weird hang.

```
 /usr/bin/localedef -v -i en_GB -c -f UTF-8 -A /usr/share/locale/locale.alias en_GB.UTF-8
```
Just hangs the pi, no crash, no backtrace, nothing in syslog, nothing on screen,
green light stays on and the pi simply hangs and reboots itself after 5-10 seconds.
_Same build with same kernel but with firmware from master branch works perfectly._

```
vcgencmd version
version 8d26b9e653bfa3160da673aad9f802058e8f7174 (clean) (release)

vcgencmd measure_temp
temp=53.7'C

vcgencmd get_throttled
throttled=0x0
```

To try and troubleshoot the issue I've disabled the all the overlays (lirc-rpi and etc) and all the overclock options, no dice.
Tried disabling preloading libarmmem.so - no difference.
Tried 2 known good sdcards and fscked them - no errors reported.
Tried running localedef within gdb - couldn't see anything, it just hangs.

Hammering the sdcard with dd - all fine.

Running the localedef with strace produces
```
<snip>
open("i18n", O_RDONLY)                  = -1 ENOENT (No such file or directory)
open("/usr/share/i18n/locales/i18n", O_RDONLY) = 4
fstat64(4, {st_mode=S_IFREG|0644, st_size=157975, ...}) = 0
mmap2(NULL, 157975, PROT_READ, MAP_SHARED, 4, 0) = 0x761e2000
_llseek(4, 157975, [157975], SEEK_SET)  = 0
brk(0x1739000)                          = 0x1739000
brk(0x1782000)                          = 0x1782000
mmap2(NULL, 684032, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7613b000
brk(0x17a7000)                          = 0x17a7000
brk(0x17e7000)                          = 0x17e7000
brk(0x1847000)                          = 0x1847000
brk(0x1887000)                          = 0x1887000
mmap2(NULL, 528384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x760ba000
mmap2(NULL, 528384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x76039000
mmap2(NULL, 528384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x75fb8000
brk(0x17e7000)                          = 0x17e7000
mmap2(NULL, 528384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x75f37000
brk(0x175a000)                          = 0x175a000
mremap(0x7613b000, 684032, 1363968, MREMAP_MAYMOVE) = 0x75dea000
mremap(0x75dea000, 1363968, 3674112, MREMAP_MAYMOVE) = 0x74d78000
mremap(0x74d78000, 3674112, 7344128, MREMAP_MAYMOVE) = 0x74677000
mremap(0x760ba000, 528384, 1052672, MREMAP_MAYMOVE) = 0x760ba000
mremap(0x76039000, 528384, 1052672, MREMAP_MAYMOVE) = 0x75e36000
mremap(0x75fb8000, 528384, 1052672, MREMAP_MAYMOVE) = 0x75fb8000
mremap(0x75f37000, 528384, 1052672, MREMAP_MAYMOVE) = 0x75d35000

open("i18n", O_RDONLY)                  = -1 ENOENT (No such file or directory)
open("/usr/share/i18n/locales/i18n", O_RDONLY) = 4
fstat64(4, {st_mode=S_IFREG|0644, st_size=157975, ...}) = 0
mmap2(NULL, 157975, PROT_READ, MAP_SHARED, 4, 0) = 0x761e2000
_llseek(4, 157975, [157975], SEEK_SET)  = 0
brk(0x1739000)                          = 0x1739000
brk(0x1782000)                          = 0x1782000
mmap2(NULL, 684032, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7613b000
brk(0x17a7000)                          = 0x17a7000
brk(0x17e7000)                          = 0x17e7000
brk(0x1847000)                          = 0x1847000
brk(0x1887000)                          = 0x1887000
mmap2(NULL, 528384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x760ba000
mmap2(NULL, 528384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x76039000
mmap2(NULL, 528384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x75fb8000
brk(0x17e7000)                          = 0x17e7000
mmap2(NULL, 528384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x75f37000
brk(0x175a000)                          = 0x175a000
mremap(0x7613b000, 684032, 1363968, MREMAP_MAYMOVE) = 0x75dea000
mremap(0x75dea000, 1363968, 3674112, MREMAP_MAYMOVE) = 0x74d78000
mremap(0x74d78000, 3674112, 7344128, MREMAP_MAYMOVE) = 0x74677000
mremap(0x760ba000, 528384, 1052672, MREMAP_MAYMOVE) = 0x760ba000
mremap(0x76039000, 528384, 1052672, MREMAP_MAYMOVE) = 0x75e36000
mremap(0x75fb8000, 528384, 1052672, MREMAP_MAYMOVE) = 0x75fb8000
mremap(0x75f37000, 528384, 1052672, MREMAP_MAYMOVE) = 0x75d35000

<HANG>
```

If there is any info you need or if I can do anything to help you troubleshoot this - just let me know,
will be glad to help.

asavah, 2017-01-11:
Edit: To add more facts to the mystery:

The issue happens only with:
```
start_x.elf
fixup_x.dat

vcgencmd version
Jan 10 2017 20:07:45
Copyright (c) 2012 Broadcom
version d877fe98bc014bc2169f378895fc15b332f9f055 (clean) (release)
```
```
start.elf
fixup.dat
```
Work fine !

popcornmix, 2017-01-11:
Are you able to provoke the same failure with raspbian and ``BRANCH=next rpi-update``?
Can you show your config.txt when it is failing?

asavah, 2017-01-11:
config.txt - failing (overclock options disabled to rule them out)
```
start_file=start_x.elf
fixup_file=fixup_x.elf
kernel=kernel7.img
framebuffer_depth=32
framebuffer_ignore_alpha=1
disable_overscan=1
disable_splash=1
#arm_freq=1300
#core_freq=500
#gpu_freq=500
#v3d_freq=500
#h264_freq=500
#sdram_freq=580
#sdram_schmoo=0x02000020
#over_voltage_sdram=5
#over_voltage=2
#force_turbo=1
gpu_mem=320
#dtparam=sd_overclock=100
dtoverlay=lirc-rpi
dtparam=watchdog=on
dtparam=random=on
dtparam=audio=on
hdmi_force_hotplug=1
hdmi_edid_file=1
decode_MPG2=<snip>
decode_WVC1=<snip>
```

config.txt - working fine (even with overclock)
```
kernel=kernel7.img
framebuffer_depth=32
framebuffer_ignore_alpha=1
disable_overscan=1
disable_splash=1
#arm_freq=1300
core_freq=500
gpu_freq=500
#v3d_greq=500
#h264_greq=500
sdram_freq=580
sdram_schmoo=0x02000020
over_voltage_sdram=5
over_voltage=2
#force_turbo=1
gpu_mem=320
dtparam=sd_overclock=100
dtoverlay=lirc-rpi
dtparam=watchdog=on
dtparam=random=on
dtparam=audio=on
hdmi_force_hotplug=1
hdmi_edid_file=1
decode_MPG2=<snip>
decode_WVC1=<snip>
```

Will try to reproduce on raspbian-pixel later.

popcornmix, 2017-01-11:
```
start_file=start_x.elf
fixup_file=fixup_x.elf
```
``start_x=1`` is preferred form.

```
kernel=kernel7.img
framebuffer_depth=32
framebuffer_ignore_alpha=1
```
I believe these are not required as they will be defaults.

Personally I suspect the start._x/start difference is a red herring. I imagine it is just moving objects around in physical memory (start_x.elf is larger) that is masking/unmasking the issue.

It may be interesting to try decreasing gpu_mem in the start_x case (and/or increasing gpu_mem in the start case). Note: gpu_mem=320 is pretty large. What do you require that for?

asavah, 2017-01-11:
I don't remember why I set 320 long time ago.
I've already tried with 256 and 128 - no difference.

Will try with tweaking the options as you recommend.

> I imagine it is just moving objects around in physical memory (start_x.elf is larger) that is masking/unmasking the issue.

I suspected that, but my low level knowledge is not that good.
On one of straces I've seen spam of brk() calls which AFAIK may suggest memleaks (?)

asavah, 2017-01-11:
Got it!
```
start_file=start_x.elf
fixup_file=fixup_x.elf
```
These options are somehow causing the issue.
```
start_x=1
```
Works fine regardless of gpu_mem amount.

popcornmix, 2017-01-11:
Ah - you have a typo:
```
start_file=start_x.elf
fixup_file=fixup_x.dat
```
You were pointing at a non-existent fixup file...

asavah, 2017-01-11:
OMG
facepalm.jpg

Sorry for all the noise.
With "dat" it works fine.

However it was working and been stable for months even with wrong filename on master.
This is typo has been there forever if my git repo doesn't lie.

Sorry again for making you waste your valuable time.

Clearly PEBCAC case.

closed 2017-01-11 by asavah

evilpower, 2017-01-13:
That's because it wasn't your typo. That was the original way ( https://www.raspberrypi.org/forums/viewtopic.php?p=269992#p269992 ).
@popcornmix If I understand correctly, start_x=1 should be used instead?

popcornmix, 2017-01-13:
@evilpower Yes, ``start_x=1`` is the recommended notation.
I've edited the linked post to correct the typo (and switch to preferred notation).