raspberrypi/firmware #716: config.txt: „total_mem“ in [piX] should not affect non-piX devices

herrmeier01, 2017-01-12:
bootloader configuration config.txt

[pi1]
total_mem=512

This option should only affect Pi1 board but not Pi2 and Pi3 board.
But it will force the Pi2 and Pi3 to use 512MB.

popcornmix, 2017-01-13:
config.txt is actually parsed by both bootcode.bin and start.elf.
bootcode.bin is very minimal. The code runs from the cache before sdram is enabled.
As such the config.txt filtering is more basic. The only filter currently supported is serial number.

Most config.txt options are start.elf options.
total_mem and gpu_mem are bootcode options hence the issue.

Let me look into parsing the processor type. If it can be done in a small amount of code, I'll add it. (pi1 is easy to identify, but Pi2/Pi3 distinction involves parsing the revision field from otp, so is a little more involved).

Note: some filters, like EDID based sections will be impossible from bootcode.bin (but fortunately you are unlikely to want to change the memory size based on monitor attached).

popcornmix, 2017-01-13:
Could you test https://drive.google.com/uc?id=0B-6zmEDJwxZEVGtIVHg0dmotZ2s&export=download
Hopefully it will fix both of your issues.

herrmeier01, 2017-01-14:
Great job! Both issues are fixed (Pi1 256M and Pi1 512M) and "total_mem" works fine now

Thanks a lot

closed 2017-01-14 by herrmeier01