raspberrypi/firmware #718: nBufferCountActual limits buffers to 8

carlonluca, 2017-01-12:
Hello, it seems that when setting `nBufferCountActual` for an `OMX_PARAM_PORTDEFINITIONTYPE` for the output port of the egl_render component to a value > 8 the `OMX_SetParameter` call fails with the error `OMX_ErrorBadParameter`. Is this correct?
I typically use 4 buffers and the result seems perfect. I'm currently working on a project where it seems the rendering engine (which is not under my control) may be retaining frames more than needed and I would like to test what happens with more than 8 output buffers. Is there any simple way to remove this limitation or is it too complex? It doesn't seem to be a memory restriction: maybe there is something I can tune in userspace libs to remove the limitation (at least temporarily for a test)? Thanks.

6by9, 2017-01-13:
Checking the component source code:
```
#define EGL_QUEUE_SIZE (1<<3) /* must be power of 2 */
...
            // Note that we impose an additional limit of EGL_QUEUE_SIZE on the maximum number of output
            // buffers
```
Ought to be documented in the component docs, but you are correct.
Whilst it looks fairly trivial to increase, another comment implies there is a link between this and something inside the GL side, so I can't say for certain what the full implications are and would therefore be wary of increasing it.

carlonluca, 2017-01-14:
I understand the check you are mentioning is in the closed portion right? Ok, no problem. I'm trying whatever I can to complete my project and this attempt was part of that, but I understand that using more than 8 buffers might not be a regular and advisable behaviour.

JamesH65, 2017-12-21:
@carlonluca Are you happy for this to be closed?

carlonluca, 2017-12-21:
Yes, no problem. You can close. Thanks.

closed 2017-12-21 by JamesH65