raspberrypi/firmware #720: USB Boot not working on the Raspberry Pi 1B

mskwon, 2017-01-14:
Trying to boot from a USB drive using a Raspberry Pi 1B (not 1B+) with a SD card containing only bootcode.bin produces the rainbow screen with green LED blinking in a 7 pulse pattern, however booting up never happens although the USB drive does boot up with a RPi2B with bootcode.bin SD card or RPi3B with MSD boot enabled. This is using the most recent firmware/bootcode.bin as of 1/14/2017.

popcornmix, 2017-01-14:
Are you saying the previous firmware does work on Pi 1? Or has it never worked on Pi 1?

mskwon, 2017-01-15:
I guess looking into it more, it's never worked and that it doesn't work has been known since at least [August 2016](https://www.raspberrypi.org/blog/pi-3-booting-part-ii-ethernet-all-the-awesome/).

A) Should this issue be closed?
B) Is USB boot for RPi1B in the works / planned?

popcornmix, 2017-01-15:
Ping @ghollingworth 
Should USB boot work on a Pi1B (with bootcode.bin on sdcard)?

ghollingworth, 2017-01-16:
It should do, although I didn't have one on my desk when I was testing it and therefore it's probably broken (Morgan's law).  Will have a look...

Gordon

ghollingworth, 2017-01-20:
Try again with the new rpiboot (note the new location)

https://github.com/raspberrypi/usbboot

closed 2017-01-20 by ghollingworth

mskwon, 2017-01-21:
You'll have to forgive me, but could I get some help setting this up?

I can't figure out how to derive the bootcode.bin to place on the SD card for the life of me.

I fear we may have had a miscommunication - I wasn't aware there is a "USB Boot" allowing for booting a RPi from a host computer. Rather than this USB Boot, I was referring instead to the PXE/MSD booting introduced August 4, 2016. Are these the same?