raspberrypi/firmware #721: USB boot fails without rootdelay cmdline parameter (Firmware 14-1-2017)

tvoverbeek, 2017-01-15:
I am testing USB boot on RPi3 again, now with an 8GB USB stick I bought at BestBuy when visiting the US. I am using the firmware from Jan 14 with kernel 4.9.3 from the firmware NEXT branch.

Without a 'rootdelay=4' in cmdline.txt I get a kernel panic because it cannot mount the root device (/dev/sda2). See the bootlogfail.txt attached which is the boot log captured from the serial console.
Adding 'rootdelay=4' to cmdline.txt results in a successful boot. See dmesg.txt attached (output from dmesg after boot).
Do not know if this is a kernel issue and/or if this workaround should be documented in the usb boot documentation.
[bootlogfail.txt](https://github.com/raspberrypi/firmware/files/706862/bootlogfail.txt)
[dmesg.txt](https://github.com/raspberrypi/firmware/files/706863/dmesg.txt)

ghollingworth, 2017-01-16:
I think this is a pretty standard thing and not a bug in the frimware but a limitation of the hardware.

http://www.linuxtopia.org/online_books/linux_kernel/kernel_configuration/re58.html

tvoverbeek, 2017-01-16:
Agreed, especially after reading the rootdelay description on linuxtopia.org you linked to.
Will close the issue here and submit a doc change/addition on https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bootmodes/msd.md

closed 2017-01-16 by tvoverbeek

ghollingworth, 2017-01-16:
Good idea!

Thanks