raspberrypi/firmware #722: Custom Resolution in EDID not accepted

Altaah, 2017-01-17:
If i want to use a custom resolution, i can onl do this by using "hdmi_timings=" in the config.txt.

In my example 
`hdmi_timings=480 0 1 41 2 272 0 2 10 2 0 0 0 60 0 9009000 3`
is working fine.

But for the application it would be essential to read out the desired resolution out of the edid.
An edid file containing only a (this) custom resolution is not accepted by the Rpi.(according to this [thread](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=171058&p=1096083#p1096083)) 

I attached a EDID file, which should have the same timings like above (if i haven´t made any mistake). 
[edid_test.zip](https://github.com/raspberrypi/firmware/files/710826/edid_test.zip)
This file is not working for me but i think it should.

Getting the correct hdmi timings automatically out of the edid is an important feature for the usage of the Rpi in embedded systems.

JamesH65, 2019-01-14:
Hi @popcornmix, is this something we should still be looking in to?

6by9, 2019-01-14:
I've had a quick look. How about if ```hdmi_group``` and ```hdmi_mode``` are set to the custom mode, but no ```hdmi_timings``` value has been provided, then take the detailed timings from the EDID in place of ```hdmi_timings```?

What's happening currently is that it will select VGA60 as it is mandatory to support that over HDMI.
The output from edidparser (which is the same logic as used in the firmware) is:
```
pi@raspberrypi:~ $ edidparser tmp/edid_test.bin
Enabling fuzzy format match...
Parsing tmp/edid_test.bin...
HDMI:EDID version 1.4, 0 extensions, landscape aspect ratio 1.77:1
HDMI:EDID features - videodef 0x92 !standby !suspend !active off; colour encoding:RGB444; sRGB is not default colourspace; preferred format is native; support discrete frame rate
HDMI:EDID ignored unknown descriptor tag 0x10
HDMI:EDID ignored unknown descriptor tag 0x10
HDMI:EDID ignored unknown descriptor tag 0x0
HDMI:EDID does not yet know monitor vertical range, setting to default 24 to 120Hz
HDMI:EDID failed to find a matching detail format for 480x272p hfp:1 hs:41 hbp:-42 vfp:2 vs:10 vbp:-12 pixel clock:9 MHz
HDMI:EDID calculated refresh rate is 69 Hz
HDMI:EDID guessing the format to be 480x272p @70 Hz
HDMI:EDID found unknown detail timing format: 480x272p hfp:1 hs:41 hbp:-42 vfp:2 vs:10 vbp:-12 pixel clock:9 MHz
HDMI:EDID established timing I/II bytes are 00 00 00
HDMI:EDID standard timings block x 8: 0x0100 0100 0100 0100 0100 0100 0100 0100 
HDMI:EDID unknown standard timing 256x160 @ 60 Hz aspect ratio (16:10)
HDMI:EDID unknown standard timing 256x160 @ 60 Hz aspect ratio (16:10)
HDMI:EDID unknown standard timing 256x160 @ 60 Hz aspect ratio (16:10)
HDMI:EDID unknown standard timing 256x160 @ 60 Hz aspect ratio (16:10)
HDMI:EDID unknown standard timing 256x160 @ 60 Hz aspect ratio (16:10)
HDMI:EDID unknown standard timing 256x160 @ 60 Hz aspect ratio (16:10)
HDMI:EDID unknown standard timing 256x160 @ 60 Hz aspect ratio (16:10)
HDMI:EDID unknown standard timing 256x160 @ 60 Hz aspect ratio (16:10)
HDMI:EDID adding mandatory support for DMT (4) 640x480p @ 60Hz
HDMI:EDID filtering formats with pixel clock > 162 MHz or h. blanking > 1023
HDMI:EDID no known preferred format has been set
HDMI:EDID filtering preferred group has been changed from Invalid to DMT
HDMI:EDID best score mode initialised to DMT (4) 640x480p @ 60 Hz with pixel clock 25 MHz (score 0)
HDMI:EDID best score mode is now DMT (4) 640x480p @ 60 Hz with pixel clock 25 MHz (score 36864)
HDMI:EDID preferred mode is updated to DMT (4) 640x480p @ 60 Hz with pixel clock 25200000 Hz
HDMI:EDID has only DVI support and no audio support
edidparser exited with code 0
```